openwrt/openwrt #12527: DIR-885L WAN speed degrade

lsd-techno, 2023-05-04:
### Describe the bug

When use OpenWrt 22.03.x firmware DIR-885L WAN speed degrade from 1000 Mbps to 400~600 Mbps

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

All

### Device

DIR-885L

### Image kind

Official downloaded image

### Steps to reproduce

Flash DIR-885L with OpenWrt 22.03.x firmware.

Configure.

### Actual behaviour

WAN to LAN speed not exceed 600 Mbps

### Expected behaviour

WAN to LAN speed 1000 Mbps

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lsd-techno, 2023-05-07:
> Degradation compared to what?

Compare to d-link firmware.

[image]

brada4, 2023-05-07:
Try enabling firewall flow offloads. They are certainly enabled in vendor firmware.