openwrt/openwrt #12530: Device CJ-Hello HYC-G920 not available in downloads.

momothefox, 2023-05-04:
### Describe the bug

Device added before 22.03.4 is out 
#12230 

available snapshot only 
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=hanyang_hyc-g920

not seen in 22.03.4 or 5 

### OpenWrt version

22.03.4

### OpenWrt target/subtarget

ramips

### Device

CJ-Hello HYC-G920

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-04:
Device support is not backported typically as per policy
https://openwrt.org/docs/guide-developer/device-support-policies#device_support_backports

momothefox, 2023-05-07:
@brada4  Can you please define what is missing?

brada4, 2023-05-07:
The backport? You have to wait for next major release like 23.x or 24.x whatever that will be.

momothefox, 2023-05-07:
Thanks @brada4

closed 2023-05-07 by momothefox

brada4, 2023-05-07:
You can follow developments here:
https://openwrt.org/docs/guide-developer/releases/goals/23.xx

momothefox, 2023-05-15:
@Ahmed5871205  check out #12568