openwrt/openwrt #12536: AX6 ipq807x: Weak Wifi signal since latest updates

Jonsonmon, 2023-05-05:
### Describe the bug

Since last updates, the wifi is limited to 23Db 5GHZ and 20Db 2.4ghz.
now the wifi signal is very weak!

Before the updates it was 29Db 5ghz and 27Db 2.4Ghz.

### OpenWrt version

r22702-cf8d861978

### OpenWrt target/subtarget

ipq807x/generic

### Device

Redmi AX6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-05:
Seems like you successfully applied regulatory domain configuration applicable to your location.

Jonsonmon, 2023-05-05:
> Seems like you successfully applied regulatory domain configuration applicable to your location.

No, it's on CN, same as always, and it was 29 dbi before few updates, not it's 23 and very weak signal.

brada4, 2023-05-05:
All matches up (last change on 01/01/22 for CN)
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n379

Jonsonmon, 2023-05-05:
> All matches up (last change on 01/01/22 for CN) https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n379

Wow, OK , good to know. Thank you.

Jonsonmon, 2023-05-08:
> All matches up (last change on 01/01/22 for CN) https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n379

OK look,  I See the 20dbi of CN is for DFS .
im not using DFS, im on 80Mhz .
And it was on 29DBi 1 month ago. and the list is updated to 2022... its not logical.
Something is wrong!

brada4, 2023-05-08:
DFS means radar detection is required to operate on particular channel, in principle that means ACS must be used (setting channel selection to automatic)
You can fine-tune ACS to exclude DFS-required channels alltogether
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n219
(or read around there on how to make hostapd prefer stronger channels)

It is to be saved into /etc/config/wireless using slightly different syntax than conf file e.g:
```
config wifi-device 'radio1'
        option type 'mac80211'
        option band '2g'
        option country 'BE'
        option channel 'auto'
        option acs_chan_bias '1:0.8 5:0.8 9:0.8 13:0.8'
        option htmode 'HT20'
```

castiel652, 2023-05-15:
Probably because of this https://github.com/openwrt/openwrt/commit/7475321f46b83cc04296638d6cf31542d11e3e78

Jonsonmon, 2023-05-16:
> Probably because of this [7475321](https://github.com/openwrt/openwrt/commit/7475321f46b83cc04296638d6cf31542d11e3e78)

I dont really understand the issue, but, you think it's gonna be resolved?

castiel652, 2023-05-18:
That's how it's supposed to work.
Before that commit ath11k couldn't get the proper region.

Jonsonmon, 2023-05-18:
> That's how it's supposed to work. Before that commit ath11k couldn't get the proper region.

I see, Thanks!

closed 2023-05-18 by Jonsonmon