openwrt/openwrt #12537: sample app build agaist libstdc++ static link failed

ptpt52, 2023-05-05:
### Describe the bug

```
CC = g++
CFLAGS = -Wall -Werror
LDFLAGS = -static -lstdc++

BINS = my_program

all: $(BINS)

%.o: %.cpp
	$(CC) $(CFLAGS) -c -o $@ $<

my_program: my_program.o
	$(CC) -o $@ $< $(LDFLAGS)

clean:
	rm -f *.o $(BINS)
```
log:
```
make[3]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mbed-tools-0.0.1'
arm-openwrt-linux-muslgnueabi-gcc -Wall -Werror -c -o my_program.o my_program.cpp
arm-openwrt-linux-muslgnueabi-gcc -o my_program my_program.o -static -lstdc++
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.2.0/../../../../arm-openwrt-linux-muslgnueabi/lib/libstdc++.a(functexcept.o):(.ARM.exidx.text.unlikely._ZSt21__throw_bad_exceptionv+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.2.0/../../../../arm-openwrt-linux-muslgnueabi/lib/libstdc++.a(functexcept.o):(.ARM.exidx.text.unlikely._ZSt17__throw_bad_allocv+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
```

### OpenWrt version

sdfs

### OpenWrt target/subtarget

sdfsf

### Device

sfdsf

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-05-05:
If I include “-lgcc_pic” it compiles OK

ptpt52, 2023-05-05:
same with https://github.com/openwrt/openwrt/issues/6710

neheb, 2023-05-15:
What’s the bug here?

neheb, 2023-05-15:
But you found the solution