openwrt/openwrt #12542: Build Failure: gnutls 3.7.8 latest master (last commit c6419d1946996aa859cef9df897a0c1c1f796d11)

AcidSlide, 2023-05-05:
### Describe the bug

Build failing on x86 (x86_64) and ramips (mipsel_24kc)

With last commit:
https://github.com/openwrt/openwrt/commit/c6419d1946996aa859cef9df897a0c1c1f796d11

### OpenWrt version

SNAPSHOT

### OpenWrt target/subtarget

x86/x86_64

### Device

x86

### Image kind

Self-built image

### Steps to reproduce

Lastest master/snapshot with last commit of: https://github.com/openwrt/openwrt/commit/c6419d1946996aa859cef9df897a0c1c1f796d11

Doing a build using: `make -j1 V=sc`

Note: Did the following just to see if it's just a dirty build files
```
$ rm -rf tmp bin build_dir staging_dir
$ ./scripts/feeds update -a && ./scripts/feeds install -a
$ make -j9 download
$ make -j1 V=sc
```

Error produced (x86):
```
make[3]: Entering directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/gnutls-3.7.8'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/gnutls-3.7.8/build-aux/missing aclocal-1.16 -I m4 -I src/gl/m4 -I lib/unistring/m4 --install
m4/gnulib-comp.m4:427: warning: macro 'AM_STDBOOL_H' not found in library
src/gl/m4/gnulib-comp.m4:802: warning: macro 'AM_STDBOOL_H' not found in library
lib/unistring/m4/gnulib-comp.m4:166: warning: macro 'AM_STDBOOL_H' not found in library
missing file gl/stdio-read.c
configure.ac:480: error: expected source file, required through AC_LIBSOURCES, not found
./lib/autoconf/general.m4:2437: AC_FATAL is expanded from...
m4/gnulib-comp.m4:180: gl_INIT is expanded from...
configure.ac:480: the top level
autom4te: error: /home/xxxxxxxx/openwrt/openwrt/staging_dir/host/bin/m4 failed with exit status: 1
aclocal.real: error: autom4te failed with exit status: 1
make[3]: *** [Makefile:2217: aclocal.m4] Error 1
make[3]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/gnutls-3.7.8'
make[2]: *** [Makefile:269: /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/gnutls-3.7.8/.built] Error 2
make[2]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#58.77#18.47#74.77
    ERROR: package/feeds/packages/gnutls failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[1]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make: *** [/home/xxxxxxxx/openwrt/openwrt/include/toplevel.mk:231: package/feeds/packages/gnutls/compile] Error 2
```

Error with ramips/mt7621
```
make[3]: Entering directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/xxxxxxxx/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8/build-aux/missing aclocal-1.16 -I m4 -I src/gl/m4 -I lib/unistring/m4 --install
m4/gnulib-comp.m4:427: warning: macro 'AM_STDBOOL_H' not found in library
src/gl/m4/gnulib-comp.m4:802: warning: macro 'AM_STDBOOL_H' not found in library
lib/unistring/m4/gnulib-comp.m4:166: warning: macro 'AM_STDBOOL_H' not found in library
missing file gl/stdio-read.c
[... 718 lines cut ...]
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vim=y
CONFIG_PACKAGE_vxlan=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_OVERRIDE_PKGS="gperf libxml2"
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AcidSlide, 2023-05-05:
I forgot to mention I also did a `make package/feeds/packages/gnutls/{clean,compile} V=sc` and same error was produced.

brada4, 2023-05-05:
It pertains ../packages repo.

AcidSlide, 2023-05-06:
Update on doing `git bisect`

Current testing commit: https://github.com/openwrt/openwrt/commit/2070a2ca27bdb2b1e4e1587274e192e42f247516

Error now with `gettext-full`: 
```
gcc -DHAVE_CONFIG_H -I. -I..  -I. -I. -I.. -I.. -Iglib -DIN_LIBTEXTSTYLE -DLIBXML_STATIC    -I./libcroco  -DDEPENDS_ON_LIBICONV=1 -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/host/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/host/include  -O2 -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/host/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/host/include -DPIC -fPIC -c getprogname.c
getprogname.c: In function 'getprogname':
getprogname.c:291:4: error: #error "getprogname module not ported to this OS"
  291 | #  error "getprogname module not ported to this OS"
      |    ^~~~~
make[8]: *** [Makefile:5125: config.h] Error 1
make[8]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/libtextstyle/lib'
make[7]: *** [Makefile:1827: all-recursive] Error 1
make[7]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/libtextstyle'
make[6]: *** [Makefile:1755: all] Error 2
make[6]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/libtextstyle'
make[5]: *** [Makefile:407: all-recursive] Error 1
make[5]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1'
make[4]: *** [Makefile:363: all] Error 2
make[4]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1'
make[3]: *** [Makefile:146: /home/xxxxxxxx/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/.built] Error 2
make[3]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#1.66#1.24#2.72
    ERROR: package/libs/gettext-full [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/gettext-full/host/compile] Error 1
make[2]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make[1]: *** [package/Makefile:110: /home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make: *** [/home/xxxxxxxx/openwrt/openwrt/include/toplevel.mk:231: world] Error 2
```

AcidSlide, 2023-05-06:
Update on doing `git bisect`

Current testing commit: https://github.com/openwrt/openwrt/commit/fe33d5ba13370027165b9acf2bd895f5b4d9a4ea6

**No build errors for both x86 and ramips**

Note: Did a full clean before doing a build, deleted `tmp`, `bin`, `build_dir`, and `staging_dir` to make sure previous build with error won't affect the test build

AcidSlide, 2023-05-06:
Update #3: Doing manual `git reset --hard` going up the commit tree

Current test build worked up to commit: https://github.com/openwrt/openwrt/commit/4437434b8a4c8c4f9000caeb773c1aa248a6d29a

ecrasy, 2023-05-06:
same error here with my build for x86

AcidSlide, 2023-05-06:
@antnyfls I'll try the PR in my next test build.  But I would also like to input an observation.  For better explanation, see the following steps I did.

1. Moved the commit head to a known working build (well to a commit a I know the build went through)
2. Do a full clean by deleting `tmp`, `bin`, `build_dir`, and `staging_dir`
3. Do a `./scripts/feeds install -a` just to reset `tmp` folder
4. Do a build `make -j9 defconfig download world`
5. Move the commit head to a known non-working build
6. Do a build `make -j9 download world` - now this will result a good build
7. Do another full clean (repeat Step 2 and 3)
8. Do a build `make -j9 defconfig download world` - now this results with the build error to gnutls

antnyfls, 2023-05-06:
@AcidSlide Try with the latest commit and apply the patch in feeds/packages
```
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
cd feeds/packages
wget https://patch-diff.githubusercontent.com/raw/openwrt/packages/pull/20940.patch
git apply 20940.patch
cd -
make -j
```

AcidSlide, 2023-05-06:
Thanks @antnyfls I'll do that later.. my current test build is still running

AcidSlide, 2023-05-06:
Once the issue is confirmed fixed by the PR in the openwrt/packages for gnutls v3.8.0, I'll close this issue.

AcidSlide, 2023-05-07:
Closing issue as the PR for the gnutls updated to v3.8.0 has fixed this issue.. so this was related to the updated gnu related packages in the main openwrt/openwrt

closed 2023-05-07 by AcidSlide