openwrt/openwrt #12554: ASUS RT-N11P boot time increase from firmware to firmware.

StaDanL, 2023-05-07:
### Describe the bug

Hi.

I`m using router Asus RT-N11P many years with openwrt. Starting from the 21.02 firmware, I noticed that the boot time has increased from 30s to 45s. After upgrading to the version 22.03 boot time increase even more, to 1 minute. This is definitely not ok for me. Is it possible to return 30s- somehow???

PS:Changing bootloader to "breed" help?

### OpenWrt version

22.03

### OpenWrt target/subtarget

22.03

### Device

ASUS RT-N11P

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-07:
It is a very low memory device. Downgrade to v21 will let it boot quickly as long as that version is supported.

richterfalko, 2023-05-07:
> It is a very low memory device. Downgrade to v21 will let it boot quickly as long as that version is supported.

OpenWrt 21.02 just got the last update. 21.02 from now on is unmaintained and end of life: https://openwrt.org/releases/21.02/notes-21.02.7

> The OpenWrt 21.02 stable series is now end of life following the [OpenWrt Security support guidelines](https://openwrt.org/docs/guide-developer/security#support_status). We encourage all users of the OpenWrt 21.02 stable series to upgrade to [OpenWrt 22.03](https://openwrt.org/releases/22.03/start). **We will not fix any security problems, even severe ones in the OpenWrt 21.02 release branch any more.**

I would not advise to downgrade to 21.02. A good advice would be migrate to newer supported devices with sufficient flash space and RAM, for example 128+ MB flash and 512+ MB RAM.

brada4, 2023-05-07:
It is just a PoC that device enters last stretch of its useful life. Thats up to user to accomodate discomfort with old gear or accomodate expense with new one.

Djfe, 2023-05-11:
Larger operating system: reading takes longer and decompression of the kernel that is required on boot also takes longer.
There's nothing that can be done about it because your hw has a fixed read speed. Also faster decompression algorithms cause (at least) slightly larger images, which in turn won't fit on your 8MB of flash.
Relevant: https://openwrt.org/supported_devices/864_warning

I got to say, I'm surprised images for your device are still being built, since it has only 32MB of RAM. I expected the minimum to already be 64MB RAM device, since support for devices with 4MB of flash and 32MB of RAM was dropped with openwrt-22.03 already. Yes, your device was "only" released in 2015/16, but the specs were very low-end even back then. We can't support these devices forever.

Here's a list of modern OpenWrt devices to look at (it also includes very expensive ones that probably aren't up your alley)
https://openwrt.org/toh/views/toh_available_16128_ax-wifi
The cheapest one on the list: a D-Link DAP-x1860. It costs only 18€ in Germany currently.
If you need more than one lan port (sorted by price kind of; it all depends on where you live and availability):
Cudy X6
Mercusys MR70X (missing on the list, but supported)
TP-Link AX23
Asus RT-AX53U / RT-AX54 (has one more lan port than the AX53U)
Netgear WAX202

Cheap when bought used:
ZyXEL Multy M1 (WSM20)

If you want good/longer lasting hardware that costs a bit more:
Netgear WAX206
Linksys E8450 / Belkin RT3200

Great bang for your buck, when it's available in your area: Dynalink DL-WRX36 AX3600

namiltd, 2023-06-11:
A long system load is acceptable if the system is stable. Disabling support for devices with 32 MB RAM just because of slow loading is not a reason. Users are restricted from accessing new versions of the system and must build them themselves, instead of just downloading them from openwrt.org

rany2, 2023-06-11:
There is no way it could be stable on 32MB of RAM. It will just OOM constantly. I had a 64MB device and that was OOMing nonstop with latest OpenWRT (but not so much with 21.02) (and it didn't have any kernel modules besides the WiFi driver and for packages it was the bare minimum for it to function). It is not worth the headache for you.

Djfe, 2023-06-11:
I didn't do the PR due to long system load 😅
Mostly because normal users like @StaDanL would expect openwrt to run stable, if we provide an official build for it.
Since it's not possible to guarantee stability anymore with just 32mb of RAM, devices with 32mb of RAM (or less) were deprecated by maintainers with the release of openwrt-21.02
https://openwrt.org/releases/21.02/notes-21.02.0#increased_minimum_hardware_requirements8_mb_flash_64_mb_ram

This github issue made me notice that there were over 100 devices left that have 32mb of RAM or less that were still being build despite this decision. People probably forgot to disable the build of these devices, so I did a PR to make up for this.

If people want current openwrt releases, then they need to run/buy hardware that we are able to support or they need to build and test releases themselves. 32MB was dropped intentionally and for reasons that we linked above.
Also: OpenWrt.22-03 will receive updates for another year.

openwrt doesn't drop support lightly. This change has been discussed back then by maintainers. My PR was only merged this fast because discussions were held and it was based on previous decisions.

Personally I wanted to prevent people with less technical background from bricking their devices by accident. Lots of devices allow first installation of openwrt without serial access after all but recovery often requires it.

brada4, 2023-06-11:
Once could include zram-swap and it will not oom right away.

namiltd, 2023-06-11:
For devices with a USB port, you can create swap memory on a USB stick to avoid OOM.

Djfe, 2023-06-12:
Could both work/help. I never tested it.
usb swap won't work for the device of @StaDanL since it doesn't have a USB port.

You could open a PR and add swap to specific devices you want to keep being build (and maybe increase compat version while doing so to inform users that their device requires a USB stick to work from now on)
The biggest issue I see here is that usb flash is often slow, can get unstable or fail (bad blocks, usb sticks might be possible with the right filesystem to handle these). especially if people choose to use cheap sticks. Depending on what you do, you might wear down the nand faster than you might imagine (swap might cause way more writes than a simple extroot)

In the end these are crutches and not a real solution, but that's just my opinion and shouldn't matter. It's fine if people want to keep their hw alive for longer. :)
But if you want to activate some of these devices again, then you need to have a solution if you want to convince the maintainers.

brada4, 2023-06-12:
@Djfe does not work on technical merit #12887 , irrelevant issue specific to one other device is used to ampify evils of 32MB. Next week same reasoning can be used to dump 64MB devices? Hope not.

Djfe, 2023-06-12:
well I suggested opening a Pull Request, not a bug report 😅
Also it's probably more like a feature request (adding support for (some) low mem devices again)
In the end I never promised maintainers wouldn't just outright close your issue without much discussion.
Maybe github wasn't the right place to discuss it after all, sorry.

I agree that low mem can be a support nightmare where you are looking for issues but find none.
I bet when he analyzed the issue with the Linksys E2000 he expected all 32M ram devices to be disabled already due to the announcement back then.

low ram is "the harder limitation" compared to flash, even if some targets have a smaller base footprint.
https://openwrt.org/supported_devices/432_warning#supportability_issues

for example: low ram can lead to issues while loading compressed data from flash. if you do your own build you could use a different compression with a smaller memory footprint on decompression. you could also do zram swap

Devs don't own all of these devices so supporting them is hard. Feel free to provide and upload your own builds somewhere, it's open source after all 😊

> irrelevant issue specific to one other device is used to ampify evils of 32MB

Well it was more like a recent example how supporting 32m can be a nightmare.
Even the devs weren't all on the same boat at first and some wanted to keep supporting all devices according to the wiki). it's a difficult topic.
Disabling official builds is a compromise. Less time spent on support requests, less user frustration (users with less technical background)

> Next week same reasoning can be used to dump 64MB devices? Hope not.

You are drawing the devil onto a wall here (German saying)
64mb should be plenty for basic operation for a while I bet. It's good that we have an official warning to avoid buying such devices new (unless you can live with the consequences of being restricted in how much packages the device can run at the same time)

Yes, he was against rediscussing what has been established already as the minimum requirements for official openwrt builds. so he just closed your issue without further debate (nothing I can do about that)