openwrt/openwrt #12558: [TP-Link MR3420 v5] Wi-Fi pretty instable since OpenWRT installation

malekairmaroc7, 2023-05-07:
### Describe the bug

Since I installed OpenWRT from the official firmware via TFTPD I noticed that Wi-Fi is pretty instable. My devices connect to it but after some time, it disappears and cannot be found in the WiFi list. I need to connect to the router via LAN and restart the WiFi chip.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-MR3420 v5

### Image kind

Official downloaded image

### Steps to reproduce

1. Turn on router
2. Wait for it to boot and connect to WiFi
3. After some time, it disconnects by itself
4. Reset WiFi device in the panel by connecting to LAN
5. Starts over again

### Actual behaviour

Wi-Fi gets disconnected and must be resetted via LAN by accessing LuCi (not the whole router; only the chip).

### Expected behaviour

Stable Wi-Fi connection without these interruptions

### Additional info

I am using the 2.4 GHz band only.

### Diffconfig

```text
-
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

richterfalko, 2023-05-07:
tl-mr3420 only has 64 MB RAM: [https://openwrt.org/toh/tp-link/tl-mr3420_v5](url)

Please read: [https://openwrt.org/supported_devices/864_warning](url)
> Insufficient RAM for stable operation
> 32 MB RAM is already deprecated. You will run into issues with an up to date OpenWrt version.
> **64 MB RAM may have some issues with stability**, depending on your hardware and use cases, although it is enough for basic usage
> 128 MB RAM or more is recommended if software past basic router/AP functionality is to be used

Current popular OpenWrt devices like Linksys E8450/Belkin RT3200, Netgear XR500 or Dynalink DL-WRX36 are equipped with 512 MB RAM or more.

brada4, 2023-05-07:
What does `logread` say around disconnects? 64MB should be plenty for single or even dozen wifi devices.
Also shall be no additional problem (sys-)upgrading to 22.03.5 for some security improvements.

malekairmaroc7, 2023-05-08:
Thank you for replying!
As you said, 64 MB is pretty much enough for basic operation. I wouldn't spend that much money for unnecessary tweaks and feature packed routers.
I have another router (TP Link Archer C60) and it performs very reliable!
I will look into `logread` as soon as it happens again. Didn't think of it actually.

brada4, 2023-05-08:
Go to LuCi -> System -> System -> Logging -> Log buffer size and set it bigger, 1024k will not hurt, you have memory (nod to deleted post ;-) )

malekairmaroc7, 2023-05-11:
It just happened again. Here is a cutout from the `logread` as a `txt` file.
[logread.txt](https://github.com/openwrt/openwrt/files/11452359/logread.txt)

Maybe you can discover something useful.

DragonBluep, 2023-05-11:
https://github.com/openwrt/openwrt/issues/10074 A famous problem about MT7628

malekairmaroc7, 2023-05-11:
Thank you for the information! Hopefully it gets fixed soon.

brada4, 2023-05-12:
You can try snapshot (mind to backup config before and install `luci-mod-admin-full` via ssh after) Mediatek firmwares are updated every few weeks (the linked issue refers to firmware)
Obviously revert if it fixes nothing (sysupgrade to release with config wipe + restore the config archive or re-enter all config)

uncharted-user, 2023-07-26:
I can also see similar issue on Netgear R6120 with mt76x8. My Wi-Fi connections (5GHz and 2.4GHz) drop even due to playing a YouTube video at 720p on a single device. I have to manually disconnect and reconnect it. I never saw that issue with factory firmware. Is there any solution yet?

DragonBluep, 2023-09-17:
This issue has been fixed some days ago.

uncharted-user, 2025-02-16:
> > [#10074](https://github.com/openwrt/openwrt/issues/10074) A famous problem about MT7628
> 
> This issue has been fixed some days ago. ref: [76b1e56](https://github.com/openwrt/openwrt/commit/76b1e564d202c09d0035315eb6e958a9b0dd4491)

The issue is re introduced from 23.05.5 and also present in 24.10