openwrt/openwrt #12565: buildsystem: tools/libtool does not compile on my target

feckert, 2023-05-08:
### Describe the bug

Since the following changes in tools/libtools Makefile, the package https://github.com/openwrt/openwrt/commit/7c0727c26330d12253ec98463279a564ae0f9efa and https://github.com/openwrt/openwrt/commit/87e902fcdcce0e3d1287deb901f934aabf7ec183 does not compile anymore. I get the following error message:
```
Applying ./patches/000-relocatable.patch using plaintext: 
patching file libtoolize.in
patching file m4/libtool.m4
Applying ./patches/100-libdir-fixes.patch using plaintext: 
patching file build-aux/ltmain.in
Applying ./patches/110-dont-use-target-dir-for-relinking.patch using plaintext: 
patching file build-aux/ltmain.in
Applying ./patches/120-strip-unsafe-dirs-for-relinking.patch using plaintext: 
patching file build-aux/ltmain.in
Applying ./patches/130-trailingslash.patch using plaintext: 
patching file build-aux/ltmain.in
Applying ./patches/140-don-t-quote-SHELL-in-Makefile.am.patch using plaintext: 
patching file Makefile.am
Applying ./patches/200-openwrt-branding.patch using plaintext: 
patching file build-aux/ltmain.in
patching file build-aux/funclib.sh
*** Calling Host/Compile/Default (libtool) ***
make[5]: Entering directory '/home/bbworker/bbworker/owrt_master_lantiq_xrx200/System6/build/openwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[5]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[5]: Leaving directory '/home/bbworker/bbworker/owrt_master_lantiq_xrx200/System6/build/openwrt/build_dir/host/libtool-2.4.7'
make[4]: *** [Makefile:68: /home/bbworker/bbworker/owrt_master_lantiq_xrx200/System6/build/openwrt/build_dir/host/libtool-2.4.7/.prepared3f554b1eff18036f2b996c39c9d267eb_6664517399ebbbc92a37c5bb081b5c53] Error 2
time: tools/libtool/compile#0.10#0.02#0.12
```

### OpenWrt version

master

### OpenWrt target/subtarget

lantiq/xrx200

### Device

---

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mpratt14, 2023-05-08:
the second commit should be unrelated

the call for Host/Compile/Default in Host/Uninstall has the minus prefix `-` so that an error is ignored

we should at least see the command to run Make in the output before it fails, can you put a more verbose log?

mpratt14, 2023-05-08:
example on my system, after running `make tools/libtool/clean` once or twice, if the directory is there

`Error 2 (ignored)` results in good exit code 0

```
 $ make tools/libtool/clean V=sc
make[1]: Entering directory '/path/to/openwrt'
make[2]: Entering directory '/path/to/openwrt/tools/libtool'
CFLAGS="-O2 -I/path/to/openwrt/staging_dir/host/include " CPPFLAGS="-I/path/to/openwrt/staging_dir/host/include " CXXFLAGS="" LDFLAGS="-L/path/to/openwrt/staging_dir/host/lib " make  -C /path/to/openwrt/build_dir/host/libtool-2.4.7  uninstall
make[3]: *** /path/to/openwrt/build_dir/host/libtool-2.4.7: No such file or directory.  Stop.
make[2]: [Makefile:59: host-clean-build] Error 2 (ignored)
(cd /path/to/openwrt/staging_dir/host/share/aclocal/ && rm -f libtool.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)
rm -rf /path/to/openwrt/build_dir/host/libtool-2.4.7 /path/to/openwrt/build_dir/host/libtool-2.4.7/.built
rm -rf /path/to/openwrt/staging_dir/host/stamp/.libtool_installed
make[2]: Leaving directory '/path/to/openwrt/tools/libtool'
time: tools/libtool/clean#0.08#0.10#0.25
make[1]: Leaving directory '/path/to/openwrt'
```

feckert, 2023-05-09:
I don't think the host-clean-build is the problem there.
The compilation fails due to the https://github.com/openwrt/openwrt/commit/7c0727c26330d12253ec98463279a564ae0f9efa
The error message is:
```
patching file build-aux/funclib.sh
*** Calling Host/Compile/Default (libtool) ***
make[5]: Entering directory '/home/bbworker/bbworker/owrt_master_lantiq_xrx200/System6/build/openwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[5]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
```

mpratt14, 2023-05-10:
I understand that the host-clean-build target is not directly involved, but both the prepare stage and host-clean-build calls Host/Uninstall. That is the only time that Make is invoked in the prepare stage, and from your first post I see that it's failing in the prepare stage...

my point is, you should be able to run `make tools/libtool/clean` as much as you want without a bad exit code. if not, I would start to question the host machine.

mpratt14, 2023-05-10:
a better example, if I do

`make tools/libtool/clean tools/libtool/prepare V=sc`

when it gets to the prepare target I have this at some point

```
Applying ./patches/200-openwrt-branding.patch using plaintext:
patching file build-aux/ltmain.in
patching file build-aux/funclib.sh
CFLAGS="-O2 -I/path/to/openwrt/staging_dir/host/include " CPPFLAGS="-I/path/to/openwrt/staging_dir/host/include " CXXFLAGS="" LDFLAGS="-L/path/to/openwrt/staging_dir/host/lib " make  -C /path/to/openwrt/build_dir/host/libtool-2.4.7  uninstall
make[3]: Entering directory '/path/to/openwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[3]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[3]: Leaving directory '/path/to/openwrt/build_dir/host/libtool-2.4.7'
make[2]: [Makefile:67: /path/to/openwrt/build_dir/host/libtool-2.4.7/.prepared436a0e79b037bdd4a9553257dc922ca8_6664517399ebbbc92a37c5bb081b5c53] Error 2 (ignored)
```

(note the "ignored" at the end)

and then it continues...

```
(cd /path/to/openwrt/staging_dir/host/share/aclocal/ && rm -f libtool.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)
( cd /path/to/openwrt/build_dir/host/libtool-2.4.7; AUTOM4TE=/path/to/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/path/to/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/path/to/openwrt/staging_dir/host/bin/automake ACLOCAL=/path/to/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/path/to/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/path/to/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/path/to/openwrt/staging_dir/host/bin/libtool M4=/path/to/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true ./bootstrap --force --skip-git --skip-po --gnulib-srcdir=/path/to/openwrt/staging_dir/host/share/gnulib )
...
...
...

```

It's worth mentioning that many tools now have a Host/Uninstall where the line that invokes Make has the `-` prefix to ignore errors. The only difference here is that it is also part of the prepare stage. In particular, `lz4` used to have a call to make uninstall *without* that prefix, which caused me build problems if I were to run `tools/clean` twice by accident. I was going to fix that by adding the `-` but then Rosen decided to convert it to a CMake build instead, which apparently saves build time as well as resolve that bad target recipe.

feckert, 2023-05-24:
@mpratt14  Thanks for your help, but the problem has been solved, it was due to a patch that I have / now had in my patch set.

cnkids, 2024-08-12:
> @mpratt14 Thanks for your help, but the problem has been solved, it was due to a patch that I have / now had in my patch set.

I got the same problem, can you give more details about how to solve the problem?