openwrt/openwrt #12566: ipv4 tethering not working with iPhone 14 pro max with nanoPI r1

stephensworld, 2023-05-08:
### Describe the bug

Hi,

I have tried to install an openwrt router with tethering feature following this tutorial: 
https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering

It works fine with iPad, iPhone SE, iPhone 12 Pro with newest iOS and iPadOS firmware 16.4.1(a)
The openwrt ethernet interface don't get an IPv4 from iPhone 14 pro max
With iPhone 14 pro max only IPv6 is working!

Thanks
Stephan

### OpenWrt version

r22765-a872570c04

### OpenWrt target/subtarget

sunxi/cortexa7

### Device

FriendlyARM NanoPi R1

### Image kind

Official downloaded image

### Steps to reproduce

opkg update
opkg install kmod-usb-net-rndis
opkg install kmod-usb-net-ipheth usbmuxd libimobiledevice usbutils
 
# Call usbmuxd
usbmuxd -v
 
# Add usbmuxd to autostart
sed -i -e "\$i usbmuxd" /etc/rc.local

Steps 3.b Web interface from tutorial
https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-08:
What do you see in `logread` regarding odhcp on that usb0? interface?

stephensworld, 2023-05-08:
Mon May  8 18:57:13 2023 kern.info kernel: [32406.801081] usb 2-1: new high-speed USB device number 4 using ehci-platform
Mon May  8 18:57:14 2023 kern.info kernel: [32407.004035] usb 2-1: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=15.03
Mon May  8 18:57:14 2023 kern.info kernel: [32407.012248] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mon May  8 18:57:14 2023 kern.info kernel: [32407.019386] usb 2-1: Product: iPhone
Mon May  8 18:57:14 2023 kern.info kernel: [32407.022983] usb 2-1: Manufacturer: Apple Inc.
Mon May  8 18:57:14 2023 kern.info kernel: [32407.027344] usb 2-1: SerialNumber: 00008120000060220FFB401E
Mon May  8 18:57:14 2023 kern.info kernel: [32407.039680] ipheth 2-1:4.2: Apple iPhone USB Ethernet device attached
Mon May  8 18:57:14 2023 daemon.notice netifd: Interface 'tethering' is enabled
Mon May  8 18:57:14 2023 daemon.err usbmuxd[2266]: [18:57:14.153][3] Connecting to new device on location 0x20004 as ID 3
Mon May  8 18:57:14 2023 daemon.err usbmuxd[2266]: [18:57:14.153][3] Connected to v2.0 device 3 on location 0x20004 with serial number 00008120-000059220EEB401E
Mon May  8 18:57:20 2023 daemon.notice netifd: Network device 'eth2' link is up
Mon May  8 18:57:20 2023 daemon.notice netifd: Interface 'tethering' has link connectivity
Mon May  8 18:57:20 2023 daemon.notice netifd: Interface 'tethering' is setting up now
Mon May  8 18:57:20 2023 daemon.notice netifd: tethering (3920): udhcpc: started, v1.36.0
Mon May  8 18:57:20 2023 daemon.notice netifd: tethering (3920): udhcpc: broadcasting discover
Mon May  8 18:57:23 2023 daemon.notice netifd: tethering (3920): udhcpc: broadcasting discover
Mon May  8 18:57:26 2023 daemon.notice netifd: tethering (3920): udhcpc: broadcasting discover
root@OpenWrt:~#

stephensworld, 2023-05-08:
and with iPhone SE and same iOS version I get this:
Mon May  8 18:59:39 2023 kern.info kernel: [32552.451071] usb 2-1: new high-speed USB device number 6 using ehci-platform
Mon May  8 18:59:39 2023 kern.info kernel: [32552.654244] usb 2-1: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=10.04
Mon May  8 18:59:39 2023 kern.info kernel: [32552.662475] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mon May  8 18:59:39 2023 kern.info kernel: [32552.669615] usb 2-1: Product: iPhone
Mon May  8 18:59:39 2023 kern.info kernel: [32552.673218] usb 2-1: Manufacturer: Apple Inc.
Mon May  8 18:59:39 2023 kern.info kernel: [32552.677578] usb 2-1: SerialNumber: 64ffff85e8a3bfae12769ceeeec9b31d123d4da0
Mon May  8 18:59:39 2023 daemon.notice netifd: Interface 'tethering' is enabled
Mon May  8 18:59:39 2023 daemon.err usbmuxd[2266]: [18:59:39.819][3] Connecting to new device on location 0x20006 as ID 5
Mon May  8 18:59:39 2023 daemon.err usbmuxd[2266]: [18:59:39.820][3] Connected to v2.0 device 5 on location 0x20006 with serial number 646d4e85e8a3bfae12769c928bc9b31d321d4da0
Mon May  8 18:59:39 2023 kern.info kernel: [32552.706630] ipheth 2-1:4.2: Apple iPhone USB Ethernet device attached
Mon May  8 18:59:40 2023 daemon.err usbmuxd[2266]: [18:59:40.105][1] ERROR: client_write: sending to fd 11 failed: Broken pipe
Mon May  8 18:59:40 2023 daemon.err usbmuxd[2266]: [18:59:40.105][1] ERROR: client_write: sending to fd 11 failed: Broken pipe
Mon May  8 18:59:40 2023 daemon.err usbmuxd[2266]: [18:59:40.105][1] connection_teardown: aborting buffer flush to client after error.
Mon May  8 18:59:41 2023 daemon.notice netifd: Network device 'eth2' link is up
Mon May  8 18:59:41 2023 daemon.notice netifd: Interface 'tethering' has link connectivity
Mon May  8 18:59:41 2023 daemon.notice netifd: Interface 'tethering' is setting up now
Mon May  8 18:59:41 2023 daemon.notice netifd: tethering (4369): udhcpc: started, v1.36.0
Mon May  8 18:59:41 2023 daemon.notice netifd: tethering (4369): udhcpc: broadcasting discover
Mon May  8 18:59:41 2023 daemon.notice netifd: tethering (4369): udhcpc: broadcasting select for 172.20.10.7, server 172.20.10.1
Mon May  8 18:59:41 2023 daemon.notice netifd: tethering (4369): udhcpc: lease of 172.20.10.7 obtained from 172.20.10.1, lease time 86400
Mon May  8 18:59:41 2023 daemon.notice netifd: Interface 'tethering' is now up
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using nameserver 172.20.10.1#53
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon May  8 18:59:41 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon May  8 18:59:41 2023 user.notice firewall: Reloading firewall due to ifup of tethering (eth2)
root@OpenWrt:~#

brada4, 2023-05-08:
One google later:
https://apple.stackexchange.com/questions/426789/iphone-hotspot-not-assigning-ipv4-dhcp-address

httpstorm, 2023-05-09:
@stephensworld 
Can you try building firmware with this commit? TLDR if you connect you iPhone for tethering to a Mac and then to OpenWRT, without this patch it won't work until you restart the phone.
https://github.com/httpstorm/openwrt/commit/b5655d5631e20ea9d53a8f0b4a94c6a796eb882e

thescouser89, 2024-04-14:
Hi!

I experienced the same issue with a TP-Link WR902AC v4 with an iphone 13. I was able to workaround it by configuring the ethx interface:
- use static instead of dhcp
- ip: 172.20.10.5 (pick one at random from 172.20.10.2 to 172.20.20.14!)
- netmask: 255.255.255.240
- gateway: 172.20.10.1
- In the advanced settings, use as DNS server your favorite one (1.1.1.1, or 8.8.8.8)

After that I was able to get internet connection flowing.

httpstorm, 2024-04-14:
@thescouser89 
Use this patch, it's been upstream for a long time, I'm not sure when it is coming to a release. Linux kernel 6.1.82 still does not include it. The commit description will tell you why you are not getting any communication. TLDR usbnet: ipheth: add CDC NCM support.
https://github.com/httpstorm/openwrt/commit/332f69143047d208428f6067de29e7c1a76ece81

thescouser89, 2024-04-14:
@httpstorm Thanks! Unfortunately it looks like it's not yet in the latest snapshot but will look into it! Thanks for pointing it out! :)

neheb, 2024-04-19:
@httpstorm those patches belong in backport, not pending. Might as well create a PR.

neheb, 2024-04-19:
btw since it seems people tether their iphones with OpenWrt, instead of doing rc.local manipulations and such, does the following hotplug script work?

/etc/hotplug.d/usb/20-usbmuxd
```
# Ported from original udev script

# Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
[ "$ACTION" = "add" ] &&
[ echo "$PRODUCT" | grep -Eq "5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*" ] &&
echo 0 > /sys/$DEVPATH/bConfigurationValue
/etc/init.d/usbmuxd start

# Exit usbmuxd when the last device is removed
[ "$ACTION" = "remove" ] &&
[ echo "$PRODUCT" | grep -Eq "5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*" ] &&
/usr/sbin/usbmuxd -x
```

neheb, 2024-04-19:
@httpstorm I know for a fact that usbmuxd must be running in order for this to work. Hence my question about the initial hotplug script.

That bind thing sounds like it should be implemented for usbmuxd.

httpstorm, 2024-04-20:
@neheb 
> @httpstorm I know for a fact that usbmuxd must be running in order for this to work. Hence my question about the initial hotplug script.

The `/etc/init.d/usbmuxd` service starts on boot, runs all the time and handles iPhone connect and disconnect events.

No `hotplug.d` script:
- [x] `usbmuxd` started on boot by `/etc/init.d/usbmuxd`
- [x] tethering works after USB connect or router reboot
- [ ] tethering does not work after iPhone restart while the USB cable remains connected

Your script: makes no difference

My script:
- [x] `usbmuxd` started on boot by `/etc/init.d/usbmuxd`
- [x] tethering works after USB connect or router reboot
- [x] tethering works after iPhone restart while the USB cable remains connected, even before entering passcode

> That bind thing sounds like it should be implemented for usbmuxd.

How do we get in touch with the `libimobiledevice` team? I have not received any replays to my support ticket yet.
https://github.com/libimobiledevice/usbmuxd/issues/218

I see there have been no releases since `1.1.1` on 2020-06-15. We can try switching to master.

neheb, 2024-04-20:
@httpstorm I meant a PR in the packages repo. Upstream is rather...slow moving. Especially when it comes to releases.

Is usbmuxd set to start on boot by default? Upstream's udev script and the hotplug.d port is for situations where it doesn't run but an iPhone is plugged in. Maybe both should be combined.

httpstorm, 2024-04-21:
@neheb 
> I meant a PR in the packages repo. Upstream is rather...slow moving. Especially when it comes to releases.

Here is a PR with my script: https://github.com/openwrt/packages/pull/23976
But if you mean fixing the issue in `usbmuxd` itself, that would require knowledge how it works and communicates with the iPhone. it might also be worth updating `usbmuxd` to master, since there are no releases in the past 4 years. Should we do this?

> Is usbmuxd set to start on boot by default? Upstream's udev script and the hotplug.d port is for situations where it doesn't run but an iPhone is plugged in. Maybe both should be combined.

I remember when I first tried tethering, the procedure was quite complicated. But that was many years ago. Thankfully things have improved, so now it's plug and play. Yes, `usbmuxd` is started on boot by the `init.d` script. So your script is no longer needed. iPhone trust is handled automatically, the tickets are stored in `/etc/lockdown`, in case you want to build an image that is already trusted by your iPhone.

> edit: I should note I don't tether my iPhone but have worked on the packages feed to get these tools working for a completely different reason. Some of the information on the wiki is outdated, especially the part where usbmuxd uses /var (patched to use /etc).

Wiki for iPhone: I use the same packages.
``` bash
opkg update
opkg install kmod-usb-net-ipheth usbmuxd libimobiledevice usbutils
 ```

These are no longer needed, the ethernet interface should be ready to use, as soon as the user confirms they trust the router.
``` bash
# Call usbmuxd
usbmuxd -v
 
# Add usbmuxd to autostart
sed -i -e "\$i usbmuxd" /etc/rc.local
```

> edi2: Out of curiosity, is tethering limited to iPhones as they have cellular radios or does it work with other iOS devices? I remember if I wanted to tether a wifi network and access it over USB I needed to jailbreak my iPhone.

I think both iPhone and iPad should work. You can query information and send some commands. If a cellular radio is available, you can tether it and get an Ethernet interface. This can also be used with Xcode to debug apps on the iPhone even though it's behind a router (once you set the IP manually). Indeed I still have that app on my ancient jailbroken iPhone 3G, I think it was called MyWi. I tested it last year and it is still able to tether to the router and provide Internet. Yes, it can expose Wi-Fi over the USB interface and do other fancy stuff. From the times when tethering with OpenWRT required manual commands. iPhone 11 is doing a bit better: 268 Mbit/s laptop-wifi-router-usb-iPhone-LTE.

commit 680f8738 (2024-04-21): kernel: backport ipeth CDC NCM support

Fixes no communication with tethered iOS devices in CDC NCM mode.
Freshly booted iOS devices start in legacy mode, but are put into
NCM mode by the official Apple driver.

[1] https://github.com/torvalds/linux/commit/a2d274c62e44b1995c170595db3865c6fe701226

Fixes: #12566
Tested-by: Georgi Valkov <gvalkov@gmail.com>

[ better reference fixed issue ]
commit 2b6b6e2f (2024-04-21): kernel: backport ipeth CDC NCM support

Fixes no communication with tethered iOS devices in CDC NCM mode.
Freshly booted iOS devices start in legacy mode, but are put into
NCM mode by the official Apple driver.

[1] https://github.com/torvalds/linux/commit/a2d274c62e44b1995c170595db3865c6fe701226

Fixes: #12566
Tested-by: Georgi Valkov <gvalkov@gmail.com>

[ better reference fixed issue ]

(cherry picked from commit 680f8738d02a1876ae4cd11aacf9cd56e520fadf)
closed 2024-04-21 by openwrt-bot