openwrt/openwrt #12569: MR8300 - 5Ghz Wifi networks not working after latest snapshot update

infantvin, 2023-05-09:
### Describe the bug

Post installing the latest update of the snapshot firmware for the Linksys MR8300, I find that the 5Ghz network on Wifi is not showing up in the Wifi networks advertised. One of them seems to show as enabled in the UI Wireless page but the other is disabled. I am attaching a screenshot for reference as well.

As soon as I revert to the previous installed version from about a week or two ago, the issue goes away and all Wifi band networks are visible to connect and use.

### OpenWrt version

OpenWrt SNAPSHOT  r22778-e882af2850

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Flash the newest snapshot image to the router. Check Wireless networks available and check on the Wifi adapter for visible networks.

The 2.5G will show up, but the 5Ghz band wifi networks dont show.

### Actual behaviour

5Ghz bands are not being advertised or visible. This is not expected behavior

### Expected behaviour

All enabled Wifi bands must be visible to connect to.

### Additional info

I have installed wifitoggle. However, this problem seems to be visible even without that being present.

If you look at the attached screenshot, you will see the 5Ghz band is not showing in the list of available wifi networks on windows. Also, one of the 5Ghz bands just does not activate even when enabled.

This is an unexpected behavior which is not seen on reverting to the previous firmware.

Please let me know if you need any more information. I tried builds from both May 8th and May 9th 2023 and its the same behavior on both. Something seems to have slipped into the downstream builds causing this.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

infantvin, 2023-05-09:
[image]

infantvin, 2023-05-09:
Hi,

It has been more than 3 hours since the router is up with this firmware and it has the same problem. Nothing has changed since the previous firmware was loaded. 

I checked the wireless page and its the same way as the screenshot uploaded shows. So I think this is some new issue associated with this firmware. Normally about 2-3 minutes brings up all the interfaces without problems.
Something has gone wrong and I need help to figure out what it is if its from my side. 

The reason I think this is a bug is because I can revert to the previous firmware and everything comes back normally. All interfaces are up and running and no issues.

This configuration has worked for almost a year now without any problem.

brada4, 2023-05-09:
UI shows radio is disabled. just tap few times on enable/disable button.
If that does not help reset configuration and redo it.
If still no luck - install back release version. Nothing changed there during last week.

robimarko, 2023-05-10:
What does `logread` via SSH show?

infantvin, 2023-05-10:
> What does `logread` via SSH show?

I am uploading the file. However, this is something I am seeing which I feel may be something wrong.

It must be noted that my wireless devices are unable to see the 5Ghz band even though one shows as enabled. I can understand if this happens on only 1 device. However if 3 phones and 2 laptops show the same problem, then something seems to have broken.

I am open to correct my configuration if the issue boils down to that being the cause too.

logread snippet below

Wed May 10 00:29:09 2023 daemon.notice hostapd: phy2-ap0: interface state HT_SCAN->DFS
Wed May 10 00:29:09 2023 daemon.notice hostapd: phy2-ap0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed May 10 00:29:09 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Wed May 10 00:29:09 2023 daemon.err hostapd: Interface initialization failed
Wed May 10 00:29:09 2023 daemon.notice hostapd: phy2-ap0: interface state DFS->DISABLED
Wed May 10 00:29:09 2023 daemon.notice hostapd: phy2-ap0: AP-DISABLED

Attaching the file as well.

[logreadmr83005gproblem.txt](https://github.com/openwrt/openwrt/files/11441443/logreadmr83005gproblem.txt)

robimarko, 2023-05-10:
Well, DFS failed hence why the AP newer gets to broadcasting.

Can you post your `/etc/config/wireless`, just remove the WPA key.

infantvin, 2023-05-10:
Thanks, here it is attached
[wirelessconfig.txt](https://github.com/openwrt/openwrt/files/11441481/wirelessconfig.txt)

robimarko, 2023-05-10:
I bet the issue was uncovered by:
https://github.com/openwrt/openwrt/commit/97d20525b24e96558f974858f4d8ad6d9148e61f

Cause, previously those channels did not have DFS requirements.
Can you try setting the channel to `auto` to see if any other DFS channel will work?

infantvin, 2023-05-10:
Hi,

I did what you said. I set the channel to auto for both 5Ghz bands and I am able to see the wifi on Qualcomm Atheros QCA9886 802.11ac/n and connect to it. The other 5Ghz band is still not visible even after setting it to auto.

So, as you say something has changed. Is this a valid change or is it something that needs to be corrected going forward?

infantvin, 2023-05-10:
@robimarko , I adjusted my channels. Left one of the 5G at auto and moved about the other to a one that works.

Thank you for helping me to find the cause and pointing to the root cause of this problems. I am happy to have found out what has caused this.  How do we dig into such commits and know if these can cause problems if they happen in the future? Its not easy unless someone is able to track the code changes made. I understand things to an administrative level on Linux, but am not efficient to know what a code update does.

robimarko, 2023-05-10:
There isn't really a magical way, but regulatory changes rarely happen.
You can see why it changed for India, original government documents are linked:
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/commit/?id=e78c450c08b73b6b291f0770313c97a85d4c865a

infantvin, 2023-05-10:
Thank you. I am closing this as I feel we have reached the solution to what was causing my problem.  I appreciate your help once again. Have good day.

closed 2023-05-10 by infantvin