openwrt/openwrt #12571: ath79/generic: TP-Link Archer C7 v5: Bandwidth with current release (22.03.5) is much lower than with old stable (21.02.7)

xosevp, 2023-05-09:
### Describe the bug

Symmetric 600 Mbps (~75 MB/s) GPON-FTTH.  Test done with 1Gb Ethernet:

- with 21.02.7, down/up: 77 / 77 MB/s
- with 22.03.5, down/up: **_30_** / **_70_** MB/s

### OpenWrt version

21.02.7 ( r16847-f8282da11e)  / 22.03.5 (r20134-5f15225c1e)

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-09:
Welcome to the club
https://github.com/openwrt/openwrt/issues/10790#issuecomment-1367501938

tjmaru, 2023-05-15:
I had a similar problem on my Archer C7 v5 with 22.03 but then I enabled `Software flow offloading` and it solved the problem

brada4, 2023-05-16:
Fixes speedtests for me but real life TCP connections freeze and eventually get dropped in 5min..1h
Absence of counters in nftable ruleset does not help, but typical packet is in (offloaded) state and both rules seem to enter early. Should be something examining every packet before that or so.

brada4, 2023-10-17:
kin principle hardware can forward packets at full speeds without offloads , but doing less copies tham currently via nft ruleset

brada4, 2023-10-17:
There is some backlog building up at fw4,  these i made to reduce per-packet work without semantic result change
V22+
https://patch-diff.githubusercontent.com/raw/openwrt/firewall4/pull/10.diff
https://patch-diff.githubusercontent.com/raw/openwrt/firewall4/pull/11.diff

V23+
https://patch-diff.githubusercontent.com/raw/openwrt/firewall4/pull/13.diff