openwrt/openwrt #12577: How to fix gettext-full undefined reference to `u8_uctomb_aux’

peterwillcn, 2023-05-10:
### Describe the bug

libtool: link: x86_64-openwrt-linux-musl-gcc -DINSTALLDIR="/usr/bin" -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1=gettext-0.21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -DPIC -fpic -znow -zrelro -o .libs/gettext gettext-gettext.o -L/home/openwrt/master/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/usr/lib -L/home/openwrt/master/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib ../gnulib-lib/libgrt.a ../intl/.libs/libintl.so
/home/openwrt/master/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../gnulib-lib/libgrt.a(libgrt_a-iconv.o): in function rpl_iconv': iconv.c:(.text+0x34b): undefined reference to u8_mbtoucr'
/home/openwrt/master/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: iconv.c:(.text+0x4ba): undefined reference to `u8_uctomb_aux'
collect2: error: ld returned 1 exit status
make[8]: *** [Makefile:1645: gettext] Error 1
make[8]: Leaving directory '/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/src'
make[7]: *** [Makefile:1643: all-recursive] Error 1
make[7]: Leaving directory '/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime'
make[6]: *** [Makefile:1548: all] Error 2
make[6]: Leaving directory '/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime'
make[5]: *** [Makefile:406: all-recursive] Error 1
make[5]: Leaving directory '/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1'
make[4]: *** [Makefile:362: all] Error 2
make[4]: Leaving directory '/home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1'
make[3]: *** [Makefile:189: /home/openwrt/master/openwrt/build_dir/target-x86_64_musl/gettext-0.21.1/.built] Error 2
make[3]: Leaving directory '/home/openwrt/master/openwrt/package/libs/gettext-full'
time: package/libs/gettext-full/compile#381.42#118.66#437.73
ERROR: package/libs/gettext-full failed to build.
make[2]: *** [package/Makefile:116: package/libs/gettext-full/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
data written to "hd.ids"
log written to "hd.log"

### OpenWrt version

master-218ce40cd7

### OpenWrt target/subtarget

x86_64

### Device

x86_64

### Image kind

Official downloaded image

### Steps to reproduce

1, ubuntu 22.04
2 time make V=s VERBOSE=1 -j$(nproc) JOBS=$(nproc) 2>&1 | tee build.log | grep -i -E "^make.*(error|[1-9]...Entering dir)"
3, branch master

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_AUTOREMOVE=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BIND_LIBJSON=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
# CONFIG_GCC_USE_VERSION_12 is not set
CONFIG_GCC_USE_VERSION_13=y
CONFIG_GCC_VERSION="13.1.0"
CONFIG_GCC_VERSION_13=y
# CONFIG_GDB is not set
CONFIG_GRUB_TIMEOUT="2"
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_HTOP_LMSENSORS=y
[... 544 lines cut ...]
CONFIG_SINGBOX_WITH_WIREGUARD=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_NONE=y
CONFIG_TARGET_KERNEL_PARTSIZE=32
CONFIG_TARGET_ROOTFS_INITRAMFS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=4096
CONFIG_USE_LLVM_BUILD=y
# CONFIG_WOLFSSL_ALT_NAMES is not set
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_PACKAGE_libxdp is not set
# CONFIG_PACKAGE_xdp-filter is not set
# CONFIG_PACKAGE_xdp-loader is not set
# CONFIG_PACKAGE_xdpdump is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

VA1DER, 2023-05-10:
I am experiencing the same issue compiling a full build of master on an aarch64 machine.
A multi-core `make -j 4 world` was first attempted, then after that failed on gettext-full, the package was cleaned and started again singly with `make V=sc package/gettext-full/clean, compile`:
```
Making all in src
make[7]: Entering directory '/mnt/nbd/devel/openwrt/build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1/gettext-runtime/src'
aarch64-openwrt-linux-musl-gcc -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H -I. -I..  -I. -I. -I.. -I../intl -I./../intl -I../gnulib-lib -I./../gnulib-lib -I/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include  -DINSTALLDIR=\"/usr/bin\" -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/nbd/devel/openwrt/build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1=gettext-0.21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC  -c -o gettext-gettext.o `test -f 'gettext.c' || echo './'`gettext.c
/bin/sh ../libtool  --tag=CC   --mode=link aarch64-openwrt-linux-musl-gcc -DINSTALLDIR=\"/usr/bin\" -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/nbd/devel/openwrt/build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1=gettext-0.21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC   -L/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -znow -zrelro  -o gettext gettext-gettext.o ../gnulib-lib/libgrt.a ../intl/libintl.la     
libtool: link: aarch64-openwrt-linux-musl-gcc "-DINSTALLDIR=\"/usr/bin\"" -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/nbd/devel/openwrt/build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1=gettext-0.21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -DPIC -fPIC -znow -zrelro -o .libs/gettext gettext-gettext.o  -L/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib ../gnulib-lib/libgrt.a ../intl/.libs/libintl.so
/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: ../gnulib-lib/libgrt.a(libgrt_a-iconv.o): in function `rpl_iconv':
iconv.c:(.text+0x3a4): undefined reference to `u8_mbtoucr'
/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: iconv.c:(.text+0x3a8): undefined reference to `u8_mbtoucr'
/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: iconv.c:(.text+0x3e4): undefined reference to `u8_uctomb_aux'
/mnt/nbd/devel/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: iconv.c:(.text+0x3e8): undefined reference to `u8_uctomb_aux'
collect2: error: ld returned 1 exit status
make[7]: *** [Makefile:1642: gettext] Error 1
```

peterwillcn, 2023-05-10:
Perhaps it may have caused a problem with a certain commit
https://github.com/openwrt/openwrt/commit/ca8577f930713999fe27bdb2bd8c33ac8afb5bc8
https://github.com/openwrt/openwrt/commit/f7fbe77115924d831dfa7e5b9dc8903b4ccaaf07
https://github.com/openwrt/openwrt/commit/d95d5d2a3a895f5374b56ab349f628578cd4149b
https://github.com/openwrt/openwrt/commit/d167adbc4418b664dd825b8b4e7085c2c7322271

Ansuel, 2023-05-11:
@mpratt14 any clue?

mpratt14, 2023-05-11:
I thought it was only going wrong on the SDK

compiles fine on my system, armhf debian

it was ok for the CI too...

I just made a new SDK earlier and was going to try later

mpratt14, 2023-05-11:
in the bulldbot output for the SDK, there was another undefined reference before this one, so not exactly the same result...

mpratt14, 2023-05-11:
with 2 of my host machines, I cannot reproduce the issue

however buildbot log provides some clues
https://downloads.openwrt.org/snapshots/faillogs/mipsel_mips32/base/gettext-full/compile.txt

there is supposed to be some objects compiled that are not being compiled for libgrt

@peterwillcn please show result of
`find build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib | grep Makefile.am | xargs diff`

also show result of 
`find build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr`

mpratt14, 2023-05-11:
@VA1DER you can post the same if you like

`find build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1/gettext-runtime/gnulib-lib | grep Makefile.am | xargs diff`

`find build_dir/target-aarch64_cortex-a53_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr`

peterwillcn, 2023-05-12:
@mpratt14 
1, find build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib | grep Makefile.am | xargs diff
```
3c3
< # Copyright (C) 2002-2022 Free Software Foundation, Inc.
---
> # Copyright (C) 2002-2023 Free Software Foundation, Inc.
7c7
< # the Free Software Foundation; either version 3 of the License, or
---
> # the Free Software Foundation, either version 3 of the License, or
86a87
> libgrt_a_CFLAGS = $(AM_CFLAGS) $(GL_CFLAG_GNULIB_WARNINGS)
138c139,164
< ## begin gnulib module atexit
---
> ## begin gnulib module assert-h
>
> BUILT_SOURCES += $(ASSERT_H)
>
> # We need the following in order to create <assert.h> when the system
> # doesn't have one that works with the given compiler.
> if GL_GENERATE_ASSERT_H
> assert.h: assert.in.h verify.h $(top_builddir)/config.status
> 	$(gl_V_at){ $(SED_HEADER_STDOUT) \
> 	      -e 's|@''INCLUDE_NEXT''@|$(INCLUDE_NEXT)|g' \
> 	      -e 's|@''PRAGMA_SYSTEM_HEADER''@|@PRAGMA_SYSTEM_HEADER@|g' \
> 	      -e 's|@''PRAGMA_COLUMNS''@|@PRAGMA_COLUMNS@|g' \
> 	      -e 's|@''NEXT_ASSERT_H''@|$(NEXT_ASSERT_H)|g' \
> 	      < $(srcdir)/assert.in.h && \
> 	  sed -e '/@assert.h omit start@/,/@assert.h omit end@/d' \
> 	      -e 's|_gl_verify|_gl_static_assert|g' \
> 	      -e 's|_GL_VERIFY|_GL_STATIC_ASSERT|g' \
> 	      -e 's|_GL\(_STATIC_ASSERT_H\)|_GL\1|g' \
> 	      < $(srcdir)/verify.h; \
> 	} > $@-t
> 	$(AM_V_at)mv $@-t $@
> else
> assert.h: $(top_builddir)/config.status
> 	rm -f $@
> endif
> MOSTLYCLEANFILES += assert.h assert.h-t
139a166
> EXTRA_DIST += assert.in.h verify.h
141c168
< EXTRA_DIST += atexit.c
---
> ## end   gnulib module assert-h
143c170,174
< EXTRA_libgrt_a_SOURCES += atexit.c
---
> ## begin gnulib module atexit
>
> if GL_COND_OBJ_ATEXIT
> libgrt_a_SOURCES += atexit.c
> endif
206,209c237,239
<
< EXTRA_DIST += canonicalize-lgpl.c
<
< EXTRA_libgrt_a_SOURCES += canonicalize-lgpl.c
---
> if GL_COND_OBJ_CANONICALIZE_LGPL
> libgrt_a_SOURCES += canonicalize-lgpl.c
> endif
231,234c261,263
<
< EXTRA_DIST += close.c
<
< EXTRA_libgrt_a_SOURCES += close.c
[... 586 lines cut ...]
> endif
2480,2482c2563
< EXTRA_DIST += windows-initguard.h windows-rwlock.c windows-rwlock.h
<
< EXTRA_libgrt_a_SOURCES += windows-rwlock.c
---
> EXTRA_DIST += windows-initguard.h windows-rwlock.h
2530a2612,2615
> distclean-local: distclean-gnulib-libobjs
> distclean-gnulib-libobjs:
> 	-rm -f @gl_LIBOBJDEPS@
> maintainer-clean-local: distclean-gnulib-libobjs
```

2, find build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr

```
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-mbtoucr.c
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-uctomb-aux.c
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-uctomb-aux.c~
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-uctomb.c
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-mbtoucr.c~
build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr/u8-uctomb.c~
```

mpratt14, 2023-05-12:
the second thing I want to see after a compile attempt, so there is object files

peterwillcn, 2023-05-12:
> the second thing I want to see after a compile attempt, so there is object files

What other logs do I need to provide?

mpratt14, 2023-05-12:
the last log you post (second item)
`find build_dir/target-x86_64_musl/gettext-0.21.1/gettext-runtime/gnulib-lib/unistr`

should be after a build attempt, there should be object files not just C files

was it that way after trying to build? so none of the objects built?

mpratt14, 2023-05-12:
but you dont have to post anymore I think I have it figured out...
it is similar to the problem with libxml #12402 

I thought it was a bad Makefile.am but it's deeper than that.
There is no difference between the new and old Makefile.am for this object

Whether the object is built at all is controlled by a symbol in gnulib-lib/Makefile.am

```
if LIBUNISTRING_COMPILE_UNISTR_U8_MBTOUCR
libgrt_a_SOURCES += unistr/u8-mbtoucr.c
endif
```

and 

```
if LIBUNISTRING_COMPILE_UNISTR_U8_UCTOMB
libgrt_a_SOURCES += unistr/u8-uctomb.c unistr/u8-uctomb-aux.c
endif
```

those symbols are defined in the configure scripts and tied to the symbol
`HAVE_LIBUNISTRING`
and there is a discrepancy between the host compile and the target compile on some (but not all) hosts in the configure script log

in host compile:
https://downloads.openwrt.org/snapshots/faillogs/mipsel_mips32/base/gettext-full/host-compile.txt

```
checking for libunistring... no, consider installing GNU libunistring
...
...
checking whether included libunistring is requested... no
checking for libunistring... no, consider installing GNU libunistring
```

and in target compile:
https://downloads.openwrt.org/snapshots/faillogs/mipsel_mips32/base/gettext-full/compile.txt

```
checking for libunistring... yes
checking how to link with libunistring... -lunistring
checking for libunistring version... 1.1.0
...
...
checking for libunistring... yes
checking how to link with libunistring... -lunistring
checking for libunistring version... 1.1.0
...
...
checking for libunistring... yes
checking how to link with libunistring... -lunistring
checking for libunistring version... 1.1.0
```

however, at link time, there is no flag to link to libunistring... It's just not there... it's still assuming that the freshly built libgrt has the functions on its own.

so the solution is to do what we just did for libxml

should be for the last time, since that's the only list of things that are not covered with the optional `--with-included`
```
$ ./configure --help | grep with-included
  --with-included-gettext use the GNU gettext library included here
  --with-included-libxml  use the libxml2 included here
  --with-included-libunistring  use the libunistring parts included here
```

does this sound right? @1715173329

1715173329, 2023-05-12:
didnt ever check it but sounds great

Ansuel, 2023-05-12:
@mpratt14 i can also see some package are failing 

https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a8_vfpv3/packages/samba4/

peterwillcn, 2023-05-12:
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a8_vfpv3/packages/gnutls/compile.txt

mpratt14, 2023-05-12:
I had test build run overnight, posting soon

this definitely has nothing to do with the compiler, it's just bad configuration

Ansuel, 2023-05-12:
@mpratt14 thanks for taking care of this hoping you find the correct solution since this is the main blocker for branching (this and xtables-addons)

micmac1, 2023-05-13:
> @mpratt14 i can also see some package are failing
> 
> https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a8_vfpv3/packages/samba4/

@Ansuel, that's due to GnuTLS, right? I posted a related pull request [here](https://github.com/openwrt/packages/pull/21007).

Ansuel, 2023-05-13:
@micmac1 i would wait @mpratt14 for the feedback and then i would merge thanks a lot !

mpratt14, 2023-05-13:
the PR from @micmac1 brings up a good point here though...

the symbol for linking is still missing for libunistring if we do exactly the same thing as what fixed libxml linking. I was going to just patch the relevant Makefile.am.

that worked, but maybe I don't have to do that if I add the libunistring-optional module to gettext-runtime with autogen.sh, or maybe both is appropriate...

but is the error for gnutls the same (undefined reference) or is it different? I don't remember

Ansuel, 2023-05-13:
@mpratt14 eh we need to find a good solution that is not hacky and would fix each broken package currently...

mpratt14, 2023-05-13:
yeah I just have to figure out which is the least hacky at this point

mpratt14, 2023-05-13:
also interesting to note is that I can reproduce the bug now, but only after adding the configure option --with-libunistring-prefix

mpratt14, 2023-05-14:
@Ansuel I think a patch to the Makefile.am is required here

it's a bug in gettext, linking directly to an external libunistring is supported in gettext-tools but not gettext-runtime, but somehow gettext-runtime is getting the configuration for that from autoconf...

the alternative would be to try to get rid of using autogen.sh or autoreconf at all, but I think it's necessary because of the "relocatable" patch, but we can try that another time...

there seems to be only 2 commits that handles it, none of them touch the gettext-runtime directory

https://git.savannah.gnu.org/gitweb/?p=gettext.git;a=commit;h=ae943bcc1a5a0600d81c736980ad5d2018637217

https://git.savannah.gnu.org/gitweb/?p=gettext.git;a=commit;h=61e21a72f048684fa63d2316646df24a4444a39a

bhaible, 2023-05-14:
> it's a bug in gettext, linking directly to an external libunistring is supported in gettext-tools but not gettext-runtime, but somehow gettext-runtime is getting the configuration for that from autoconf...

I disagree that it's a bug in gettext. The directories `gettext-tools` and `gettext-runtime` have different `configure.ac` and thus different `configure` files. The only way I can imagine an interference between these two directories is if you passed a `--cache-file=...` option or if you have the `CONFIG_SITE` environment variable set.

> the alternative would be to try to get rid of using autogen.sh or autoreconf at all

Re-running the autogen.sh script should produce identical results; so why would you do that?

Running `autoreconf` is not supported. I provide tarballs of my packages, ready to be used with a correct `configure` file. If the first thing you do is to run `autoreconf` and thus replace a working `configure` file with a dysfunctional one, I can't help.

mpratt14's pull request looks OK to me. It will have the effect that the programs `gettext`, `ngettext`, `envsubst` will have an extra link dependency that was not intended upstream; that's the only negative effect.

mpratt14, 2023-05-14:
Sorry....It's actually not correct at this point to say that it only affects some hosts

what actually happens before my PR is:

- host build without libunistring detected on system --> good
- target build without libunistring detected on system --> good
- host build with libunistring detected on system --> good
- target build with libunistring detected on system --> bad

the CI build using the sdk, and some people here, do a `make world` which coincidentally builds libunistring before gettext,
otherwise, even with the strange configuration change to gettext-runtime, it would use the included objects.

bhaible, 2023-05-15:
> I only notice an autom4te.cache after using autogen.sh. Would disabling caching for this build make a difference?

The `autom4te.cache` directories cannot be disabled; they can only be deleted after the autotools have run.
But they are only a cosmetic problem, AFAIK. I haven't seen them cause any malfunction.

> One thing thats clear to me is, gettext-runtime building does require some objects from libunistring. So if I understand you right, the link dependency still exists, but upstream wants the link to only be with libgrt with the included libunistring objects and not a link to an external library, and the only difference should be size and performance right?

Correct.

> it seems that $(LIBUNISTRING) is supposed to be added to $(LIBS)

No it isn't. The Makefile is supposed to link only specific binaries against $(LIBUNISTRING).

> we have a site config file but there is nothing in it about libunistring as far as I can tell.

Can you temporary remove the site config and try to build gettext again? The site config is my primary suspect.

> target build with libunistring detected on system --> bad

So let's focus on what you call the "target builds" (a cross-compilation build, AFAIU). Can you please attach
1. the `configure` command line and environment variables for the "target build",
2. the configure output (stdout, stderr) of that `configure` command,
3. the files `config.log` and `config.status` from the `gettext-runtime` directory.
This should let us progress a bit.

closed 2023-05-15 by openwrt-bot

mpratt14, 2023-05-16:
even though we have a fix for now I would like to keep investigating this

probably I'll make a gist to share files

bhaible, 2023-05-16:
> probably I'll make a gist to share files

You don't need a gist in order to include files in a github issue. Just compress the file and drag-and-drop it into the comment area. Like this:
[dummy.c.gz](https://github.com/openwrt/openwrt/files/11485326/dummy.c.gz)

mpratt14, 2023-05-16:
I was thinking it would be easier for both of us, to be able to have line numbers on screen and keep track of changes between logs

mpratt14, 2023-05-16:
by the way, I think the problem is during autogen.sh, that's when I see a diff between the new and old configure script including the `with-libunistring-prefix` option appearing. if you want I can patch autogen.sh to show autoconf / automake in verbose

but if you really want to see configure log I'll do that too

mpratt14, 2023-05-24:
@bhaible here is all the output

https://gist.github.com/5eff278e48e9ab2285ff4b740e24c08b

here is a direct link to a zip if you prefer that
https://gist.github.com/mpratt14/5eff278e48e9ab2285ff4b740e24c08b/archive/ee76049cd54ebed89de12dbd7beaa8125d54e32d.zip

with no site file, it still happens
I included config.log and config.status for all subdirs except "examples"

in nohup.out:
no site config file warning at line 4
environment variables starts line 37
(using the variables in make, which is a little messy but I filtered some openwrt specific things)
autogen.sh starts line 673
configure starts line 7758
make all starts line 9788

mpratt14, 2023-05-24:
by the way, the site config would consist of these two combined, if I did not delete them for testing

https://github.com/openwrt/openwrt/blob/main/include/site/mipsel
https://github.com/openwrt/openwrt/blob/main/include/site/linux

bhaible, 2023-05-24:
> by the way, the site config would consist of these two combined, if I did not delete them for testing
> 
> https://github.com/openwrt/openwrt/blob/main/include/site/mipsel https://github.com/openwrt/openwrt/blob/main/include/site/linux

Among these settings, some look fishy or dangerous:
```
ac_cv_c_long_double=no
as_cv_unaligned_access=yes
```

mpratt14, 2023-05-25:
> So, apparently, you have modified `gettext-runtime/configure` in some way, other than calling `autoreconf`, and this is what caused the issue. No issue is present in the gettext-0.21.1 tarball.

Everything you say sounds good and correct, but I don't see evidence of configure being altered, which is why I'm totally lost...

For example, our patches for gettext
https://github.com/openwrt/openwrt/tree/master/package/libs/gettext-full/patches

1. gettext-tools/misc
2. gettext-tools/man gettext-runtime/man (which I believe can be removed now since I disabled man build in Makefile)
3. libxml.diff
4. gettext-runtime/src/Makefile.am

@bhaible what do you think about me editing autogen.sh to show verbose output from autoconf / automake and post that? would it be helpful?

bhaible, 2023-05-26:
> I don't see evidence of configure being altered (except by autogen.sh)

I have no idea which effects the two autogen.sh invocations in https://github.com/openwrt/openwrt/blob/master/package/libs/gettext-full/Makefile have. I am not familiar with openwrt's build system in general, either.

> what do you think about me editing autogen.sh to show verbose output from autoconf / automake and post that? would it be helpful?

It wouldn't be helpful to me, since I've already drawn my conclusion. What may be helpful to you, for finding what caused gettext-runtime/configure to be modified, is:
1. temporarily revert `200-libunistring-missing-link.patch`,
2. add a printout of `md5sum gettext-runtime/configure` after each step.

mpratt14, 2023-05-27:
in the branch I'm testing with, I have always had that patch removed

I discovered that the gettext-runtime/configure script is modified the same way by autogen.sh in both the host build and the target build (cross-compile), because they end up with the same checksums. So the difference is something else...

I'm going to look at the config.status for host build and I can post all the output from a host build if it helps. I would do a `diff` between files of the different builds but it might be messy because of the different directories.

mpratt14, 2023-05-27:
@bhaible should we not be including our aclocal while doing autogen.sh?

verbose output from autotools when gettext-runtime/configure is replaced:

```
...
...
aclocal.real: found macro gl_LIBUNISTRING_MODULE in gnulib-m4/libunistring-base.m4: 22
aclocal.real: found macro gl_LIBUNISTRING_LIBHEADER in gnulib-m4/libunistring-base.m4: 48
aclocal.real: found macro gl_LIBUNISTRING_LIB_PREPARE in gnulib-m4/libunistring-base.m4: 67
aclocal.real: found macro gl_LIBUNISTRING_VERSION_CMP in gnulib-m4/libunistring-base.m4: 103
...
...
aclocal.real: found macro gl_LIBUNISTRING in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring.m4: 26
aclocal.real: found macro gl_LIBUNISTRING_CORE in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring.m4: 61
aclocal.real: found macro gl_LIBUNISTRING_OPTIONAL in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring-optional.m4: 15
aclocal.real: ignoring macro gl_LIBUNISTRING_MODULE in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring-base.m4: 22
aclocal.real: ignoring macro gl_LIBUNISTRING_LIBHEADER in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring-base.m4: 48
aclocal.real: ignoring macro gl_LIBUNISTRING_LIB_PREPARE in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring-base.m4: 67
aclocal.real: ignoring macro gl_LIBUNISTRING_VERSION_CMP in /path/to/openwrt/staging_dir/host/share/aclocal/libunistring-base.m4: 103
...
...
aclocal.real: saw macro gl_LIBUNISTRING_LIBHEADER
aclocal.real: saw macro gl_LIBUNISTRING_MODULE
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_CONDITIONAL
aclocal.real: saw macro gl_LIBUNISTRING_VERSION_CMP
aclocal.real: saw macro gl_LIBUNISTRING_LIBHEADER
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_LIBUNISTRING_VERSION_CMP
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_LIBUNISTRING
aclocal.real: saw macro gl_LIBUNISTRING
aclocal.real: saw macro gl_LIBUNISTRING_MODULE
aclocal.real: saw macro gl_LIBUNISTRING_LIBHEADER
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_LIBUNISTRING_OPTIONAL
aclocal.real: saw macro gl_LIBUNISTRING_OPTIONAL
aclocal.real: saw macro gl_LIBUNISTRING
aclocal.real: saw macro gl_LIBUNISTRING
aclocal.real: saw macro gl_LIBUNISTRING_CORE
aclocal.real: saw macro gl_LIBUNISTRING_CORE
aclocal.real: saw macro gl_LIBUNISTRING_CORE
...
...
aclocal.real: saw macro gl_LIBUNISTRING_VERSION_CMP
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_LIBUNISTRING_LIB_PREPARE
aclocal.real: saw macro gl_LIBUNISTRING
aclocal.real: saw macro gl_MODULE_INDICATOR
aclocal.real: saw macro gl_LIBUNISTRING_MODULE
aclocal.real: saw macro gl_MODULE_INDICATOR
aclocal.real: saw macro gl_LIBUNISTRING_MODULE
aclocal.real: saw macro gl_LIBUNISTRING_LIBHEADER
aclocal.real: saw macro gl_LIBUNISTRING_LIBHEADER
aclocal.real: saw macro gl_LIBUNISTRING_MODULE
...
...
autom4te: allowed   tokens: ^am__quote$|^AS_FLAGS$|^SHELL$|^PATH_SEPARATOR$|^PACKAGE_NAME$|^PACKAGE_TARNAME$|^PACKAGE_VERSION$|^PACKAGE_STRING$|^PACKAGE_BUGREPORT$|^PACKAGE_URL$|^exec_prefix$|^prefix$|^program_transform_name$|^bindir$|^sbindir$|^libexecdir$|^datarootdir$|^datadir$|^sysconfdir$|^sharedstatedir$|^localstatedir$|^runstatedir$|^includedir$|^oldincludedir$|^docdir$|^infodir$|^htmldir$|^dvidir$|^pdfdir$|^psdir$|^libdir$|^localedir$|^mandir$|^PACKAGE_NAME$|^PACKAGE_TARNAME$|^PACKAGE_VERSION$|^PACKAGE_STRING$|^PACKAGE_BUGREPORT$|^PACKAGE_URL$|^DEFS$|^ECHO_C$|^ECHO_N$|^ECHO_T$|^LIBS$|^build_alias$|^host_alias$|^target_alias$|^AM_[A-Z]+FLAGS$|^INSTALL_PROGRAM$|^INSTALL_SCRIPT$|^INSTALL_DATA$|^am__isrc$|^CYGPATH_W$|^PACKAGE$|^VERSION$|^PACKAGE$|^VERSION$|^ACLOCAL$|^AUTOCONF$|^AUTOMAKE$|^AUTOHEADER$|^MAKEINFO$|^install_sh$|^STRIP$|^INSTALL_STRIP_PROGRAM$|^MKDIR_P$|^mkdir_p$|^AWK$|^SET_MAKE$|^am__leading_dot$|^AMTAR$|^am__tar$|^am__untar$|^CTAGS$|^ETAGS$|^CSCOPE$|^AM_V$|^AM_DEFAULT_V$|^AM_DEFAULT_VERBOSITY$|^AM_BACKSLASH$|^CC$|^CFLAGS$|^LDFLAGS$|^LIBS$|^CPPFLAGS$|^CC$|^CC$|^CC$|^CC$|^ac_ct_CC$|^CC$|^EXEEXT$|^OBJEXT$|^DEPDIR$|^am__include$|^AMDEP_TRUE$|^AMDEP_FALSE$|^AMDEPBACKSLASH$|^am__nodep$|^CCDEPMODE$|^am__fastdepCC_TRUE$|^am__fastdepCC_FALSE$|^JAVA_CHOICE$|^CPP$|^CPPFLAGS$|^CPP$|^GREP$|^EGREP$|^HAVE_GCJ_IN_PATH$|^HAVE_JAVAC_IN_PATH$|^HAVE_JIKES_IN_PATH$|^CONF_JAVAC$|^CLASSPATH$|^CLASSPATH_SEPARATOR$|^HAVE_JAVAC_ENVVAR$|^HAVE_GCJ_C$|^HAVE_JAVAC$|^HAVE_JIKES$|^JAR$|^CONF_JAVAC$|^HAVE_JAVAC_ENVVAR$|^HAVE_GCJ_C$|^HAVE_JAVAC$|^HAVE_JIKES$|^BUILDJAVA$|^CSHARP_CHOICE$|^CSHARP_CHOICE_MONO$|^HAVE_MCS_IN_PATH$|^HAVE_CSC_IN_PATH$|^HAVE_MCS$|^HAVE_CSC$|^CSHARPCOMPFLAGS$|^BUILDCSHARP$|^build$|^build_cpu$|^build_vendor$|^build_os$|^host$|^host_cpu$|^host_vendor$|^host_os$|^AR$|^AR$|^ARFLAGS$|^RANLIB$|^RANLIB$|^gl_ES$|^gl_LIBOBJS$|^gl_LTLIBOBJS$|^STDC_HEADERS$|^_ALL_SOURCE$|^_DARWIN_C_SOURCE$|^_GNU_SOURCE$|^_HPUX_ALT_XOPEN_SOCKET_API$|^_NETBSD_SOURCE$|^_OPENBSD_SOURCE$|^_POSIX_PTHREAD_SEMANTICS$|^__STDC_WANT_IEC_60559_ATTRIBS_EXT__$|^__STDC_WANT_IEC_60559_BFP_EXT__$|^__STDC_WANT_IEC_60559_DFP_EXT__$|^__STDC_WANT_IEC_60559_EXT__$|^__STDC_WANT_IEC_60559_FUNCS_EXT__$|^__STDC_WANT_IEC_60559_TYPES_EXT__$|^__STDC_WANT_LIB_EXT2__$|^__STDC_WANT_MATH_SPEC_FUNCS__$|^_TANDEM_SOURCE$|^_MINIX$|^_POSIX_SOURCE$|^_POSIX_1_SOURCE$|^__EXTENSIONS__$|^_XOPEN_SOURCE$|^_ISOC11_SOURCE$|^CXX_CHOICE$|^CXX$|^ac_ct_CXX$|^CXX$|^CXXFLAGS$|^CXX$|^ANSICXX_TRUE$|^ANSICXX_FALSE$|^CXXDEPMODE$|^am__fastdepCXX_TRUE$|^am__fastdepCXX_FALSE$|^a [... cut ...] LL_TRUE$|^WOE32DLL_FALSE$|^WOE32DLL$|^PERL$|^CROSS_COMPILING$|^CXX$|^ac_ct_CXX$|^CXX$|^CXXFLAGS$|^CXX$|^ANSICXX_TRUE$|^ANSICXX_FALSE$|^CXXDEPMODE$|^am__fastdepCXX_TRUE$|^am__fastdepCXX_FALSE$|^am__fastdepCXX_TRUE$|^am__fastdepCXX_FALSE$|^LIBSCHED$|^ENABLE_LIBASPRINTF_TRUE$|^ENABLE_LIBASPRINTF_FALSE$|^ENABLE_SHARED$|^subdirs$|^LIBOBJS$|^LTLIBOBJS$|^am__EXEEXT_TRUE$|^am__EXEEXT_FALSE$|^gl_LIBOBJS$|^gl_LTLIBOBJS$|^gl_LIBOBJDEPS$|^gltests_LIBOBJS$|^gltests_LTLIBOBJS$|^gltests_LIBOBJDEPS$
...
...
```

PR #12606 (2023-05-15): gettext-full: link to libunistring
Fixes #12577