openwrt/openwrt #12587: Unable to build package/libs/libusb

graysky2, 2023-05-12:
### Describe the bug

Using the latest commit, I cannot build package/libs/libusb for bcm27xx/rpi-4 on x86_64

### OpenWrt version

d5182de637986a62bf654d75d6d8a8dc720b1f28

### OpenWrt target/subtarget

bcm27xx/rpi-4

### Device

raspberry pi 4

### Image kind

Self-built image

### Steps to reproduce

1. Create a fresh `.config`
2. Try to build the toolchain

```
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
```

### Actual behaviour

Build errors on `package/libs/libusb`

Complete build log below, but relevant part:
```
/bin/sh ../libtool  --tag=CC   --mode=compile aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..   -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/usr/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/include/fortify -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/include  -std=gnu11  -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -pthread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/openwrt/build_dir/target-aarch64_cortex-a72_musl/libusb-1.0.26=libusb-1.0.26 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC  -c -o core.lo core.c
libtool: compile:  aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/usr/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/include/fortify -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/include -std=gnu11 -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -pthread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/openwrt/build_dir/target-aarch64_cortex-a72_musl/libusb-1.0.26=libusb-1.0.26 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -c core.c  -fPIC -DPIC -o .libs/core.o
In file included from os/events_posix.h:24,
                 from libusbi.h:116,
                 from core.c:23:
/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.3.0_musl/include/fortify/poll.h:42:30: error: "_REDIR_TIME64" is not defined, evaluates to 0 [-Werror=undef]
   42 | #if defined(_GNU_SOURCE) && !_REDIR_TIME64
      |                              ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[6]: *** [Makefile:662: core.lo] Error 1
```

### Expected behaviour

Build should not error out

### Additional info

Here is the `make -j1 V=s` output
[build.log.txt](https://github.com/openwrt/openwrt/files/11466649/build.log.txt)

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_GETOPT_LONG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SEAMLESS_XZ=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SORT_BIG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_TAR_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_GETOPT=y
CONFIG_BUSYBOX_CONFIG_MOUNTPOINT=y
CONFIG_BUSYBOX_CONFIG_UNXZ=y
CONFIG_BUSYBOX_CONFIG_XZ=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_flock=y
CONFIG_PACKAGE_getopt=y
# CONFIG_PACKAGE_hostapd-common is not set
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-zz-legacy=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-ikconfig=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-macvlan=y
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-sched-cake=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-net-asix-ax88179=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-rtl8150=y
CONFIG_PACKAGE_kmod-usb-net-rtl8152=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-xhci-hcd=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-veth=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap=y
# CONFIG_PACKAGE_libf2fs is not set
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnfnetlink=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_libustream-mbedtls=m
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lscpu=y
CONFIG_PACKAGE_lua=y
# CONFIG_PACKAGE_mkf2fs is not set
CONFIG_PACKAGE_mount-utils=y
CONFIG_PACKAGE_r8152-firmware=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_xtables-legacy=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_ROOTFS_PARTSIZE=256
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2023-05-12:
It looks like this comes from ddfe5678a448ac8875e94f2fb4ddca67416fa14a

https://github.com/openwrt/openwrt/blob/ddfe5678a448ac8875e94f2fb4ddca67416fa14a/toolchain/fortify-headers/patches/001-__ppoll_time64.patch#L8

`fortify-headers` uses the pattern of [1]:
```c
#if defined(_FORTIFY_SOURCE) && _FORTIFY_SOURCE > 0 && defined(__OPTIMIZE__) && __OPTIMIZE__ > 0
```

@httpstorm, could you please have a look over this logic?
Untested, only throwing an idea out there, _maybe_ something like:
```patch
diff --git a/toolchain/fortify-headers/patches/001-__ppoll_time64.patch b/toolchain/fortify-headers/patches/001-__ppoll_time64.patch
index 9c6fea3e661d..05f8c717148e 100644
--- a/toolchain/fortify-headers/patches/001-__ppoll_time64.patch
+++ b/toolchain/fortify-headers/patches/001-__ppoll_time64.patch
@@ -5,7 +5,7 @@
  }
  
 -#ifdef _GNU_SOURCE
-+#if defined(_GNU_SOURCE) && !_REDIR_TIME64
++#if defined(_GNU_SOURCE) && !(defined(_REDIR_TIME64) && _REDIR_TIME64 > 0)
  #undef ppoll
  _FORTIFY_FN(ppoll) int ppoll(struct pollfd *__f, nfds_t __n, const struct timespec *__s,
                               const sigset_t *__m)

```
`_REDIR_TIME64` is defined by musl (seemingly for 32-bit archs) [2], [3]

[1]: https://git.2f30.org/fortify-headers/file/include/unistd.h.html#l24
[2]: https://git.musl-libc.org/cgit/musl/commit/arch?id=38143339646a4ccce8afe298c34467767c899f51
[3]: https://elixir.bootlin.com/musl/v1.2.4/source/arch/arm/bits/alltypes.h.in

httpstorm, 2023-05-12:
@john-tho  
I am sorry I did not consider undefined macros can trigger an error when the compiler is more strict.
Yes, your suggestion should fix it. I think we can skip `> 0` and just do
``` c
#if defined(_GNU_SOURCE) && !(defined(_REDIR_TIME64) && _REDIR_TIME64)
```

I made a PR: https://github.com/openwrt/openwrt/pull/12590
Currently running a test build.

@hauke @neheb @graysky2

graysky2, 2023-05-13:
I applied the PR but still got a build failure on libusb.

httpstorm, 2023-05-13:
@graysky2 
> I applied the PR but still got a build failure on libusb.

I successfully compiled a fresh clone with the `diffconfig` you provided. Try this:

```
make toolchain/fortify-headers/{clean,compile} V=sc
make package/libs/libusb/{clean,compile} V=sc
```

graysky2, 2023-05-13:
https://github.com/openwrt/openwrt/pull/12590 fixed this for me, great job!

graysky2, 2023-05-13:
@httpstorm - yes, I did not remove the old toolchain ... deleted my comment and confirm successful build when the toolchain was rebuilt.  Recommend merging this.

httpstorm, 2023-05-13:
@graysky2 
Yes, I also noticed `make toolchain/fortify-headers/{clean,compile}` is required after you add the patch. To confirm if the fix is applied:
``` bash
cat staging_dir/toolchain-*/include/fortify/poll.h | grep 'defined(_REDIR_TIME64)'
```

should return this line
``` c
#if defined(_GNU_SOURCE) && !(defined(_REDIR_TIME64) && _REDIR_TIME64)
```

commit 9145c4fb (2023-05-13): fortify-headers: fix build error when _REDIR_TIME64 is not defined

some targets do not define the _REDIR_TIME64 macro resulting in a
build error regression since ddfe5678a448ac8875e94f2fb4ddca67416fa14a
fix by checking if the macro is defined

[1] https://github.com/openwrt/openwrt/issues/12587

Fixes: #12587
Fixes: ddfe5678a448 ("fortify-headers: fix inconsistent time_t version of ppoll")
commit 30fca541 (2023-05-18): fortify-headers: fix build error when _REDIR_TIME64 is not defined

some targets do not define the _REDIR_TIME64 macro resulting in a
build error regression since ddfe5678a448ac8875e94f2fb4ddca67416fa14a
fix by checking if the macro is defined

[1] https://github.com/openwrt/openwrt/issues/12587

Fixes: #12587
Fixes: ddfe5678a448 ("fortify-headers: fix inconsistent time_t version of ppoll")

(cherry picked from commit 9145c4fbcb655b2f56c76012da85cb203fdbc72b)
closed 2023-05-13 by openwrt-bot