openwrt/openwrt #12605: openssl: build error with LTO

M95D, 2023-05-13:
### Describe the bug

When CONFIG_USE_LTO is enabled, openssl build fails with errors like these:

```
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /build/openwrt/tmp/cc3JcyV3.ltrans0.ltrans.o: in function `legacy_get_params':
<artificial>:(.text.legacy_get_params+0xb8): undefined reference to `ossl_prov_is_running'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text.legacy_get_params+0xbc): undefined reference to `ossl_prov_is_running'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /build/openwrt/tmp/cc3JcyV3.ltrans0.ltrans.o: in function `legacy_teardown':
<artificial>:(.text.legacy_teardown+0x4): undefined reference to `ossl_prov_ctx_get0_libctx'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text.legacy_teardown+0x8): undefined reference to `ossl_prov_ctx_get0_libctx'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text.legacy_teardown+0x34): undefined reference to `ossl_prov_ctx_free'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text.legacy_teardown+0x38): undefined reference to `ossl_prov_ctx_free'
/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /build/openwrt/tmp/cc3JcyV3.ltrans0.ltrans.o: in function `OSSL_provider_init':
<artificial>:(.text.OSSL_provider_init+0x14): undefined reference to `ossl_prov_ctx_new'
```

### OpenWrt version

git master

### OpenWrt target/subtarget

BCM27xx

### Device

Raspberry Pi 3B+

### Image kind

Self-built image

### Steps to reproduce

Check LTO option in Global build settings.
Check OpenSSL if not already enabled.
Try to build.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2023-05-23:
ping @cotequeiroz

mrkiko, 2023-06-11:
I may have encountered this issue as well - disabling the option I was able to succesfully build.

stklcode, 2023-07-31:
Should be resolved by 906616d20183bb7be4eb71812ef5e76cb3af56a0

closed 2023-07-31 by ynezz