openwrt/openwrt #12608: build error with netfilter

ilxp, 2023-05-14:
### Describe the bug

 CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND objtool
  CHK     include/generated/compile.h
  AR      net/netfilter/built-in.a
  CC [M]  net/netfilter/nf_conntrack_ecache.o
net/netfilter/nf_conntrack_ecache.c:136:12: error: '__nf_conntrack_eventmask_report' defined but not used [-Werror=unused-function]
  136 | static int __nf_conntrack_eventmask_report(struct nf_conntrack_ecache *e,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[7]: *** [scripts/Makefile.build:289: net/netfilter/nf_conntrack_ecache.o] Error 1
make[6]: *** [scripts/Makefile.build:552: net/netfilter] Error 2
make[5]: *** [Makefile:1906: net] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.111'
make[4]: *** [Makefile:25: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.111/.modules] Error 2
make[3]: *** [Makefile:11: compile] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#12.88#3.30#16.15
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   111  100    55  100    56    136    138 --:--:-- --:--:-- --:--:--   275
100   111  100    55  100    56    136    138 --:--:-- --:--:-- --:--:--   274
{"ok":false,"error_code":404,"description":"Not Found"}
Error: Process completed with exit code 1.

How to do？

### OpenWrt version

master

### OpenWrt target/subtarget

master

### Device

x86-64

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2023-05-14:
@ilxp current configuration used to repro this?

ilxp, 2023-05-15:
this is  https://github.com/coolsnowwolf/lede/issues/11211

Ansuel, 2023-05-15:
@ilxp sorry this is a downstream project and the error comes from qsdk patch about chain events for ECM module... ask to the downstream project maintainer to fix the unused function or disable WERROR kernel flag if not possible.