openwrt/openwrt #12610: Incorrect links in staging_dir

savostyanov, 2023-05-15:
### Describe the bug

I am compiling the latest snapshot and get an error. It seems that the `python3` link in my `staging_dir`  points to `python3.10`, but my system switched to `python3.11` already. How do I fix the links in staging_dir?

```
:~/openwrt> uname -a
Linux blaq 6.3.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 11 May 2023 16:40:42 +0000 x86_64 GNU/Linux
:~/openwrt> ls -la /usr/bin/python3*
lrwxrwxrwx 1 root root    10 Apr  5 16:52 /usr/bin/python3 -> python3.11
-rwxr-xr-x 1 root root 14280 Apr  5 16:52 /usr/bin/python3.11
-rwxr-xr-x 1 root root  3248 Apr  5 16:52 /usr/bin/python3.11-config
lrwxrwxrwx 1 root root    17 Apr  5 16:52 /usr/bin/python3-config -> python3.11-config

```

I proceed to update OpenWrt as follows:

```
:~/openwrt> git pull
Already up to date.
:~/openwrt> make menuconfig
:~/openwrt> ./scripts/feeds update -a && ./scripts/feeds install -a && make -j1 V=sc
[... some lines skipped ...]
make[3]: Entering directory '/home/draug/openwrt/tools/pkgconf'
/home/draug/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'gcc',|" -e "s|@CXX@|'g++',|" -e "s|@PKGCONFIG@|/home/draug/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/draug/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/draug/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/home/draug/openwrt/staging_dir/host/include', '-I/home/draug/openwrt/staging_dir/host/include',|" -e "s|@CXXFLAGS@|'-I/home/draug/openwrt/staging_dir/host/include',|" -e "s|@LDFLAGS@|'-L/home/draug/openwrt/staging_dir/host/lib',|" -e "s|@PREFIX@|/home/draug/openwrt/staging_dir/host|" < /home/draug/openwrt/staging_dir/host/lib/meson/openwrt-native.txt.in > /home/draug/openwrt/build_dir/host/pkgconf-1.9.5/openwrt-native.txt
/home/draug/openwrt/staging_dir/host/bin/python3 /home/draug/openwrt/staging_dir/host/bin/meson.py  --native-file /home/draug/openwrt/build_dir/host/pkgconf-1.9.5/openwrt-native.txt -Ddefault_library=static -Dtests=disabled /home/draug/openwrt/build_dir/host/pkgconf-1.9.5/openwrt-build /home/draug/openwrt/build_dir/host/pkgconf-1.9.5/openwrt-build/..
bash: line 1: /home/draug/openwrt/staging_dir/host/bin/python3: No such file or directory
make[3]: *** [Makefile:42: /home/draug/openwrt/build_dir/host/pkgconf-1.9.5/.configured] Error 127
make[3]: Leaving directory '/home/draug/openwrt/tools/pkgconf'
time: tools/pkgconf/compile#0.06#0.01#0.06
    ERROR: tools/pkgconf failed to build.
make[2]: *** [tools/Makefile:226: tools/pkgconf/compile] Error 1
make[2]: Leaving directory '/home/draug/openwrt'
make[1]: *** [tools/Makefile:222: /home/draug/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/draug/openwrt'
make: *** [/home/draug/openwrt/include/toplevel.mk:231: world] Error 2
```

I look at the cause of the error:
```
:~/openwrt> ls -la /home/draug/openwrt/staging_dir/host/bin/python3
lrwxrwxrwx 1 draug users 19 Mar  8  2022 /home/draug/openwrt/staging_dir/host/bin/python3 -> /usr/bin/python3.10
:~/openwrt> ls -la /usr/bin/python3*
lrwxrwxrwx 1 root root    10 Apr  5 16:52 /usr/bin/python3 -> python3.11
-rwxr-xr-x 1 root root 14280 Apr  5 16:52 /usr/bin/python3.11
-rwxr-xr-x 1 root root  3248 Apr  5 16:52 /usr/bin/python3.11-config
lrwxrwxrwx 1 root root    17 Apr  5 16:52 /usr/bin/python3-config -> python3.11-config
```

### OpenWrt version

draug@blaq:~/openwrt> git log | head -1 commit 7ad1fa12efd99dd2a4e4ffa01365280f63e99e55

### OpenWrt target/subtarget

lantiq / xrx200

### Device

BT Home Hub 5A

### Image kind

Self-built image

### Diffconfig

```text
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

savostyanov, 2023-05-15:
@Ansuel

mpratt14, 2023-05-17:
@savostyanov have you tried my PR?

commit f7520403 (2023-05-21): prereq-build: allow host command symlinks to update

This makes the prereq stage update the symlinks
installed into staging_dir/host/bin
by rearrainging the way they are verified.

Before, seeing or installing a symlink would result in
a successful exit code, and not installing a symlink
would result is a failed exit code. However,
that is not able to account for the difference
between existing good and bad links, or whether
the link would be the same if it was reinstalled,
because the check can match the program to a different path.

Instead, let a success exit code represent
identifying an e [... cut]
commit fdfb8484 (2023-05-21): prereq-build: allow host command symlinks to update

This makes the prereq stage update the symlinks
installed into staging_dir/host/bin
by rearrainging the way they are verified.

Before, seeing or installing a symlink would result in
a successful exit code, and not installing a symlink
would result is a failed exit code. However,
that is not able to account for the difference
between existing good and bad links, or whether
the link would be the same if it was reinstalled,
because the check can match the program to a different path.

Instead, let a success exit code represent
identifying an e [... cut]
closed 2023-05-21 by openwrt-bot