openwrt/openwrt #12611: TP-Link TL-WR902AC v3 - mt76x0e 0000:01:00.0: MAC error detected

twildATgmxde, 2023-05-15:
### Describe the bug

Have several identical TP-Link TL-WR902AC. 2 of 9 have the problem that if I activate 5GHz, channel doesn't matter, the systems reports 

[ 4199.398734] mt76x0e 0000:01:00.0: MAC error detected
[ 4200.868210] mt76x0e 0000:01:00.0: MAC error detected

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR902AC v3

### Image kind

Official downloaded image

### Steps to reproduce

Set config to:

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option country 'DE'
        option cell_density '3'
        option htmode 'HT40'
        option channel '52'
        option txpower '7'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option encryption 'psk2+ccmp'
        option key 'XXXXXX'
        option macfilter 'allow'
        option ssid 'chn052@wr03'

activate 5GHz, get 
[ 4232.228213] mt76x0e 0000:01:00.0: MAC error detected
[ 4233.457945] mt76x0e 0000:01:00.0: MAC error detected
[ 4234.907883] mt76x0e 0000:01:00.0: MAC error detected
[ 4232.228213] mt76x0e 0000:01:00.0: MAC error detected
[ 4233.457945] mt76x0e 0000:01:00.0: MAC error detected
[ 4234.907883] mt76x0e 0000:01:00.0: MAC error detected
 

### Actual behaviour

activate 5GHz, get 
[ 4232.228213] mt76x0e 0000:01:00.0: MAC error detected
[ 4233.457945] mt76x0e 0000:01:00.0: MAC error detected
[ 4234.907883] mt76x0e 0000:01:00.0: MAC error detected
[ 4232.228213] mt76x0e 0000:01:00.0: MAC error detected
[ 4233.457945] mt76x0e 0000:01:00.0: MAC error detected
[ 4234.907883] mt76x0e 0000:01:00.0: MAC error detected

### Expected behaviour

%

### Additional info

%

### Diffconfig

```text
%
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

twildATgmxde, 2023-05-15:
Many Tnx for the answer. I did this already all in that way. Channel auto, fix. Power max, low and auto. 

I guess hardware failure one two devices? The the other identical 7 devices are working fine.

brada4, 2023-05-15:
There are some reports on mt76 radios locking up. The exact conditions are not clear.
You need to physically remove power between tries, best to have device in reach:
- try disabling DFS channels and DFS survey.
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n219
- try setting non-dfs channel like 36
- snapshot

twildATgmxde, 2023-05-17:
Thank again. I found the problem. But it sounds really strange.

It is simply related to the location. The problem appears as soon I bring the device to my USB Hub 3.0 with disk.  If a move the device away from that Disk, at least 1m away, the error reporting stops and the radio come up, if i move back to the usb devices the errors logging started and the radio 5GHz goes offline. 

That's reproducible. Doesn't matter of Power or Channel.  

My solution, i located the device to an other place. Now it works fine and stable.

That's really curious to me.

brada4, 2023-05-17:
Try to identify if it is hub or disk, there is no known predictable interference from USB3.0 5GHz +/- .5% signalling rate to 5GHz wifi (5.1ghz+), should be really some arc discharge to gave such strong noise acompanied with electronics burn smell.