openwrt/openwrt #12623: firewall4: IPv6 MLD ICMP types are not recognized

jonathanunderwood, 2023-05-16:
### Describe the bug

When adding some of the ICMPv6 ICMP types to a firewall rule, fw4 doesn't recognize them. Example below.

Note that these types are listed in `fw4.uc` and so should be handled.

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Belkin rt3200

### Image kind

Official downloaded image

### Steps to reproduce

Add a stanza to `/etc/config/firewall`:
```
config rule                                           
        option name 'Allow-MLDv6-Test'                
        option family 'ipv6'                          
        list proto 'icmp'                             
        list icmp_type 'v2-multicast-listener-report'
        list icmp_type 'multicast-listener-query' 
        list icmp_type 'multicast-listener-report'
        list icmp_type 'multicast-listener-done'
        option src 'REDNET'       
        option target 'ACCEPT'
```

On restarting the firewall, observe the error message about invalid options:
```
# /etc/init.d/firewall restart
Section @rule[23] (Allow-MLDv6-Test) option 'icmp_type' specifies invalid value 'v2-multicast-listener-report'
Section @rule[23] (Allow-MLDv6-Test) skipped due to invalid options
```

Note: commenting out `list icmp_type 'v2-multicast-listener-report'` results in the same message about the `multicast-listener-query` line, and so on.

### Actual behaviour

On restarting the firewall, observe the error message about invalid options:
```
# /etc/init.d/firewall restart
Section @rule[23] (Allow-MLDv6-Test) option 'icmp_type' specifies invalid value 'v2-multicast-listener-report'
Section @rule[23] (Allow-MLDv6-Test) skipped due to invalid options
```

### Expected behaviour

Correct firewall rules are generated.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jonathanunderwood, 2023-05-16:
@systemcrash this relates to a change you made - any ideas?

dave14305, 2023-05-16:
What if you specify the `proto` as `icmpv6` or `ipv6-icmp`?

jonathanunderwood, 2023-05-16:
> Those new types were not yet backported to firewall4 in 22.03.

Aha! That makes sense. 

I wonder then if there's still chance to rename them to be more consistent with the nft terms for these types?

jonathanunderwood, 2023-05-18:
Closing this - not actually a bug.

closed 2023-05-18 by jonathanunderwood