openwrt/openwrt #12626: nft -o crashes

brada4, 2023-05-16:
### Describe the bug

nft "optimize" crashes

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Steps to reproduce

```
root@OpenWrt:~# nft list ruleset | nft -o -f -
Merging:
/dev/stdin:5:3-37:              ct state established,related accept comment "!fw4: Allow inbound established and related flows"
/dev/stdin:6:3-23:              ct state invalid drop comment "!fw4: Drop flows with invalid conntrack state"
into:
        ct state vmap { established : accept, related : accept, invalid : drop } comment "!fw4: Allow inbound established and related flows"
Segmentation fault
root@OpenWrt:~#
```

### Actual behaviour

as in reproducer

### Expected behaviour

all vmaps get presented like from any linux
`ssh openwrt.lan nft list ruleset | nft -o -f -`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-05-16:
FWIW, this problem is not present in nftables 1.0.7 in master/snapshot, so it has likely been fixed between nftables 1.0.2 and 1.0.7.

Maybe in this commit:
http://git.netfilter.org/nftables/commit/?id=59bd944f6d75e99fe0c8d743e7fd482672640c2d

brada4, 2023-05-16:
seems similar but not exactly, mine prints one merge then dies. Give me few days to ckeck vs snapshot (though repeater is straightforward if you have snapshot install at hand)

brada4, 2023-05-16:
Yes, no problem with snapshot on same device. Lets wait for some commiter to close with or without action taken.

brada4, 2023-05-16:
No crash.