openwrt/openwrt #12630: ramips: Asus RT-AX54 (AX1800S): bad block in ubi region (excluded from nmbm) results in soft brick

Znevna, 2023-05-17:
### Describe the bug

Hello, if we have a device that develops a bad block in the ubi region, the said device will apparently soft brick.
Only flashing the stock firmware works at this point. Flashing OpenWrt again will result in the same soft brick.
The only thing that seems to work for now is adjusting the remap-range to also include the ubi partition range, that way nmbm catches the bad block and ubi is running happily over it, but this seems to be a discouraged practice.

### OpenWrt version

snapshot from 20 days ago.

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX54

### Image kind

Official downloaded image

### Steps to reproduce

Have a device that develops a bad block in the ubi partition (nmbm excluded range), watch it bootlooping.

### Actual behaviour

The stars align against you and your precious NAND develops a bad block in the wrong sector.
Your device is now a bootlooper.

### Expected behaviour

If the said device develops a bad block, the device shouldn't bootloop but instead deal with the bad block properly.

### Additional info

Forum post with all the details provided by the affected user: https://forum.openwrt.org/t/openwrt-support-for-asus-ax1800hp/149891/31

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2023-05-17:
@hackpascal It looks like you should _not_ return `-EBADMSG` in `mt7621_nfc_read_page_hwecc`.

@Znevna If you are able to build a custom firmware, replace the only `-EBADMSG` with `0` in `target/linux/ramips/files/drivers/mtd/nand/raw/mt7621_nand.c`.

Znevna, 2023-05-17:
Ok, I'm building an image with this change and making it available on the forum for the user to test.
<details><summary>diff</summary>

```
--- a/target/linux/ramips/files/drivers/mtd/nand/raw/mt7621_nand.c
+++ b/target/linux/ramips/files/drivers/mtd/nand/raw/mt7621_nand.c
@@ -1043,7 +1043,7 @@ static int mt7621_nfc_read_page_hwecc(struct nand_chip *nand, uint8_t *buf,
                        dev_dbg(nfc->dev,
                                 "Uncorrectable ECC error at page %d.%d\n",
                                 page, i);
-                       bitflips = -EBADMSG;
+                       bitflips = 0;
                        mtd->ecc_stats.failed++;
                } else if (bitflips >= 0) {
                        bitflips += rc;
```
</details>

The bad block will not get recorded by nmbm, will it? this will only help ubi to deal with it?

LGA1150, 2023-05-18:
It's only for ubi.

Djfe, 2023-05-26:
This is fixed, you may close this. Backport PRs exist

Znevna, 2023-05-26:
The PR is hanging for some reason.
And the user that reported this still didn't reply.
I'll give it a few more days :)

Znevna, 2023-05-28:
The user confirmed that current snapshots work fine.
Hope the backport PR gets accepted for 23.05.

closed 2023-05-28 by Znevna