openwrt/openwrt #12634: Hostapd update causes regression on Raspberry Pi 4B + Cisco enterprise APs

Entropy512, 2023-05-17:
### Describe the bug

After commit 304423a4ff841129f1b6b7504c83dd11c1f71c84, OpenWRT running on a Raspberry Pi 4B is not able to connect reliably to a Cisco enterprise AP (not sure which model).  wpa_supplicant will associated and the interface will pass traffic for 10 seconds, then fail with an "Authentication timeout" even though the interface clearly succeeded in authentication with WPA2-PSK as it is passing traffic/responding to pings.

### OpenWrt version

Anything after commit 304423a4ff841129f1b6b7504c83dd11c1f71c84

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.1

### Image kind

Self-built image

### Steps to reproduce

Flash any build that has the new hostapd version
Attempt to connect in client mode to a Cisco enterprise AP (not sure if normal APs are affected, I can only test these devices against the local network)

### Actual behaviour

Debug log showing failure - this is the section of the log where clear and unambiguous differences from a "working" setup are seen.
```
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: State: ASSOCIATING -> ASSOCIATED
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: nl80211: Set phy0-sta0 operstate 0->0 (DORMANT)
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: netlink: Operstate: ifindex=6 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: Associated to a new BSS: BSSID=bc:26:c7:fe:4f:4f
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: WPA: Update cipher suite selection based on IEs in driver-generated WPA/RSNE in AssocReq - hexdump(len=98): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00 2d 1a 63 00 17 ff 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 00 00 00 00 00 00 00 40 bf 0c 32 50 80 0f fe ff 00 00 fe ff 00 00 dd 09 00 10 18 02 00 00 10 00 00 dd 07 00 50 f2 02 00 01 00 00 00 00 00
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: AP key_mgmt 0x2 network key_mgmt 0x2; available key_mgmt 0x2
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: using KEY_MGMT WPA2-PSK and proto 2
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: AP pairwise cipher 0x10 network pairwise cipher 0x18; available pairwise cipher 0x10
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: using PTK CCMP
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: AP group cipher 0x10 network group cipher 0x18; available group cipher 0x10
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: using GTK CCMP
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: AP mgmt_group_cipher 0x20 network mgmt_group_cipher 0x0; available mgmt_group_cipher 0x20
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: STA/AP is not MFP capable; AP RSNE caps 0x28
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: not using MGMT group cipher
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: clearing AP WPA IE
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: WPA: set AP RSN IE - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 28 00
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: clearing AP RSNXE
Tue May 16 14:26:40 2023 daemon.notice wpa_supplicant[1220]: phy0-sta0: Associated with bc:26:c7:fe:4f:4f
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: Association event - clear replay counter
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: Clear old PTK
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: External notification - portEnabled=0
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: External notification - portValid=0
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: External notification - EAP success=0
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: External notification - portEnabled=1
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: SUPP_PAE entering state CONNECTING
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: enable timer tick
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: EAPOL: SUPP_BE entering state IDLE
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: Setting authentication timeout: 10 sec 0 usec
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: Cancelling scan request
Tue May 16 14:26:40 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: ASSOC INFO: wait for driver port authorized indication
Tue May 16 14:26:40 2023 daemon.notice wpa_supplicant[1220]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: startWhen --> 0
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: disable timer tick
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: SUPP_PAE entering state CONNECTING
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: enable timer tick
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: txStart
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: WPA: drop TX EAPOL in non-IEEE 802.1X mode (type=1 len=0)
Tue May 16 14:26:50 2023 daemon.notice wpa_supplicant[1220]: phy0-sta0: Authentication with bc:26:c7:fe:4f:4f timed out.
Tue May 16 14:26:50 2023 daemon.notice wpa_supplicant[1220]: BSSID bc:26:c7:fe:4f:4f ignore list count incremented to 7, ignoring for 1800 seconds
Tue May 16 14:26:50 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: WPA: Clear old PMK and PTK
Tue May 16 14:26:50 2023 daemon.debug wpa_supplicant[1220]: phy0-sta0: Request to deauthenticate - bssid=bc:26:c7:fe:4f:4f pending_bssid=00:00:00:00:00:00 reason=3 (DEAUTH_LEAVING) state=ASSOCIATED
[... 44 lines cut ...]
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: phy0-sta0: radio_work_free('connect'@0x7f8dc8e680): num_active_works --> 0
Wed May 17 10:21:34 2023 daemon.notice wpa_supplicant[1415]: phy0-sta0: CTRL-EVENT-CONNECTED - Connection to bc:26:c7:fe:4f:40 completed [id=1 id_str=]
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Set phy0-sta0 operstate 0->1 (UP)
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: netlink: Operstate: ifindex=6 linkmode=-1 (no change), operstate=6 (IF_OPER_UP)
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: External notification - portValid=1
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: External notification - EAP success=1
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: SUPP_PAE entering state AUTHENTICATING
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: SUPP_BE entering state SUCCESS
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAP: EAP entering state DISABLED
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: SUPP_PAE entering state AUTHENTICATED
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: Supplicant port status: Authorized
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Set supplicant port authorized for bc:26:c7:fe:4f:40
Wed May 17 10:21:34 2023 kern.info kernel: [  373.023919] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: SUPP_BE entering state IDLE
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL authentication completed - result=SUCCESS
Wed May 17 10:21:34 2023 daemon.notice wpa_supplicant[1415]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: RTM_NEWLINK: ifi_index=6 ifname=phy0-sta0 operstate=5 linkmode=1 ifi_family=0 ifi_flags=0x11003 ([UP][LOWER_UP])
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Set IF_OPER_UP again based on ifi_flags and expected operstate
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: netlink: Operstate: ifindex=6 linkmode=-1 (no change), operstate=6 (IF_OPER_UP)
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: RTM_NEWLINK: ifi_index=6 ifname=phy0-sta0 operstate=6 linkmode=1 ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Entropy512, 2023-05-17:
These seem to be the critical three lines in the failure where things diverge significantly between "working" and "not:
```
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: EAPOL: txStart
Tue May 16 14:26:42 2023 daemon.debug wpa_supplicant[1220]: WPA: drop TX EAPOL in non-IEEE 802.1X mode (type=1 len=0)
Tue May 16 14:26:50 2023 daemon.notice wpa_supplicant[1220]: phy0-sta0: Authentication with bc:26:c7:fe:4f:4f timed out.
```

In the "working" scenario, the following occurs instead of txStart:
`Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: EAPOL: SUPP_BE entering state IDLE`

Entropy512, 2023-05-17:
OK, another headscratcher here (It looks like there is relevant info in the logs I have that precedes the snippets I posted...  But posting the entire logs or even extended logs is going to require me to spend a significant amount of time reviewing them to ensure nothing sensitive leaked) - in the "good" logs, it appears that as soon as the association happens, the kernel now reports link state of "up" to the rest of the system, and in the logs you can see the system responding to this:

```
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Drv Event 46 (NL80211_CMD_CONNECT) received for phy0-sta0
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Connect event (status=0 ignore_next_local_disconnect=0)
Wed May 17 10:21:34 2023 daemon.notice netifd: Network device 'phy0-sta0' link is up
Wed May 17 10:21:34 2023 daemon.notice netifd: Interface 'wwan' has link connectivity
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Associated on 2412 MHz
Wed May 17 10:21:34 2023 daemon.notice netifd: Interface 'wwan' is setting up now
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Associated with bc:26:c7:fe:4f:40
Wed May 17 10:21:34 2023 daemon.info avahi-daemon[2392]: Joining mDNS multicast group on interface phy0-sta0.IPv4 with address REDACTED.
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Operating frequency for the associated BSS from scan results: 2412 MHz
Wed May 17 10:21:34 2023 daemon.info avahi-daemon[2392]: New relevant interface phy0-sta0.IPv4 for mDNS.
Wed May 17 10:21:34 2023 daemon.info avahi-daemon[2392]: Registering new address record for REDACTED on phy0-sta0.IPv4.
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Associated on 2412 MHz
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Associated with bc:26:c7:fe:4f:40
Wed May 17 10:21:34 2023 daemon.notice netifd: Interface 'wwan' is now up
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: nl80211: Set drv->ssid based on scan res info to 'Automation'
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: phy0-sta0: Event ASSOC (0) received
Wed May 17 10:21:34 2023 daemon.debug wpa_supplicant[1415]: phy0-sta0: Association info event
```

The netifd link state handling results are completely absent from the "bad" logs.

I have no idea at all why changing hostapd would cause the kernel to not report link state handling changes properly...

danpawlik, 2023-05-18:
Could you share some config files not just logs?

Entropy512, 2023-05-18:
```

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option country 'US'
        option channel 'auto'
        option hwmode '11a'
        option htmode 'VHT40'
        option cell_density '0'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'sta'
        option network 'wwan'
        option ssid 'REDACTED'
        option encryption 'psk2'
        option key 'REDACTED'
```
Pretty much a basic client setup - no bridging for WLAN in this configuration.  Config works prior to the hostapd upgrade, not afterwards.

Running out of time before I have to take care of other tasks that have piled up for the past few days, going to try and grab some of the association traffic with another device in monitor mode to look for anything that stands out.  Some of these behaviors scream "FortiNAC interference" to me, but IT swears it's still not enforced on the network in question, and MAC address doesn't change before/after the hostapd change anyway.

Entropy512, 2023-05-18:
Reverting https://w1.fi/cgit/hostap/commit/?id=edfcb2f1a9f8eef9ddfa8b67d642f4bd318a6a2f resolves the issue

I suspect that is why the driver stopped reporting link state changes to the rest of the system - it looks like in MLO/MLD, wpa_supplicant handles a bunch of link state that used to be done by kernel.  But why in the world would wpa_supplicant saying it can handle MLO/MLD fail with an interface that most definitely does not support MLO/MLD?  (MLO is a part of 802.11be, Pi4 only does 802.11ac)  Shouldn't the driver just be ignoring this attribute that it clearly doesn't support, instead of going into la la land?

Entropy512, 2023-06-26:
So this issue appears to have gone away sometime in the past month.

I can't see anything obvious that would have fixed it, so I am GUESSING that hostapd may have had some sort of kernel dependency that was not present in 5.15 and was outside of the mac80211 backports package.  I may run an experiment later this week to confirm that the issue went away when bcm27xx switched to 6.1

Entropy512, 2023-07-05:
So I spoke too soon, I was likely deceived by the fact that there's a backoff mechanism here such that the first few failures after initial boot recover very quickly, but eventually reaches a 2 minutes off/10 seconds on pattern.

6.1 did NOT solve the issue and it is still present.

Entropy512, 2023-09-14:
@nbd168 Taking the conversation we started form #13439 here

I confirmed that the regression affects a single-band network at home.  Which means my theory that the multilink stuff is the problem was wrong, and your theory is more likely to be relevant.

Good news is that since I can reproduce by taking the hardware home, I don't have to worry nearly as much about leaking anything sensitive in the logs.  I'll try to collect the logs you requested tomorrow.

dhewg, 2023-09-14:
Drive by comment, but does yesterday's hostapd bump resolve this (noticed some MLO/MLD related commits)?