openwrt/openwrt #12637: Bootloop happens when flashing OpenWRT 22.03.5 kernel.bin from OEM UI on the Strong1200 router

PaulRadchenko, 2023-05-17:
### Describe the bug

I'm flashing OpenWRT 22.03.5 kernel image on the Strong 1200 router using the OEM web UI and it cause a bootloop. I perform steps as described in the https://openwrt.org/toh/strong/1200#installation doc. There are described that "openwrt-ramips-mt7621-mtc_wr1201-initramfs-kernel.bin" image should be flashed firstly to prevent the bootloop, but actually this leads to it.
Image are flashing to 100%, no power supply issues was observed, then router reboots, then continuous reboots appears.
No ping, no SSH, no network connection able to establish.
No able to boot router, so I can't provide a logs or screenshots, can't execute commands.
I search a forums, and see that this issue appears not only with me, and seems it appears from April, 2023. 

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

N/A

### Device

Strong 1200

### Image kind

Official downloaded image

### Steps to reproduce

1. Open https://openwrt.org/toh/strong/1200#installation page
2. Download provided openwrt-ramips-mt7621-mtc_wr1201-initramfs-kernel.bin and openwrt-ramips-mt7621-mtc_wr1201-squashfs-sysupgrade.bin
3. Login into the Strong 1200 OEM Web UI
4. go to System Tools > Firmware Upgrade and flash openwrt-ramips-mt7621-mtc_wr1201-initramfs-kernel.bin
5. Wait for the progress bars for Flashing and Reboot reach 100%  

### Actual behaviour

Continuous reboots appears, router not able to boot to OpenWrt
Router not reachable from the PC, no SSH access, no network connection

### Expected behaviour

Router reboots successfully and boot OpenWRT successfully. 
Network connection available on PC, router works as expected

### Additional info

I don't have a programmer actually to make a dump from the ROM or some cable to connect directly to the board. But if I will find it somewhere, I will update this ticket with results.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-17:
You can try to boot flashed image into failsafe mode if it is functional at all.
Need to press reset button when system bootloader switches to openwrt and the activity/power LED starts blinking 5x/sec

PaulRadchenko, 2023-05-17:
@brada4  Hi, thank you for the quick response.
I tried to boot in the Failsafe mode and use Reset button before I open this bug. This doesn't helps. 
And I didn't observe 5x LEDs blinking - after power On, LEDs switch On for 1 sec, then turn Off, then turns On for the 5-6 seconds and then reboot and all repeats again - seems boot interrupted before the bootloader are loaded completely. Device are not responding on any LAN port connection or USB.
And one thing that I found - TF card was inserted in the router at time when I performed flashing. I take it out and check it content - no extra files was written. But can this affects on the devices listing/order for the flashing, for example?

brada4, 2023-05-17:
Check here (sorry no way to recovery without opening case and connecting serial)
https://forum.openwrt.org/t/strong-1200-mtc-wr1201-bricked-tftp-or-serial-recovery/59680
Developer might know about micdosd better @vk496

brada4, 2023-05-23:
If you have good recovery setup - can you try sysupgrade to v22? i.e if problem is with 22.03 or with later factory images.

PaulRadchenko, 2023-05-23:
Ok, I will try this in couple of days.

PaulRadchenko, 2023-05-28:
I made some testing and I able to upgrade from 19.07.7 to 22.03.1 using the Force Upgrade option, with a warning:
 "**Device mtc,wr1201 not supported by this image. Supported devices: mtc,wr1201 - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall.
Reason: Config cannot be migrated from swconfig to DSA Image check failed.**

**The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform**" 
(Github don't allow me attach a screenshot..). 
Then I able to upgrade from 22.03.1 to 22.03.5 w/o any issues. -sysupgrade.bin images was used. Seems that images, and my device are correct, because finally all it was installed, the issue happens on the stage when move from OEM to OpenWRT -kernel image.

leskanic, 2023-06-29:
> TF card was inserted in the router at time when I performed flashing. I take it out and check it content - no extra files was written. But can this affects on the devices listing/order for the flashing, for example?

I know you recovered successfully, but in case someone else runs into this issue: nope, presence of the card in the slot has nothing to do with it - I just landed in the same hole as you with brand new device with no card inserted. Ordered some cabling and soldering stuff, hopefully I'll be able to recover via serial link.

leskanic, 2023-07-01:
> Ordered some cabling and soldering stuff, hopefully I'll be able to recover via serial link.

I can confirm that this can be fixed using serial access. To avoid issues with Windows drivers for the TTL converter, I used Ubuntu 23 as a platform (Live running off USB stick), with minicom (/dev/ttyUSB0 since that's what showed up in dmesg for my particular pl2303 cable, 57600, 8N1) and tftpd-hpa for serial and tftp comms respectively. From there on, it's sufficient to follow the linked forums article/discussion, initial initramfs version 19.07.10 worked just fine and I force jumped from it right to 22.03.5. The router hung after sysupgrade, but no issues after power cycling it afterwards - perhaps could/should try 22.03.0 first (doesn't really matter for me since I will be putting on custom imagebuilder-based image anyways). Still waiting for wiki access to warn others that direct OEM -> 22.03 upgrade is NOT the way to go.

Thanks for all the hints!