openwrt/openwrt #12639: ../lib/.libs/libgnutls.so: undefined reference to `uc_general_category_or'

snakwu, 2023-05-17:
/home/snakwu/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gc
c/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld:
 ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Nl'
/home/snakwu/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gc
c/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld:
 ../lib/.libs/libgnutls.so: undefined reference to `uninorm_nfkc'
/home/snakwu/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gc
c/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld:
 ../lib/.libs/libgnutls.so: undefined reference to `uc_general_category_or'
collect2: error: ld returned 1 exit status
make[7]: *** [Makefile:2637: psktool] Error 1
make[7]: Leaving directory '/home/snakwu/openwrt/openwrt/build_dir/target-x86_64
_musl/gnutls-3.8.0/src'
make[6]: *** [Makefile:2464: all] Error 2
make[6]: Leaving directory '/home/snakwu/openwrt/openwrt/build_dir/target-x86_64
_musl/gnutls-3.8.0/src'
make[5]: *** [Makefile:2278: all-recursive] Error 1
make[5]: Leaving directory '/home/snakwu/openwrt/openwrt/build_dir/target-x86_64
_musl/gnutls-3.8.0'
make[4]: *** [Makefile:2203: all] Error 2
make[4]: Leaving directory '/home/snakwu/openwrt/openwrt/build_dir/target-x86_64
_musl/gnutls-3.8.0'
make[3]: *** [Makefile:269: /home/snakwu/openwrt/openwrt/build_dir/target-x86_64
_musl/gnutls-3.8.0/.built] Error 2
make[3]: Leaving directory '/home/snakwu/openwrt/openwrt/feeds/packages/libs/gnu
tls'
time: package/feeds/packages/gnutls/compile#188.36#14.62#201.59
    ERROR: package/feeds/packages/gnutls failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/gnutls/compile] Error
 1
make[2]: Leaving directory '/home/snakwu/openwrt/openwrt'
make[1]: *** [package/Makefile:114: /home/snakwu/openwrt/openwrt/staging_dir/tar
get-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt/openwrt'
make: *** [/home/snakwu/openwrt/openwrt/include/toplevel.mk:231: world] Error 2
snakwu@compiler-server:~/openwrt/openwrt$

Ansuel, 2023-05-18:
@mpratt14 noticing also on CI? https://downloads.openwrt.org/snapshots/faillogs/mips_4kec/packages/gnutls/compile.txt

mpratt14, 2023-05-18:
fun fun fun

looks exactly the same as the issue with gettext

seems like everyone in https://github.com/openwrt/packages/pull/21007 was testing without having libunistring built first

micmac1, 2023-05-18:
mpratt14 that's not correct. The fix was merged today and this issue was raised yesterday (in the wrong repo I might add).

mpratt14, 2023-05-18:
oh i see now, my bad

this is a duplicate of issue https://github.com/openwrt/packages/issues/20969

mpratt14, 2023-05-18:
@Ansuel it may take multiple days for the fix to show in CI logs, unless you restart it manually, right?

mpratt14, 2023-05-23:
I don't see issues in the bulldbot logs anymore

@snakwu good to close this?

snakwu, 2023-05-23:
@mpratt14 ok，close this

closed 2023-05-23 by snakwu