openwrt/openwrt #12646: IPQ4019 Fails to boot after soft reset

Codemonkey1973, 2023-05-18:
### Describe the bug

Device boots seemingly without issue after a hardware reset but hangs at random points in the bootloader after a soft reset (reboot command).

### OpenWrt version

228e0e10398ba2b82e8ac07df3369dbfa15d1517

### OpenWrt target/subtarget

ipq40xx/generic

### Device

8devices Habanero DVK

### Image kind

Self-built image

### Steps to reproduce

Perform a hardware reset and allow the device to boot. Once running, issue a reboot command.

### Actual behaviour

Device hangs in the bootloader after software reset

### Expected behaviour

Device should boot

### Additional info

Images built with 9145c4fbcb655b2f56c76012da85cb203fdbc72b work as expected, images built with 228e0e10398ba2b82e8ac07df3369dbfa15d1517 (kernel update) do not.

**Example of performing a software reset:**

`root@OpenWrt:/# reboot
root@OpenWrt:/# [  199.527984] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[  199.568828] device phy0-ap0 left promiscuous mode
[  199.569309] br-wan: port 2(phy0-ap0) entered disabled state
[  199.708114] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[  199.758212] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[  199.839003] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 0
[  199.839126] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 0
[  199.918205] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[  200.028023] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[  200.179088] ath10k_ahb a000000.wifi: mac flush null vif, drop 0 queues 0xffff
[  200.182328] br-lan: port 2(lan1) entered disabled state
[  200.185368] br-lan: port 1(eth0) entered disabled state
[  200.190934] br-wan: port 3(mesh0) entered disabled state
[  200.196918] br-lan: port 1(eth0) entered disabled state
[  200.267356] device lan1 left promiscuous mode
[  200.267650] br-lan: port 2(lan1) entered disabled state
[  200.319970] br-wan: port 3(mesh0) entered blocking state
[  200.320092] br-wan: port 3(mesh0) entered forwarding state
[  200.326862] qca8k-ipq4019 c000000.switch lan1: Link is Down
[  200.336289] device lan2 left promiscuous mode
[  200.336694] br-lan: port 3(lan2) entered disabled state
[  200.384049] device lan3 left promiscuous mode
[  200.385401] br-lan: port 4(lan3) entered disabled state
[  200.449487] device lan4 left promiscuous mode
[  200.449851] br-lan: port 5(lan4) entered disabled state
[  200.653330] br-wan: port 3(mesh0) entered disabled state
[  200.653456] br-wan: port 1(wan) entered disabled state
[  200.663619] device mesh0 left promiscuous mode
[  200.663809] br-wan: port 3(mesh0) entered disabled state
[  200.719376] device wan left promiscuous mode
[  200.719670] br-wan: port 1(wan) entered disabled state
[... 783 lines cut ...]
CONFIG_PCRE_JIT_ENABLED=y
# CONFIG_WOLFSSL_HAS_ECC25519 is not set
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
ubuntu@ubuntu-VirtualBox:~/openwrt$
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-05-18:
Ok, I managed to bisect this and the culprit is:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/firmware/qcom_scm.c?h=v5.15.111&id=a3ea89b5978dbcd0fa55f675c5a1e04611093709

robimarko, 2023-05-18:
The following PR should fix it for now: #12649 
I have replied to the original patch so there will probably be an upstream fix as well

Codemonkey1973, 2023-05-18:
Thanks for looking into this, much appreciated!

mrkiko, 2023-05-18:
Would be very nice to have a more precise overview of the affected devices.
That said, I don't seem to be able to reproduce this on a FRITZ!BOX 7530, even tough I did only one reboot so far.

Codemonkey1973, 2023-05-18:
The device i'm using are 8devices Habanero modules on a custom PCB.

robimarko, 2023-05-18:
I reproduced and tested this on Habanero DVK, it could depend on the QSEE version but its bad even if one board stops rebooting

Djfe, 2023-05-26:
This can be closed now, right?

robimarko, 2023-05-26:
Yes as the revert was merged

closed 2023-06-05 by stintel