openwrt/openwrt #12655: Summarize status of mt76 drivers

lukasz1992, 2023-05-18:
### Describe the bug

There are many different problems with mt76 drivers including all generations of chipset:

- https://github.com/openwrt/mt76/issues/777
- https://github.com/openwrt/mt76/issues/776
- https://github.com/openwrt/mt76/issues/755
- https://github.com/openwrt/mt76/issues/754
- https://github.com/openwrt/mt76/issues/743
- https://github.com/openwrt/mt76/issues/719

Previously betweenwhiles some developers post short replies, sometimes fixing the issues.
But generally problems stay unresolved (and left without answer) for a long time. Some of them are critical issues, because stable work is impossible (like device/firmware hangs).

Could you @nbd168 , @LorenzoBianconi , @ryderlee1110 do any summarization? Like:
- clarifing what are the priorities, which drivers currently receive support and which are abadoned? It would help many people either to wait for resolving problems, switch to another device or debug issues
- which of these problems are being worked out
- when you do not know how, when or just do not intend to solve problems (like these specified examples) please write about it, for people to not expect things that gonna not happen
- tell if any help is needed, especially with cleaning up the issues (mt76 has over 200 open issues, probably the most of them are duplicates or were solved in later versions)

### OpenWrt version

any

### OpenWrt target/subtarget

any

### Device

any

### Image kind

Official downloaded image

### Steps to reproduce

Install any mt76 driver

### Actual behaviour

Hangs, stability issues

### Expected behaviour

stable work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ThiloteE, 2023-05-18:
## With regard to the many open issues:

This encompases not only the mt76 repository, but the whole of OpenWrt. Most of the time issues are not closed, even if the issue is resolved. I only can speculate about why that is the case, so i will not talk about the reasons for why, but I have two proposals that could mitigate the problem at least a little bit:

## 1. Use GitHub syntax to link pull-requests to issues:

See here: https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword. Using these keywords, issues can be closed automatically once the pull-request is merged. The key problem with this "solution" is that OpenWrt's current policy is to not merge pull-requests and close them instead (and merge them via staging trees into openwrt.git) and maintainers of the mt76 repository do so only in very rare circumstances. That means since pull-requests do not get merged, issues will not get closed. Still, linking a pull-request to an issue can go a long way. It shows that

- a) issues have been resolved
- b) or there is an idea for a fix;
- c) reduces the number of open issues

Also, linking is not limited to OpenWrt or mt76 maintainers or Mediatek employees. Anybody can do it.

#### Here some examples for GitHub's Syntax:

You can link a pull request to an issue by using a supported keyword in the pull request's description or in a commit message. The pull request must be on the default branch.

    close
    closes
    closed
    fix
    fixes
    fixed
    resolve
    resolves
    resolved

Linked issue | Syntax | Example
-- | -- | --
Issue in the same repository | KEYWORD #ISSUE-NUMBER | Closes #12655
Issue in a different repository | KEYWORD OWNER/REPOSITORY#ISSUE-NUMBER | Fixes octo-org-example/octo-repo-example#12655
Multiple issues | Use full syntax for each issue | Resolves #12655, resolves #12655, resolves octo-org-example/octo-repo-example#12655

## 2. Using GitHub actions to close issues once PR has been merged

The idea here would be to

A)

1. manually link pull-request to the issue using special GitHub syntax.
2. automatically add the label "merged" to the PR once it was merged.
3. A github action automatically detects the label "merged" and automatically closes the linked issue.

I know for a fact that step one and three are possible. I don't know if step two is feasible.

lukasz1992, 2023-05-18:
Many thanks @ThiloteE for detailed description. But at this point - with my little knowledge of kernel programming - I am not able yet to fix issues and improve drivers in any way. Sometimes I have good observations (can participate with debugging/resolving issues) but this is not enough to help in most of the issues.

Here I prefer to hear/read general reference from developers or OpenWrt team rather than asking them to fix the issues (I could ask them in every reported issue). Depsite I belive they are working hard, too little communication is between them and OpenWRT users.

jow-, 2023-06-08:
Problem: there's 6 open tickets
Solution: let's make a new ticket asking about those 6 open tickets
Problem: there's 7 open tickets
...

Please use mailing lists or forums for discussion.

closed 2023-06-08 by jow-

lukasz1992, 2023-06-09:
@jow- https://github.com/openwrt/mt76/issues - 251 actually.

wirl, 2025-02-19:
> clarifing what are the priorities, which drivers currently receive support and which are abadoned? It would help many people either to wait for resolving problems, switch to another device or debug issues

Missing an answer to this. Some issues stay unanswered and open for months with nobody taking any interest. I understand nobody is entitled to support but it is kind of weird that those who spend time and effort to file bug reports don't even get a response let alone someone trying to reproduce their issue and fix the bug.

It kind of sucks that you have to be "lucky" with the router you buy cuz if it has any bugs for things you need they might never be fixed even if you report them and patiently wait for weeks. It will just put users off from filing bug reports in the first place, which might be for the best given the number of unanswered and open issues.

I've been using OpenWrt since White Russian on wrt54g's and I guess with time the scale grew larger and larger and now it's pretty much developers fixing bugs they run into themselves and maybe fix some related ones if you're lucky. Reporting bugs is pointless when so many bug reports don't even get an answer, let alone a fix. I'm sure this is demotivating to developers too and not only to users.

Clarification and managing users expectations realistically would be great. If it is too much for users to expect a developer to look at their issue and try to reproduce it make that clear in the issue template. That way no users will waste their time collecting logs, doing tests and filing a bug report in the hopes to get their issue or reported bug fixed.

Once more; nobody is entitled to support I get that, just make sure you manage your users expectations properly.

https://www.psychologytoday.com/intl/blog/the-wisdom-of-anger/202309/how-to-manage-expectations