openwrt/openwrt #12661: MT7921u drivers hangs when starts access point

OnlyCharacters, 2023-05-19:
### Describe the bug

`
[   13.047646] mt7921u 2-1:1.3: Message 00020003 (seq 6) timeout
[   13.053416] phy0-ap0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
[   16.247579] mt7921u 2-1:1.3: vendor request req:63 off:d02c failed:-110
[   19.467546] mt7921u 2-1:1.3: vendor request req:63 off:d054 failed:-110
[   22.056415] passwall2: restart when wan ifup
[   22.676952] mt7921u 2-1:1.3: vendor request req:63 off:d058 failed:-110
[   25.887021] mt7921u 2-1:1.3: vendor request req:63 off:53b8 failed:-110
[   29.106829] mt7921u 2-1:1.3: vendor request req:63 off:53c4 failed:-110
[   31.704241] passwall2: restart when wanv6 ifup
[   32.306604] mt7921u 2-1:1.3: vendor request req:66 off:53c4 failed:-110
[   35.506410] mt7921u 2-1:1.3: vendor request req:63 off:d698 failed:-110
[   38.706237] mt7921u 2-1:1.3: vendor request req:63 off:d520 failed:-110
[   41.906062] mt7921u 2-1:1.3: vendor request req:63 off:d518 failed:-110
[   45.115859] mt7921u 2-1:1.3: vendor request req:63 off:d688 failed:-110
[   48.315695] mt7921u 2-1:1.3: vendor request req:63 off:d690 failed:-110
[   51.515532] mt7921u 2-1:1.3: vendor request req:63 off:d558 failed:-110
[   54.715376] mt7921u 2-1:1.3: vendor request req:63 off:d564 failed:-110
[   57.915228] mt7921u 2-1:1.3: vendor request req:63 off:d568 failed:-110
[   61.115072] mt7921u 2-1:1.3: vendor request req:63 off:d7a8 failed:-110
[   64.314940] mt7921u 2-1:1.3: vendor request req:63 off:a150 failed:-110
[   67.514794] mt7921u 2-1:1.3: vendor request req:63 off:a158 failed:-110
[   70.714651] mt7921u 2-1:1.3: vendor request req:63 off:d780 failed:-110
[   73.914523] mt7921u 2-1:1.3: vendor request req:63 off:d770 failed:-110
[   77.114400] mt7921u 2-1:1.3: vendor request req:63 off:d774 failed:-110
[   80.314290] mt7921u 2-1:1.3: vendor request req:63 off:d55c failed:-110
[   83.513857] mt7921u 2-1:1.3: vendor request req:63 off:10e0 failed:-110
[   86.714019] mt7921u 2-1:1.3: vendor request req:63 off:10e4 failed:-110
[   89.914461] mt7921u 2-1:1.3: vendor request req:63 off:10e8 failed:-110
[   93.114886] mt7921u 2-1:1.3: vendor request req:63 off:10ec failed:-110
[   96.315310] mt7921u 2-1:1.3: vendor request req:63 off:10f0 failed:-110
[   99.515701] mt7921u 2-1:1.3: vendor request req:63 off:10f4 failed:-110
[  102.716087] mt7921u 2-1:1.3: vendor request req:63 off:10f8 failed:-110
[  105.916444] mt7921u 2-1:1.3: vendor request req:63 off:10fc failed:-110
[  109.116786] mt7921u 2-1:1.3: vendor request req:63 off:d7dc failed:-110
[  112.317143] mt7921u 2-1:1.3: vendor request req:63 off:d7ec failed:-110
[  115.517432] mt7921u 2-1:1.3: vendor request req:63 off:d7e0 failed:-110
[  118.717454] mt7921u 2-1:1.3: vendor request req:63 off:d7f0 failed:-110
[  121.917669] mt7921u 2-1:1.3: vendor request req:63 off:d7e4 failed:-110
[  125.117298] mt7921u 2-1:1.3: vendor request req:63 off:d7f4 failed:-110
[  128.317497] mt7921u 2-1:1.3: vendor request req:63 off:d7e8 failed:-110
[  131.517434] mt7921u 2-1:1.3: vendor request req:63 off:d7f8 failed:-110
[  131.746515] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230331110902a
[  131.746515] 
[  131.766324] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20230331110939
[  134.502562] br-lan: port 1(phy0-ap0) entered disabled state
[  134.510904] device phy0-ap0 left promiscuous mode
[  134.515733] br-lan: port 1(phy0-ap0) entered disabled state
[  135.766014] br-lan: port 1(phy0-ap0) entered blocking state
[  135.771761] br-lan: port 1(phy0-ap0) entered disabled state
[  135.777662] device phy0-ap0 entered promiscuous mode
[  135.782869] br-lan: port 1(phy0-ap0) entered blocking state
[  135.788510] br-lan: port 1(phy0-ap0) entered forwarding state
`

From system boot to ap ready, it takes nearly 150 seconds.

### OpenWrt version

r22910-a7747e8670

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.4
[... 384 lines cut ...]
CONFIG_PREINITOPT=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_PREINIT_BROADCAST="192.168.0.255"
CONFIG_TARGET_PREINIT_IFNAME="br-lan"
CONFIG_TARGET_PREINIT_IP="192.168.0.2"
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set

`
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-19:
mt76 USB devices need loadable firmware:
`opkg install mt7601u-firmware`

lukasz1992, 2023-05-19:
@brada4 this is bulshit, only mt7601u (not supported by mt76 driver) needs it

OnlyCharacters, 2023-05-19:
> mt76 USB devices need loadable firmware: `opkg install mt7601u-firmware`

Thanks for replay.
But It seems not work.

brada4, 2023-05-19:
Plug ddvice in USB2 port and supply (attach) dmesg of both ports. Your cut does not include initial fw version report.

dhewg, 2023-05-25:
I'm seeing this now too on a build from yesterdays master, on a ipq4019 board, which only has one xhci port.
r22438 didn't have this problem.

brada4, 2023-05-25:
You can unload (rmmod) xhci driver and stay with ehci till reboot

backslashxx, 2023-06-08:
I'm seeing this too on 23.05-rc1

dhewg, 2023-06-24:
With today's e0fb38f4eeed6cd56d4b5178aeb91442b4fe92a5 and this on top [0001-XXX-mt76-test-possible-fix.patch.txt](https://github.com/openwrt/openwrt/files/11855869/0001-XXX-mt76-test-possible-fix.patch.txt) it seems the situation is improved:

```
Sat Jun 24 09:23:11 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan6) --> new PHY
Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.271067] br-lan: port 8(wlan6) entered blocking state
Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.271198] br-lan: port 8(wlan6) entered disabled state
Sat Jun 24 09:23:11 2023 daemon.notice hostapd: wlan6: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.280467] device wlan6 entered promiscuous mode
Sat Jun 24 09:23:15 2023 kern.err kernel: [  217.517950] mt7921u 1-1:1.3: Message 00020003 (seq 14) timeout
Sat Jun 24 09:23:15 2023 kern.err kernel: [  217.518038] wlan6: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
Sat Jun 24 09:23:18 2023 kern.err kernel: [  220.708264] mt7921u 1-1:1.3: vendor request req:63 off:d02c failed:-110
Sat Jun 24 09:23:21 2023 kern.err kernel: [  223.899235] mt7921u 1-1:1.3: vendor request req:63 off:d054 failed:-110
Sat Jun 24 09:23:24 2023 kern.err kernel: [  227.088463] mt7921u 1-1:1.3: vendor request req:63 off:d058 failed:-110
Sat Jun 24 09:23:27 2023 kern.err kernel: [  230.278103] mt7921u 1-1:1.3: vendor request req:63 off:53b8 failed:-110
Sat Jun 24 09:23:31 2023 kern.err kernel: [  233.478425] mt7921u 1-1:1.3: vendor request req:63 off:53c4 failed:-110
Sat Jun 24 09:23:34 2023 kern.err kernel: [  236.679201] mt7921u 1-1:1.3: vendor request req:66 off:53c4 failed:-110
Sat Jun 24 09:23:37 2023 kern.err kernel: [  239.758205] mt7921u 1-1:1.3: Message 00020003 (seq 15) timeout
Sat Jun 24 09:23:37 2023 kern.err kernel: [  239.758699] wlan6: failed to set key (4, ff:ff:ff:ff:ff:ff) to hardware (-110)
Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.964214] mt7921u 1-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.964214]
Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.986083] mt7921u 1-1:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.375295] IPv6: ADDRCONF(NETDEV_CHANGE): wlan6: link becomes ready
Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.378345] br-lan: port 8(wlan6) entered blocking state
Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.381123] br-lan: port 8(wlan6) entered forwarding state
Sat Jun 24 09:23:40 2023 daemon.notice hostapd: wlan6: interface state COUNTRY_UPDATE->ENABLED
Sat Jun 24 09:23:40 2023 daemon.notice hostapd: wlan6: AP-ENABLED
Sat Jun 24 09:23:40 2023 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Sat Jun 24 09:23:41 2023 daemon.notice netifd: Wireless device 'radio2' is now up
Sat Jun 24 09:23:41 2023 daemon.notice netifd: Network device 'wlan6' link is up
```

There's still a ~20s delay during those `vendor request failed` errors, but after that the radio is at least usable (where the system basically complete hung in the past)

OnlyCharacters, 2023-06-24:
> With today's [e0fb38f](https://github.com/openwrt/openwrt/commit/e0fb38f4eeed6cd56d4b5178aeb91442b4fe92a5) and this on top [0001-XXX-mt76-test-possible-fix.patch.txt](https://github.com/openwrt/openwrt/files/11855869/0001-XXX-mt76-test-possible-fix.patch.txt) it seems the situation is improved:
> 
> ```
> Sat Jun 24 09:23:11 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan6) --> new PHY
> Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.271067] br-lan: port 8(wlan6) entered blocking state
> Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.271198] br-lan: port 8(wlan6) entered disabled state
> Sat Jun 24 09:23:11 2023 daemon.notice hostapd: wlan6: interface state UNINITIALIZED->COUNTRY_UPDATE
> Sat Jun 24 09:23:11 2023 kern.info kernel: [  214.280467] device wlan6 entered promiscuous mode
> Sat Jun 24 09:23:15 2023 kern.err kernel: [  217.517950] mt7921u 1-1:1.3: Message 00020003 (seq 14) timeout
> Sat Jun 24 09:23:15 2023 kern.err kernel: [  217.518038] wlan6: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
> Sat Jun 24 09:23:18 2023 kern.err kernel: [  220.708264] mt7921u 1-1:1.3: vendor request req:63 off:d02c failed:-110
> Sat Jun 24 09:23:21 2023 kern.err kernel: [  223.899235] mt7921u 1-1:1.3: vendor request req:63 off:d054 failed:-110
> Sat Jun 24 09:23:24 2023 kern.err kernel: [  227.088463] mt7921u 1-1:1.3: vendor request req:63 off:d058 failed:-110
> Sat Jun 24 09:23:27 2023 kern.err kernel: [  230.278103] mt7921u 1-1:1.3: vendor request req:63 off:53b8 failed:-110
> Sat Jun 24 09:23:31 2023 kern.err kernel: [  233.478425] mt7921u 1-1:1.3: vendor request req:63 off:53c4 failed:-110
> Sat Jun 24 09:23:34 2023 kern.err kernel: [  236.679201] mt7921u 1-1:1.3: vendor request req:66 off:53c4 failed:-110
> Sat Jun 24 09:23:37 2023 kern.err kernel: [  239.758205] mt7921u 1-1:1.3: Message 00020003 (seq 15) timeout
> Sat Jun 24 09:23:37 2023 kern.err kernel: [  239.758699] wlan6: failed to set key (4, ff:ff:ff:ff:ff:ff) to hardware (-110)
> Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.964214] mt7921u 1-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
> Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.964214]
> Sat Jun 24 09:23:37 2023 kern.info kernel: [  239.986083] mt7921u 1-1:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
> Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.375295] IPv6: ADDRCONF(NETDEV_CHANGE): wlan6: link becomes ready
> Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.378345] br-lan: port 8(wlan6) entered blocking state
> Sat Jun 24 09:23:39 2023 kern.info kernel: [  242.381123] br-lan: port 8(wlan6) entered forwarding state
> Sat Jun 24 09:23:40 2023 daemon.notice hostapd: wlan6: interface state COUNTRY_UPDATE->ENABLED
> Sat Jun 24 09:23:40 2023 daemon.notice hostapd: wlan6: AP-ENABLED
> Sat Jun 24 09:23:40 2023 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
> Sat Jun 24 09:23:41 2023 daemon.notice netifd: Wireless device 'radio2' is now up
> Sat Jun 24 09:23:41 2023 daemon.notice netifd: Network device 'wlan6' link is up
> ```
> 
> There's still a ~20s delay during those `vendor request failed` errors, but after that the radio is at least usable (where the system basically complete hung in the past)

I have built it and yes that it seems better.

dhewg, 2023-07-04:
With today's master of ce8c639a6c539534be14a540e7c3e9933d3a34ef, which has the two mentioned mt76 patches above merged, it got worse again. It looks like mt76 tries something in a loop with interrupts disabed, because that long hang causes everything else to deteriorate too, like the ppp connection.

This is with the single net on the mt7921u in disabled state, rebooted, and then the log upon enabling that net:

```
Tue Jul  4 18:49:44 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan6) --> new PHY
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.490522] br-lan: port 8(wlan6) entered blocking state
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.490583] br-lan: port 8(wlan6) entered disabled state
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.495418] device wlan6 entered promiscuous mode
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.501128] br-lan: port 8(wlan6) entered blocking state
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.505214] br-lan: port 8(wlan6) entered forwarding state
Tue Jul  4 18:49:45 2023 kern.info kernel: [  660.511721] br-lan: port 8(wlan6) entered disabled state
Tue Jul  4 18:49:45 2023 daemon.notice hostapd: wlan6: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Jul  4 18:49:48 2023 kern.err kernel: [  663.913594] mt7921u 1-1:1.3: Message 00020003 (seq 14) timeout
Tue Jul  4 18:49:48 2023 kern.err kernel: [  663.913738] wlan6: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
Tue Jul  4 18:49:51 2023 kern.err kernel: [  667.091195] mt7921u 1-1:1.3: vendor request req:63 off:d02c failed:-110
Tue Jul  4 18:49:55 2023 kern.err kernel: [  670.263745] mt7921u 1-1:1.3: vendor request req:63 off:d054 failed:-110
Tue Jul  4 18:49:58 2023 kern.err kernel: [  673.450618] mt7921u 1-1:1.3: vendor request req:63 off:d058 failed:-110
Tue Jul  4 18:50:01 2023 kern.err kernel: [  676.653315] mt7921u 1-1:1.3: vendor request req:63 off:53b8 failed:-110
Tue Jul  4 18:50:04 2023 kern.err kernel: [  679.841642] mt7921u 1-1:1.3: vendor request req:63 off:53c4 failed:-110
Tue Jul  4 18:50:07 2023 kern.err kernel: [  683.013264] mt7921u 1-1:1.3: vendor request req:66 off:53c4 failed:-110
Tue Jul  4 18:50:11 2023 kern.err kernel: [  686.202665] mt7921u 1-1:1.3: vendor request req:63 off:d698 failed:-110
Tue Jul  4 18:50:14 2023 kern.err kernel: [  689.373652] mt7921u 1-1:1.3: vendor request req:63 off:d520 failed:-110
Tue Jul  4 18:50:14 2023 daemon.notice netifd: radio2 (7261): Command failed: ubus call hostapd config_add {"iface":"wlan6", "config":"/var/run/hostapd-phy0.conf"} (Request timed out)
Tue Jul  4 18:50:14 2023 daemon.notice netifd: radio2 (7261): Device setup failed: HOSTAPD_START_FAILED
Tue Jul  4 18:50:14 2023 daemon.notice netifd: Wireless device 'radio2' set retry=0
Tue Jul  4 18:50:14 2023 daemon.crit netifd: Wireless device 'radio2' setup failed, retry=0
Tue Jul  4 18:50:14 2023 daemon.notice netifd: Wireless device 'radio2' is now down
Tue Jul  4 18:50:17 2023 kern.err kernel: [  692.553012] mt7921u 1-1:1.3: vendor request req:63 off:d518 failed:-110
Tue Jul  4 18:50:20 2023 kern.err kernel: [  695.712978] mt7921u 1-1:1.3: vendor request req:63 off:d688 failed:-110
Tue Jul  4 18:50:23 2023 kern.err kernel: [  698.873066] mt7921u 1-1:1.3: vendor request req:63 off:d690 failed:-110
Tue Jul  4 18:50:26 2023 kern.err kernel: [  702.072916] mt7921u 1-1:1.3: vendor request req:63 off:d558 failed:-110
Tue Jul  4 18:50:30 2023 kern.err kernel: [  705.275561] mt7921u 1-1:1.3: vendor request req:63 off:d564 failed:-110
Tue Jul  4 18:50:33 2023 kern.err kernel: [  708.474432] mt7921u 1-1:1.3: vendor request req:63 off:d568 failed:-110
Tue Jul  4 18:50:36 2023 kern.err kernel: [  711.662882] mt7921u 1-1:1.3: vendor request req:63 off:d7a8 failed:-110
Tue Jul  4 18:50:39 2023 kern.err kernel: [  714.873155] mt7921u 1-1:1.3: vendor request req:63 off:a150 failed:-110
Tue Jul  4 18:50:42 2023 kern.err kernel: [  718.062569] mt7921u 1-1:1.3: vendor request req:63 off:a158 failed:-110
Tue Jul  4 18:50:46 2023 kern.err kernel: [  721.252816] mt7921u 1-1:1.3: vendor request req:63 off:d780 failed:-110
Tue Jul  4 18:50:49 2023 kern.err kernel: [  724.452812] mt7921u 1-1:1.3: vendor request req:63 off:d770 failed:-110
Tue Jul  4 18:50:52 2023 kern.err kernel: [  727.652711] mt7921u 1-1:1.3: vendor request req:63 off:d774 failed:-110
Tue Jul  4 18:50:55 2023 kern.err kernel: [  730.843933] mt7921u 1-1:1.3: vendor request req:63 off:d55c failed:-110
Tue Jul  4 18:50:58 2023 kern.err kernel: [  734.035599] mt7921u 1-1:1.3: vendor request req:63 off:10e0 failed:-110
Tue Jul  4 18:51:02 2023 kern.err kernel: [  737.203198] mt7921u 1-1:1.3: vendor request req:63 off:10e4 failed:-110
Tue Jul  4 18:51:05 2023 kern.err kernel: [  740.402847] mt7921u 1-1:1.3: vendor request req:63 off:10e8 failed:-110
Tue Jul  4 18:51:08 2023 kern.err kernel: [  743.602871] mt7921u 1-1:1.3: vendor request req:63 off:10ec failed:-110
Tue Jul  4 18:51:11 2023 kern.err kernel: [  746.799380] mt7921u 1-1:1.3: vendor request req:63 off:10f0 failed:-110
Tue Jul  4 18:51:14 2023 kern.err kernel: [  749.982319] mt7921u 1-1:1.3: vendor request req:63 off:10f4 failed:-110
Tue Jul  4 18:51:18 2023 kern.err kernel: [  753.182183] mt7921u 1-1:1.3: vendor request req:63 off:10f8 failed:-110
Tue Jul  4 18:51:21 2023 kern.err kernel: [  756.392070] mt7921u 1-1:1.3: vendor request req:63 off:10fc failed:-110
[... cut ...]
Tue Jul  4 18:51:46 2023 kern.info kernel: [  782.105894] mt7921u 1-1:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
Tue Jul  4 18:51:49 2023 daemon.notice pppd[4878]: Modem hangup
Tue Jul  4 18:51:49 2023 daemon.notice pppd[4878]: Connection terminated.
Tue Jul  4 18:51:49 2023 daemon.info pppd[4878]: Connect time 10.7 minutes.
Tue Jul  4 18:51:49 2023 daemon.info pppd[4878]: Sent 0 bytes, received 339502 bytes.
Tue Jul  4 18:51:49 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Tue Jul  4 18:51:49 2023 daemon.info pppd[4878]: Exit.
```

brada4, 2023-07-04:
Does the device work in a normal PC? Might be USB power shortage.

dhewg, 2023-07-04:
It works with an openwrt build from a few months ago without changing any hardware or cables, so it has to be a software or firmware problem

OnlyCharacters, 2023-07-05:
I found something strange.
In the Network -> Interface, edit your lan Interface, if you set the Wireless Network as Device instead of br-lan, there is no error when reboot and wifi up the radio. But still has error if you try to config the Access Point like restart it.

[image]
Config in the picture has no error after reboot. But still has error if you try to config the Access Point like restart it.

dhewg, 2023-07-05:
My last working build was d98c4fb8bfabe86d4cfdb8bb5497ec6791d1e36d. That had the mt76 package at c32d6d849c43792abd8007e13e468b12d6d6e0b7. I tried using that revision on a recent master, including the firmware, to no avail - the issue persists.
That makes me think it's probably a kernel bump or patch?

dhewg, 2023-07-05:
Reverting to the used kernel from d98c4fb8bfabe86d4cfdb8bb5497ec6791d1e36d (5.15.104) doesn't help on its own

dhewg, 2023-07-05:
Reverting e722b667c5a59bbd5aff787002f7f5457d0bf721 doesn't help either, no idea what's up

OnlyCharacters, 2023-07-05:
@dhewg Thanks to your work.
Ignore the error when starts the AP, another problem is that the driver is not reliable.
I do a stress test with iperf3 on the rpi 4B plug  with mt7921au.
My phone connected to the ap as iperf3 client, a computer as iperf3 server which connect to a switch, rpi 4B connect to the switch too.  After transfer 13GB data, the AP disapear. By checking the log, the driver was reloaded.

castelan-jesus, 2023-07-05:
> It works with an openwrt build from a few months ago without changing any hardware or cables, so it has to be a software or firmware problem

Personally, I have been dealing with this particular issue since a couple of months or so and I found that if that "vendor request req" error appears, just unplug the USB adapter and then run 'service network restart' or just 'wifi'.

That workaround works even in STA+AP mode.

In my opinion, this bug is present since a time ago. Maybe it went unnoticed because it works perfectly immediately after flashing and problems starts to happen after first reboot.

Tested on:
- Gl.iNet AR300m16 running 22.0.5 (with updated drivers from mediatek repo)
- Raspberry Pi Zero 2 W running snapshot (r23507-32ed976446)

dhewg, 2023-07-06:
I only tried one time so far, but it seems reverting 304423a4ff841129f1b6b7504c8 gets rid of the problem

dhewg, 2023-07-06:
The good news is that updating hostapd fixes it too. This patch works for me, can you guys confirm? 
[0001-hostapd-update-to-2023-06-22.patch.txt](https://github.com/openwrt/openwrt/files/11965114/0001-hostapd-update-to-2023-06-22.patch.txt)

Going back to the 2023-03-29-bb945b98 version breaks it again, so it's not just a once-after-boot/flash issue.

castelan-jesus, 2023-07-07:
> > The good news is that updating hostapd fixes it too. This patch works for me, can you guys confirm? [0001-hostapd-update-to-2023-06-22.patch.txt](https://github.com/openwrt/openwrt/files/11965114/0001-hostapd-update-to-2023-06-22.patch.txt)
> > 
> > Going back to the 2023-03-29-bb945b98 version breaks it again, so it's not just a once-after-boot/flash issue.
> 
> I have applied that patch and I'm getting this error when turning on STA+AP:
> ```
> 
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Interface phy1-ap0.IPv6 no longer relevant for mDNS.
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Leaving mDNS multicast group on interface phy1-ap0.IPv6 with address fe80::e65f:1ff:fe80:3935.
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Withdrawing address record for fe80::e65f:1ff:fe80:3935 on phy1-ap0.
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: Remove interface 'phy0-ap0'
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: phy0-ap0: AP-DISABLED
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
> Tue Jan  1 00:00:39 2030 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.036674] device phy0-ap0 left promiscuous mode
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.043206] br-lan: port 3(phy0-ap0) entered disabled state
> Tue Jan  1 00:00:39 2030 kern.err kernel: [  580.093162] ieee80211 phy1: brcmf_cfg80211_get_channel: chanspec failed (-52)
> Tue Jan  1 00:00:39 2030 daemon.notice netifd: Network device 'phy0-ap0' link is down
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Interface phy0-ap0.IPv6 no longer relevant for mDNS.
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Leaving mDNS multicast group on interface phy0-ap0.IPv6 with address fe80::e2e1:a9ff:fe35:d13d.
> Tue Jan  1 00:00:39 2030 daemon.info avahi-daemon[1105]: Withdrawing address record for fe80::e2e1:a9ff:fe35:d13d on phy0-ap0.
> Tue Jan  1 00:00:39 2030 daemon.notice netifd: Wireless device 'radio0' is now down
> Tue Jan  1 00:00:39 2030 daemon.notice netifd: radio0 (6623): command failed: No error information (-524)
> Tue Jan  1 00:00:39 2030 daemon.notice netifd: Wireless device 'radio1' is now down
> Tue Jan  1 00:00:39 2030 daemon.notice netifd: radio0 (6623): command failed: I/O error (-5)
> Tue Jan  1 00:00:39 2030 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.573173] br-lan: port 2(phy1-ap0) entered blocking state
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.579959] br-lan: port 2(phy1-ap0) entered disabled state
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.586984] device phy1-ap0 entered promiscuous mode
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.593171] br-lan: port 2(phy1-ap0) entered blocking state
> Tue Jan  1 00:00:39 2030 kern.info kernel: [  580.599790] br-lan: port 2(phy1-ap0) entered forwarding state
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  580.783330] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
> Tue Jan  1 00:00:40 2030 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->ENABLED
> Tue Jan  1 00:00:40 2030 daemon.notice hostapd: phy1-ap0: AP-ENABLED
> Tue Jan  1 00:00:40 2030 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.069692] br-lan: port 3(phy0-ap0) entered blocking state
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.076814] br-lan: port 3(phy0-ap0) entered disabled state
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.083895] device phy0-ap0 entered promiscuous mode
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.090312] br-lan: port 3(phy0-ap0) entered blocking state
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.097063] br-lan: port 3(phy0-ap0) entered forwarding state
> Tue Jan  1 00:00:40 2030 kern.info kernel: [  581.481021] br-lan: port 3(phy0-ap0) entered disabled state
> Tue Jan  1 00:00:41 2030 daemon.notice netifd: Wireless device 'radio0' is now up
> Tue Jan  1 00:00:41 2030 daemon.notice netifd: Network device 'phy1-ap0' link is up
> Tue Jan  1 00:00:42 2030 daemon.info avahi-daemon[1105]: Joining mDNS multicast group on interface phy1-ap0.IPv6 with address fe80::e65f:1ff:fe80:3935.
> Tue Jan  1 00:00:42 2030 daemon.info avahi-daemon[1105]: New relevant interface phy1-ap0.IPv6 for mDNS.
> Tue Jan  1 00:00:42 2030 daemon.info avahi-daemon[1105]: Registering new address record for fe80::e65f:1ff:fe80:3935 on phy1-ap0.*.
> Tue Jan  1 00:00:43 2030 kern.err kernel: [  584.230921] mt7921u 1-1.3:1.3: Message 00020003 (seq 14) timeout
> Tue Jan  1 00:00:43 2030 kern.err kernel: [  584.238269] phy0-ap0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
> Tue Jan  1 00:00:46 2030 kern.err kernel: [  587.430926] mt7921u 1-1.3:1.3: vendor request req:63 off:d02c failed:-110
> Tue Jan  1 00:00:49 2030 kern.err kernel: [  590.630926] mt7921u 1-1.3:1.3: vendor request req:63 off:d054 failed:-110
> Tue Jan  1 00:00:53 2030 kern.err kernel: [  593.830937] mt7921u 1-1.3:1.3: vendor request req:63 off:d058 failed:-110
[... cut ...]
> Tue Jan  1 00:00:45 2030 kern.info kernel: [  708.566451] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
> Tue Jan  1 00:00:45 2030 kern.info kernel: [  708.574412] br-lan: port 3(phy0-ap0) entered blocking state
> Tue Jan  1 00:00:45 2030 kern.info kernel: [  708.581104] br-lan: port 3(phy0-ap0) entered forwarding state
> ```
> 
> After a minute or two, the AP turns on but I have no internet access. It's working fine on STA mode.
> 
> - Tested on Raspberry Pi Zero 2 W (openWrt snapshot)

I'm sorry, I just checked again and it was my mistake. The patch actually works!

dhewg, 2023-07-07:
> I'm sorry, I just checked again and it was my mistake. The patch actually works!

Nice, thanks!

See #13064

commit 8d7d9aa4 (2023-07-20): hostapd: update to 2023-06-22

Removed, merged upstream:
- 170-wpa_supplicant-fix-compiling-without-IEEE8021X_EAPOL.patch

Manually refreshed:
- 040-mesh-allow-processing-authentication-frames-in-block.patch
- 600-ubus_support.patch
- 761-shared_das_port.patch

Fixes: #12661
Fixes: 304423a4 ("hostapd: update to 2023-03-29")

(cherry picked from commit cd804c1ebb34c9740f8ad9fe393a81f3efb7b009)
closed 2023-07-07 by openwrt-bot