openwrt/openwrt #12665: tools/meson: current version cannot compile newest GStreamer

flyn-org, 2023-05-19:
### Describe the bug

I am the maintainer of OpenWrt's GStreamer packages. The latest version of GStreamer requires Meson >= 0.62, and thus OpenWrt will not compile it. I noticed in the Git log that Nick Hainke updated Meson to 0.63.1, but Daniel Golle later reverted that commit. A log message from Rosen Penev on June 20, 2022 indicates that newer Mesons require Python, so I suspect that is why OpenWrt has not yet adopted them. That might be the case, but I expect more software will start requiring newer versions of Meson than we provide.

### OpenWrt version

master

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-05-19:
I think the issue was with meson requiring Python 3.7 while Ubuntu 18.04 and like were shipping 3.6 only.
After branching it maybe time to loosen that requirement and bit

Entropy512, 2023-05-19:
At this point 18.04 is TWO LTS revisions behind, and we're now more than halfway to the next LTS...

brada4, 2023-05-19:
No wonder python scripts require python.  Just bump version and checksum in local makefile and cross fingers that glib and other things using it will not break.

eli-schwartz, 2023-05-21:
Relevant:
#12186
#10543
#10649

Python 3.6 is:
- EOL, and not supported by the Python developers
- not supported by Ubuntu, even to the extent that Python developers released security fixes for 3.6.15
- shipped in an LTS distro that is now merely 10 days away from going out of standard support
- no longer listed on https://packages.ubuntu.com which already de-listed Ubuntu Bionic entirely
- irrelevant even on that ancient Ubuntu, since you can install python 3.8 there -- it just isn't the default python

eli-schwartz, 2023-05-21:
> I think the issue was with meson requiring Python 3.7 while Ubuntu 18.04 and like were shipping 3.6 only.

This is false. Ubuntu 18.04 ships both 3.6 and 3.8.

xback, 2023-05-22:
@flyn-org 
https://patchwork.ozlabs.org/project/openwrt/patch/20230217091810.1381853-1-koen.vandeputte@citymesh.com/

xback, 2023-05-22:
https://patchwork.ozlabs.org/project/openwrt/patch/20230217091810.1381853-1-koen.vandeputte@citymesh.com/

robimarko, 2023-05-22:
Well, Python 3.7 was just merged as the minimal supported version so meson can be updated to 1.0.2.

hnyman, 2023-05-22:
> so meson can be updated to 1.0.2.

There is one commit updating meson already in the PR #12186 by @neheb,
hidden by the PR title talking only about Python requirement.

https://github.com/openwrt/openwrt/pull/12186/commits/c19a9edff7810a5c5ac843e07f7806b56481825d
"tools/meson: update to 1.0.1 "

neheb, 2023-05-22:
Latest version of meson is 1.1.0. I leave it for someone else to update it as I am away currently.

flyn-org, 2023-05-22:
With PR #12186 merged by OpenWrt, I manually applied PR #10543 to my local tree. This allowed GStreamer 1.22.2 to build. Can we merge PR #10543 now an migrate to an even newer Meson later?

robimarko, 2023-05-24:
Meson 1.1.0 was merged so it should work again

flyn-org, 2023-05-24:
Merging PR #10543 fixed this. Thank you!

closed 2023-05-24 by flyn-org