openwrt/openwrt #12666: error when update list on wrt1900acs SNAPSHOT r22962-cf19fd0cb5 / LuCI Master git-23.118.79121-6fb185f

lopez67876, 2023-05-19:
### Describe the bug

Errors
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/5.15.112-1-e8ff30c3eb11e99ff53d0a64a0978063/Packages.gz, wget returned 8.
The opkg update command failed with code 1.

### OpenWrt version

wrt1900acs SNAPSHOT r22962-cf19fd0cb5

### OpenWrt target/subtarget

SNAPSHOT r22962-cf19fd0cb5

### Device

wrt1900acs

### Image kind

Official downloaded image

### Steps to reproduce

press update list on luci

### Actual behaviour

Errors
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/5.15.112-1-e8ff30c3eb11e99ff53d0a64a0978063/Packages.gz, wget returned 8.
The opkg update command failed with code 1.

### Expected behaviour

nomal behabour

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

reusu, 2023-05-20:
same 5.15.112 packages does not exists

lopez67876, 2023-05-21:
Fixed on OpenWrt SNAPSHOT r23001. Thanks.

closed 2023-05-21 by lopez67876