openwrt/openwrt #12668: DFS CAC + DSA + Secondary interfaces = wrong VLAN assignment

F-i-f, 2023-05-19:
### Describe the bug

I've found an issue that occurs when all three conditions are met:
- DSA is in use and wifi interfaces are assigned to different VLANs
- There are secondary wifi interfaces on the 5 GHz phy.
- The configured 5 GHz WiFi channel uses DFS/CAC.

In that case, when the network comes up, the following happens:
- All 2.5 GHz interfaces come up initially and are assigned to the bridge with the correct VLANs.
- The first (primary?) 5 GHZ comes up, is assigned to the bridge with the correct VLANs.
- The secondary 5 GHz interfaces are not added until CAC/DFS runs to completion.
- After ~1 minute, CAC/DFS completes, the secondary 5 GHZ interfaces get added to the bridge, but with VLAN id 1, not the configured VLAN id.

### OpenWrt version

r22894-d53fe5d9ce

### OpenWrt target/subtarget

ipq807x/generic

### Device

Dynalink DL-WRX36

### Image kind

Official downloaded image

### Steps to reproduce

## Configure network and wifi like this:

- `/etc/config/network`:
  ```
  config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

  config globals 'globals'
	option ula_prefix 'fd45:84f4:8331::/48'

  config device
	option name 'br-vlan'
	option type 'bridge'
	option macaddr 'xx:xx:xx:xx:xx:xx'
	option stp '1'
	option priority '32768'
	option igmp_snooping '1'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

  config device
	option name 'br-vlan.11'
	option type '8021q'
	option ifname 'br-vlan'
	option vid '11'

  config device
	option name 'br-vlan.12'
	option type '8021q'
	option ifname 'br-vlan'
	option vid '12'

  config device
	option name 'br-vlan.13'
	option type '8021q'
	option ifname 'br-vlan'
	option vid '13'

  config device
	option name 'br-vlan.18'
	option type '8021q'
	option ifname 'br-vlan'
	option vid '18'

  config interface 'br_vlan_12'
	option device 'br-vlan.12'
	option proto 'none'

  config interface 'br_vlan_13'
	option device 'br-vlan.13'
	option proto 'none'

  config interface 'br_vlan_18'
	option device 'br-vlan.18'
	option proto 'dhcp'

  config bridge-vlan
	option device 'br-vlan'
	option vlan '11'
	list ports 'lan1'
	list ports 'wan:t'

  config bridge-vlan
	option device 'br-vlan'
	option vlan '12'
	list ports 'lan2'
	list ports 'wan:t'

  config bridge-vlan
	option device 'br-vlan'
	option vlan '13'
	list ports 'lan3'
	list ports 'wan:t'

  config bridge-vlan
	option device 'br-vlan'
	option vlan '18'
	list ports 'lan4:u*'
	list ports 'wan:t'

  ```
- `/etc/config/wireless`:
  ```
  config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option country 'US'
	option cell_density '0'
	option band '5g'
	option he_bss_color '43'
	option channel '100'
	option htmode 'HE160'
	option disabled '0'

  config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option country 'US'
	option cell_density '0'
	option band '2g'
	option he_bss_color '43'
	option channel '11'
	option htmode 'HE20'
	option disabled '0'

  config wifi-iface 'wifinet0'
	option ifname 'wl_wlan_5'
	option network 'br_vlan_12'
	option device 'radio0'
	option mode 'ap'
	option ssid 'NET1'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  config wifi-iface 'wifinet1'
	option ifname 'wl_wlan_25'
	option network 'br_vlan_12'
	option device 'radio1'
	option mode 'ap'
	option ssid 'NET1'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  config wifi-iface 'wifinet2'
	option ifname 'wl_wlan_g_5'
	option network 'br_vlan_12'
	option device 'radio0'
	option mode 'ap'
	option ssid 'NET2'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  config wifi-iface 'wifinet3'
	option ifname 'wl_wlan_g_25'
	option network 'br_vlan_12'
	option device 'radio1'
	option mode 'ap'
	option ssid 'NET2'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  config wifi-iface 'wifinet4'
	option ifname 'wl_outlan_5'
	option network 'br_vlan_13'
	option isolate '1'
	option device 'radio0'
	option mode 'ap'
	option ssid 'NET3'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  config wifi-iface 'wifinet5'
	option ifname 'wl_outlan_25'
	option network 'br_vlan_13'
	option isolate '1'
	option device 'radio1'
	option mode 'ap'
	option ssid 'NET3'
	option encryption 'psk2+ccmp'
	option key 'XXX'
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
	option max_inactivity '900'
	option disabled '0'

  ```

## Then run:
```
/etc/init.d/network restart
```

### Actual behaviour

Check the bridge VLAN configuration immediately after (re-)starting the network:
```
# bridge vlan
port              vlan-id  
lan4              18 PVID Egress Untagged
lan3              13 PVID Egress Untagged
lan2              12 PVID Egress Untagged
lan1              11 PVID Egress Untagged
wan               11
                  12
                  13
                  18
br-vlan           10
                  11
                  12
                  13
                  18
wl_wlan_5         12 PVID Egress Untagged
wl_wlan_25        12 PVID Egress Untagged
wl_wlan_g_25      12 PVID Egress Untagged
wl_outlan_25      13 PVID Egress Untagged
```

As described previously, all 3 2.5 GHz interfaces and the primary 5 GHz network are up and assigned the correct VLAN ids.
Now wait a minute for CAC on channel 100 to complete and check again:
```
# bridge vlan
port              vlan-id  
lan4              18 PVID Egress Untagged
lan3              13 PVID Egress Untagged
lan2              12 PVID Egress Untagged
lan1              11 PVID Egress Untagged
wan               11
                  12
                  13
                  18
br-vlan           10
                  11
                  12
                  13
                  18
wl_wlan_5         12 PVID Egress Untagged
wl_wlan_25        12 PVID Egress Untagged
wl_wlan_g_25      12 PVID Egress Untagged
wl_outlan_25      13 PVID Egress Untagged
wl_wlan_g_5       1 PVID Egress Untagged
wl_outlan_5       1 PVID Egress Untagged
```

The two secondary 5 GHz interfaces have been assigned the wrong VLAN ids.

### Expected behaviour

After a network restart and waiting for CAC to complete, the bridge VLAN filter should be configured:
```
# bridge vlan
port              vlan-id  
lan4              18 PVID Egress Untagged
lan3              13 PVID Egress Untagged
lan2              12 PVID Egress Untagged
lan1              11 PVID Egress Untagged
wan               11
                  12
                  13
                  18
br-vlan           10
                  11
                  12
                  13
                  18
wl_wlan_5         12 PVID Egress Untagged
wl_wlan_25        12 PVID Egress Untagged
wl_wlan_g_25      12 PVID Egress Untagged
wl_outlan_25      13 PVID Egress Untagged
wl_wlan_g_5       12 PVID Egress Untagged
wl_outlan_5       13 PVID Egress Untagged
```

### Additional info

As a work-around, a hotplug script can correct the problem.
Note that the VLAN ID assignments are hard-coded and it is a poor solution:
`/etc/hotplug.d/net/20-bridge-vlans`:
```
[ "$ACTION" = add -a "$SUBSYSTEM" = net -a "$DEVTYPE" = wlan ] && {
	sleep 1
	wanted_pvid=
	case "$INTERFACE" in
		wl_wlan*) wanted_pvid=12;;
		wl_outlan*) wanted_pvid=13;;
	esac
	if [ -n "$wanted_pvid" ]
	then
		pvid="$(bridge vlan ls dev "$INTERFACE" | sed -nre "s@$INTERFACE[[:space:]]+([0-9]+) PVID Egress Untagged\$@\1@ p")"
		if [ "$pvid" != "$wanted_pvid" ]
		then
			logger -t 20-bridge-vlans "$INTERFACE: pvid=$pvid, correcting to $wanted_pvid"
			bridge vlan del vid $pvid dev "$INTERFACE"
			bridge vlan add vid $wanted_pvid dev "$INTERFACE" pvid untagged
		fi
	fi
}
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-19:
can you check logread restarting `wpad` service
also post definitions of br_vlan_13 and alike. Normally vlans are noted eth0.123 style

F-i-f, 2023-05-20:
> can you check logread restarting `wpad` service

Are are the results:
- After boot: wrong VLAN id on wl_outlan_5 and wl_wlan_g_5.
- After `/etc/init.d/network restart`: wrong VLAN id on wl_outlan_5 and wl_wlan_g_5.
- After `wifi`: correct VLAN ids.
- After `/etc/init.d/wpad restart`: correct VLAN ids.

Logs:
1. Generated during `/etc/init.d/network restart` and waiting for CAC to complete. After that, the VLAN are wrong from wl_outlan_5 and wl_wlan_g_5, which are the two interfaces that appear after CAC is complete. See above for the output of `bridge vlan` under the _Actual behavior_ section. 
[network-restart.log](https://github.com/openwrt/openwrt/files/11521603/network-restart.log)
2.  Generated during `/etc/init.d/wpad restart` and waiting for CAC to complete. After that, VLANs are all correct. See above for the output of `bridge vlan` under the _Expected behavior_ section.
[wpad-restart.log](https://github.com/openwrt/openwrt/files/11521606/wpad-restart.log)

> also post definitions of br_vlan_13 and alike. Normally vlans are noted eth0.123 style

The full definitions are in the `/etc/config/network` file in the OP.
Is there anything else needed?
The UCI devices / kernel interfaces are named in the recommended style, eg. `br-vlan.12`.
The UCI interfaces have to be UCI identifiers, and cannot have dots, that's why they're named br_vlan_12.

robimarko, 2023-05-21:
Well, IPQ807x has no DSA support nor it has proper VLAN filtering, its all just a SW bridge

F-i-f, 2023-05-21:
> Well, IPQ807x has no DSA support nor it has proper VLAN filtering, its all just a SW bridge

Well it does seem to behave properly. If it uses the kernel's bridge code, then VLAN filtering should be working right?
Does that have any bearing on the issue at hand (wrong VLAN pvids)?

F-i-f, 2023-05-21:
I've amended the bug as VLAN ids are not the only thing wrong with DFS secondary interfaces, some other bridge parameters such as `mcast_to_unicast` and `hairpin` can be misconfigured.

brada4, 2023-05-22:
Last line seems to contradict documentation slightly
https://openwrt.org/docs/guide-user/base-system/basic-networking#config_switch_vlan

```
  list ports 'lan1'
  list ports 'lan2'
  list ports 'lan3'
  list ports 'lan4:u*'
```

F-i-f, 2023-05-23:
> Last line seems to contradict documentation slightly https://openwrt.org/docs/guide-user/base-system/basic-networking#config_switch_vlan

I'm assuming you're talking about this line:

> ```
>   list ports 'lan4:u*'
> ```

I created that config with luci (and cleaned it up by hand afterwards).
I think the link you sent only document the old-style bridge configuration.
This is using a DSA-like driver and is configured with `config bridge-vlan` stanzas, and is documented at https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial?s[]=dsa#multiple_networks_using_vlan_tagging

F-i-f, 2023-05-23:
This example https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial?s[]=dsa#multiple_networks_using_vlan_tagging uses `<port>:u*` on one untagged port (and `<port>` on the others).
But this other example https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial?s[]=dsa#firewall_zones_for_vlans uses `<port>:u*` for all untagged ports.
I can try removing the `:u*` and report if it changes anything.

Is there a good explanation of the difference between `<port>` and `<port>:u*`  for untagged ports?
Luci says it's for marking the primary VLAN on a port or bridge?
[image]

brada4, 2023-05-23:
It is just a spottable difference, otherwise yout configs look impeccable.

john-tho, 2023-05-24:
From https://github.com/openwrt/openwrt/commit/cf45caeff16256f9db777e0e652ec3a38cd476a4 (hostapd OpenWrt patches: https://github.com/openwrt/openwrt/tree/main/package/network/services/hostapd/patches),
it looks like (at some stage), hostapd https://w1.fi/cgit/hostap/tree/src/ap was patched to allow netifd https://git.openwrt.org/?p=project/netifd.git;a=blob;f=wireless.c;hb=HEAD https://lxr.openwrt.org/source/netifd/wireless.c to manage bridging in some situations.

Could we please set netifd to be more verbose: https://openwrt.org/docs/techref/netifd#debugging
Then restart it, and collect some logs

Check the hostapd config
```
grep -E '^ssid|bridge|iface|vlan' /var/run/hostapd-phy*.conf
```

As well as querying ubus before and after device gets wrong vlan (will need to change `wifi` to your interface name)
```
ubus -v call "network.interface.wifi" status | grep device
ubus -v call "network.wireless" status | grep -EA1 "ghz|section|network"
```

Might reduce log noise from the other interfaces if you can trigger this with `ifup` & `ifdown`.

Some more temporary debug prints through netifd should help to work out what/when/where (if the issue is in `netifd`): https://lxr.openwrt.org/source/netifd/wireless.c#L434

F-i-f, 2023-05-25:
@john-tho, here are the results:

> Could we please set netifd to be more verbose: https://openwrt.org/docs/techref/netifd#debugging Then restart it, and collect some logs

Sure.
Made the following changes to `/etc/init.d/network`:
```
# diff -u {/rom,}/etc/init.d/network 
--- /rom/etc/init.d/network	2023-05-23 23:29:35.000000000 -0700
+++ /etc/init.d/network	2023-05-25 14:29:07.000000000 -0700
@@ -16,8 +16,10 @@
 	init_switch
 
 	procd_open_instance
-	procd_set_param command /sbin/netifd
+	procd_set_param command /sbin/netifd -l 5 -d 15
 	procd_set_param respawn
+	procd_set_param stdout 1
+	procd_set_param stderr 1
 	procd_set_param watch network.interface
 	[ -e /proc/sys/kernel/core_pattern ] && {
 		procd_set_param limits core="unlimited"
```
Then restarted the network, waited for CAC to settle, then launched a logread capture, and restarted the network:
```
# /etc/init.d/network restart
# sleep 60
# logread -f > /tmp/logread.log &
# /etc/init.d/network restart
# sleep 60
# kill %-
```
Output [attached in logread.log](https://github.com/openwrt/openwrt/files/11569632/logread.log)

Did the same thing for a wpad reload:
```
# logread -f > /tmp/logread2.log &
# /etc/init.d/network restart
# sleep 60
# kill %-
```
Output [attached in logread2.log](https://github.com/openwrt/openwrt/files/11569801/logread2.log)

> Check the hostapd config 
> ```
> grep -E '^ssid|bridge|iface|vlan' /var/run/hostapd-phy*.conf
> ```

Output:
```
/var/run/hostapd-phy0.conf:ssid=NET1
/var/run/hostapd-phy0.conf:bridge=br-vlan
/var/run/hostapd-phy0.conf:wds_bridge=
/var/run/hostapd-phy0.conf:snoop_iface=br-vlan.12
/var/run/hostapd-phy0.conf:dynamic_vlan=0
/var/run/hostapd-phy0.conf:vlan_naming=1
/var/run/hostapd-phy0.conf:vlan_no_bridge=1
/var/run/hostapd-phy0.conf:vlan_file=/var/run/hostapd-wl_wlan_5.vlan
/var/run/hostapd-phy0.conf:ssid=NET2
/var/run/hostapd-phy0.conf:bridge=br-vlan
/var/run/hostapd-phy0.conf:wds_bridge=
/var/run/hostapd-phy0.conf:snoop_iface=br-vlan.12
/var/run/hostapd-phy0.conf:dynamic_vlan=0
/var/run/hostapd-phy0.conf:vlan_naming=1
/var/run/hostapd-phy0.conf:vlan_no_bridge=1
/var/run/hostapd-phy0.conf:vlan_file=/var/run/hostapd-wl_wlan_g_5.vlan
/var/run/hostapd-phy0.conf:ssid=NET3
/var/run/hostapd-phy0.conf:bridge=br-vlan
/var/run/hostapd-phy0.conf:wds_bridge=
/var/run/hostapd-phy0.conf:snoop_iface=br-vlan.13
[... 102 lines cut ...]
					],

```
The VLANs assignments look correct to me.
The above output does not change whether is properly configured or not. 

> Might reduce log noise from the other interfaces if you can trigger this with `ifup` & `ifdown`.

I cannot bring the wireless interfaces up or down with `ifup` or `ifdown`.
I found I could remove an wireless interface with:
```
ubus -v call hostapd config_remove '{"iface":"WIRELESS-INTERFACE-NAME"}'
```
But then I cannot re-add it without restarting `wpad`.

> Some more temporary debug prints through netifd should help to work out what/when/where (if the issue is in `netifd`): https://lxr.openwrt.org/source/netifd/wireless.c#L434

There are some key differences between [logread.log (captured during `/etc/init.d/network restart`)](https://github.com/openwrt/openwrt/files/11569632/logread.log) and [logread2.log (captured during `wifi`)](https://github.com/openwrt/openwrt/files/11569801/logread2.log).
In particular, when `/etc/init.d/network restart` runs, the two failing devices (`wl_wlan_g_5` and `wl_outlan_5`) are never claimed properly:
```
logread.log:Thu May 25 14:33:35 2023 daemon.err netifd[3317]: device_claim(555): claim Network device wl_wlan_g_5 failed: -1
logread.log:Thu May 25 14:33:36 2023 daemon.err netifd[3317]: device_claim(555): claim Network device wl_outlan_5 failed: -1
```
Log message is from https://lxr.openwrt.org/source/netifd/device.c#L555
Meaning that the [up event is never sent](https://lxr.openwrt.org/source/netifd/device.c#L553).