openwrt/openwrt #12674: ASUS RT-AX53U - problem with instalaltion via tFTP

PiotrX1, 2023-05-20:
### Describe the bug

I have problem with installation on asus rt-ax53u. I'm connecting to router via serial port and installing from tftp server.
After successful firmaware upgrade I'm getting 2 errors:

`RT-AX53U bootloader version: 1.0.0.0

Input U-Boot's IP address: 192.168.1.1
Input TFTP server's IP address: 192.168.1.2
Input IP netmask: 255.255.255.0
Input file name: openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-factory.bin

Using eth@1e100000 device
TFTP from server 192.168.1.2; our IP address is 192.168.1.1
Filename 'openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-factory.bin'.
Load address: 0x83000000
Loading: T #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ################################
         3.1 MiB/s
done
Bytes transferred = 9043968 (8a0000 hex)

*** Loaded 9043968 (0x8a0000) bytes at 0x83000000 ***

Erasing from 0x3e0000 to 0xc7ffff, size 0x8a0000 ... OK
Writting from 0x83000000 to 0x3e0000, size 0x8a0000 ... OK

*** Firmware upgrade completed! ***

Error: no Image found at offset 0x003e0000
Error: no Image found at offset 0x003e0040
=>`

Then I have to reboot by 'reset' command or physical button.

`RT-AX53U bootloader version: 1.0.0.0

CPU:   MediaTek MT7621AT ver 1, eco 3
Clocks: CPU: 880MHz, DDR: 600MHz (1200MT/s), Bus: 220MHz, XTAL: 40MHz
Model: MediaTek MT7621 reference board (NAND)
DRAM:  256 MiB
NAND:  128 MiB

Initializing NMBM ...
Signature found at block 1023 [0x07fe0000]
First info table with writecount 0 found in block 960
Second info table with writecount 0 found in block 963
NMBM has been successfully attached

Loading Environment from NMBM... *** Warning - bad CRC, using default environment

In:    uartlite0@1e000c00
Out:   uartlite0@1e000c00
Err:   uartlite0@1e000c00
Net:
Warning: eth@1e100000 (eth0) using random MAC address - b2:2c:7c:04:74:43
eth0: eth@1e100000
gpio_init ......
Hit any key to stop autoboot:  0

  *** U-Boot Boot Menu ***

[... 267 lines cut ...]
[    5.313655] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    5.321899] Rebooting in 1 seconds..`

Device restarting over and over again.
Am I doing something wrong?

### OpenWrt version

all version & newest snapshot 

### OpenWrt target/subtarget

cant boot

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-05-20:
Please try these steps:
1. Use `initramfs kernel` image instead of `factory` image.
2. Flash `initramfs` image via tftp and boot into openwrt.
3. Execute `mtd -r erase firmware` in the openwrt shell:
4. Repeat step 2
5. Upload `sysupgrade` image into `/tmp` and execute `sysupgrade -n /tmp/*.bin`

PiotrX1, 2023-05-20:
`root@OpenWrt:/# mtd -r erase firmware
Unlocking firmware ...
Erasing firmware ...
Failed to get erase block status`

Then tried install by sysupgrade. Same errors.

mrkiko, 2023-05-20:
Would you mind trying reverting commit 09a0efbe835428eaf436b17ae6709f0b22e48c5a and see how it goes?
In case it doesn't work, would you mind trying reverting commit d343a15314195023c7c77884c86df4629f1bfa76 ?

Enrico

PiotrX1, 2023-05-20:
I dont think it will help. I have same errors on stable releases. Tried since 22.03.3

DragonBluep, 2023-05-20:
This problem is caused by nand bad blocks. It's very strange that mtd does not skip bad blocks.
@PiotrX1 Do you know how to install packages via `opkg`? If yes, you can try to install `nand-utils` and execute `nandtest -m /dev/mtd5` before step 3.

PiotrX1, 2023-05-21:
```
root@OpenWrt:~# opkg install nand-utils_2.1.4-1_mipsel_24kc.ipk
Installing nand-utils (2.1.4-1) to root...
Configuring nand-utils.
root@OpenWrt:~# nandtest -m /dev/mtd5
ECC corrections: 0
ECC failures   : 0
Bad blocks     : 1
BBT blocks     : 0
005c0000: reading (1 of 4)...
 4 bit(s) ECC corrected at 005c0000
005c0000: reading (2 of 4)...
 4 bit(s) ECC corrected at 005c0000
005c0000: reading (3 of 4)...
 4 bit(s) ECC corrected at 005c0000
005c0000: reading (4 of 4)...
 4 bit(s) ECC corrected at 005c0000
Bad block at 0x02a20000 4)...
031e0000: checking...of 4)...
Finished pass 1 successfully
root@OpenWrt:~# mtd -r erase firmware
Unlocking firmware ...
Erasing firmware ...
Failed to get erase block status
```

PiotrX1, 2023-05-21:
Sysupgrade errors:

`Writing from <stdin> to kernel ...
[  256.665434] ubi0: attaching mtd7
[  256.786243] UBI: EOF marker found, PEBs from 36 will be erased
[  257.307568] ubi0 error: 0x80473ba0: error -77 while checking if PEB 305 is bad
[  257.314873] ubi0 error: 0x8046db70: failed to attach mtd7, error -77
ubiattach: error!: cannot attach mtd7
           error 77 (Bad message)
ubiformat: mtd7 (nand), size 48234496 bytes (46.0 MiB), 368 eraseblocks of 131072 bytes (128.0 KiB), min. I/O size 2048 bytes
li[  258.470657] ubi0: attaching mtd78 % complete
libscan: scanning eraseblo[  258.593544] UBI: EOF marker found, PEBs from 36 will be erased
libscan: scanning eraseblock 305 -- 83 % complete  libmtd: error!: MEMGETBADBLOCK ioctl failed for eraseblock 305 (mtd7)
        error 77 (Bad message)
ubiformat: error!: failed to scan mtd7 (/dev/mtd7)
[  259.121683] ubi0 error: 0x80473ba0: error -77 while checking if PEB 305 is bad
[  259.128987] ubi0 error: 0x8046db70: failed to attach mtd7, error -77
ubiattach: error!: cannot attach mtd7
           error 77 (Bad message)
libubi: error!: "/dev/" is not a character device
ubiupdatevol: error!: error while probing "/dev/"
              error 22 (Invalid argument)
tar: write error: Broken pipe
sysupgrade successful
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[  259.199503] reboot: Restarting system`

Djfe, 2023-05-21:
Looks similar to the issues with the RT-AX54, described in this thread:
https://forum.openwrt.org/t/openwrt-support-for-asus-ax1800hp/149891/66?u=djfe

DragonBluep, 2023-05-21:
@PiotrX1 Can you try this firmware:
[asus_rt-ax53u.tar.gz](https://github.com/openwrt/openwrt/files/11524845/asus_rt-ax53u.tar.gz)

PiotrX1, 2023-05-21:
**Thank you very much! This firmware works for me! So what's next? Will it be fixed in new versions or i will have to install custom firmware with every update?**

root@OpenWrt:/tmp# mtd -r erase firmware
Unlocking firmware ...
Erasing firmware ...
Rebooting ...

Writing from <stdin> to kernel ...
[   79.491553] ubi0: attaching mtd7
[   80.451770] ubi0: scanning is finished
[   80.455608] ubi0 error: ubi_read_volume_table: the layout volume was not found
[   80.463077] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
ubiattach: error!: cannot attach mtd7
           error 22 (Invalid argument)
ubiformat: mtd7 (nand), size 48234496 bytes (46.0 MiB), 368 eraseblocks of 131072 bytes (128.0 KiB), min. I/O size 2048 bytes
libscan: scanning eraseblock 364 -- 50 % complete
ubiformat: formatting eraseblock[   82.993641] ubi0: attaching mtd7
ubiformat: formatting eraseblock 367 -- 100 % c
[   84.176846] ubi0: scanning is finished
[   84.197041] ubi0: attached mtd7 (name "ubi", size 46 MiB)
[   84.202494] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[   84.209356] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[   84.216132] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[   84.223083] ubi0: good PEBs: 368, bad PEBs: 0, corrupted PEBs: 0
[   84.229069] ubi0: user volume: 0, internal volumes: 1, max. volumes count: 128
[   84.236269] ubi0: max/mean erase counter: 0/0, WL threshold: 4096, image sequence number: 1039141616
[   84.245386] ubi0: available PEBs: 345, total reserved PEBs: 23, PEBs reserved for bad PEB handling: 19
[   84.254726] ubi0: background thread "ubi_bgt0d" started, PID 4180
UBI device number 0, total 368 LEBs (46727168 bytes, 44.5 MiB), available 345 LEBs (43806720 bytes, 41.7 MiB), LEB size 126976 bytes (124.0 KiB)
Volume ID 0, size 34 LEBs (4317184 bytes, 4.1 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs", alignment 1
Set volume size to 39489536
Volume ID 1, size 311 LEBs (39489536 bytes, 37.6 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs_data", alignment 1
Unlocking kernel ...

Writing from <stdin> to kernel ...
sysupgrade successful
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[   87.620862] mt7530 mdio-bus:1f lan1: Link is Down
[   87.627266] br-lan: port 1(lan1) entered disabled state
[   87.632813] br-lan: port 2(lan2) entered disabled state
[   87.638209] br-lan: port 3(lan3) entered disabled state
[   87.646334] device lan1 left promiscuous mode
[   87.650764] br-lan: port 1(lan1) entered disabled state
[   87.742025] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=lan1
[   87.750913] device lan2 left promiscuous mode
[   87.755272] br-lan: port 2(lan2) entered disabled state

[   87.830202] device lan3 left promiscuous mode
[   87.834562] br-lan: port 3(lan3) entered disabled state
[   87.901517] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=lan3
[   87.910915] device eth0 left promiscuous mode
[   87.980201] reboot: Restarting system

Djfe, 2023-05-26:
This is fixed, you may close this. Backport PRs exist :)

freaked1234, 2024-01-09:
Finding this Thread I was very hopeful that it would fix my issue too, since it sounds exactly the same as mine:
https://github.com/openwrt/openwrt/issues/14371

So I flashed the factory file from @DragonBluep post via TFTP, which let me boot ONCE again (see my thread for details, booting once is all I ever get)
Then I flashed it again and applied the sysupgrade file included via luci but after successfully upgrading i get send right back into kernel panic bootloop, the same as with all the other versions and combinations I tried
I have no idea what to do with the kernel file thats included?!

Any help dearly appreciated!

DragonBluep, 2024-01-10:
> Finding this Thread I was very hopeful that it would fix my issue too, since it sounds exactly the same as mine: #14371
>
> So I flashed the factory file from @DragonBluep post via TFTP, which let me [...]

Do not skip step 3. see: https://github.com/openwrt/openwrt/issues/12674#issuecomment-1555922307

freaked1234, 2024-01-12:
OK somehow I got it working but I still think someone from the dev-team should look into this issue, since I am very sure that I did everything exactly as it should be. I could try to replicate stuff for you if needed but else I´ll stay away from updates for now.

What I did:

- Asus firmware repair tool
- Flash 22.03.5 via TFTP
- Same bootloop as before
- Flash the kernel file from the prev mentioned github post
- Boots into failsafe mode
- Flash the FW-upgrade from the github post via luci
- works stable for 3 days now, reboots and all