openwrt/openwrt #12679: Partition "kernel" doesn't end on an erase/write block

JannesMeyer, 2023-05-21:
### Describe the bug

OpenWRT forgets all settings on every reboot. `df -h` shows that it's using `tmpfs` for some reason:

```
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.8M      3.8M         0 100% /rom
tmpfs                    28.1M    228.0K     27.9M   1% /tmp
tmpfs                    28.1M     64.0K     28.1M   0% /tmp/root
overlayfs:/tmp/root      28.1M     64.0K     28.1M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

There seems to be an issue with the partition alignments? (see error message in the log)

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link RE305 v1

### Image kind

Official downloaded image

### Steps to reproduce

Happens on every boot

### Actual behaviour

```
[    1.275738] 4 fixed-partitions partitions found on MTD device spi0.0
[    1.288382] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    1.302891] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    1.317866] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    1.332414] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    1.347817] Creating 4 MTD partitions on "spi0.0":
[    1.357376] 0x000000000000-0x000000020000 : "u-boot"
[    1.372073] 0x000000020000-0x000000600000 : "firmware"
[    1.383843] 2 tplink-fw partitions found on MTD device firmware
[    1.395649] Creating 2 MTD partitions on "firmware":
[    1.405489] 0x000000000000-0x00000021b67b : "kernel"
[    1.415318] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
[    1.434714] 0x00000021b67b-0x0000005e0000 : "rootfs"
[    1.444628] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
[    1.464080] mtd: setting mtd3 (rootfs) as root device
[    1.475204] 1 squashfs-split partitions found on MTD device rootfs
[    1.487530] 0x0000005b0000-0x0000005e0000 : "rootfs_data"
[    1.499540] 0x000000600000-0x000000650000 : "config"
[    1.512600] 0x0000007f0000-0x000000800000 : "radio"
[    1.579587] rt3050-esw 10110000.esw: mediatek esw at 0xb0110000, irq 25 initialized
[    1.595664] mtk_soc_eth 10100000.ethernet eth0: mediatek frame engine at 0xb0100000, irq 5
[    1.614022] NET: Registered protocol family 10
[    1.628786] Segment Routing with IPv6
[    1.636262] NET: Registered protocol family 17
[    1.645206] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    1.670886] 8021q: 802.1Q VLAN Support v1.8
[    1.686541] VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
[    1.707525] Freeing unused kernel memory: 1228K
[    1.716519] This architecture does not have kernel memory protection.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-05-22:
> There seems to be an issue with the partition alignments? (see error message in the log)

It's a feature of squashfs which can save a little rom space.

The problem on your device is caused by rom space runs out. The support for RE305 v1 is end up now. Please use old version instead.

brada4, 2023-05-22:
kernel+squashfs are read-only under normal operation, so they need not to be aligned.

You can stretch 8MB flash a bit longer using `auc` aka `attended-sysupgrade` in place of installing updates in ever-shrinking overlay.

JannesMeyer, 2023-05-22:
@DragonBluep thank you very much! I flashed 21.02.7 and it works as expected now. I'm still a little bit confused about what actually caused this problem.

@brada4 thanks for the hint. I'll have to do some reading about that. Is [this](https://p.clonehub.org/openwrt.org/docs/guide-user/installation/attended.sysupgrade) a good starting point to learn more?

JannesMeyer, 2023-05-22:
> in place of installing updates in ever-shrinking overlay.

I'm confused. Why does the overlay fs shrink when I install an update?

brada4, 2023-05-23:
> Why the FS

because updated files install in /overlay until you sysupgrade next and kernel does not get smaller over major upgrades.

> interesting

/proc/mtd should tell if you have any space in permanent overlayfs (called rootfs_data in the list)

brada4, 2023-05-23:
`[    1.487530] 0x0000005b0000-0x0000005e0000 : "rootfs_data"`

... 192k ...

That is 3 erase blocks, absolute minimum, unlikely you can install any package or update there, it should mount after flash with erase, though any configuration change might be last.

JannesMeyer, 2023-05-23:
Thank you!! That is extremely helpful! I will give your suggestions a try when I have some time to tinker with it

closed 2023-05-23 by JannesMeyer