openwrt/openwrt #12687: Compile failure with GCC 13

timocapa, 2023-05-22:
### Describe the bug

I think somewhere between 5.15.108 and 5.15.112 a compile failure with GCC 13 has been introduced

```
kernel/workqueue.c: In function 'get_work_pwq':
kernel/workqueue.c:705:24: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  705 |                 return (void *)(data & WORK_STRUCT_WQ_DATA_MASK);
      |                        ^
kernel/workqueue.c: In function 'get_work_pool':
kernel/workqueue.c:733:25: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  733 |                 return ((struct pool_workqueue *)
      |                         ^
kernel/workqueue.c: In function 'get_work_pool_id':
kernel/workqueue.c:755:25: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  755 |                 return ((struct pool_workqueue *)
      |                         ^
cc1: all warnings being treated as errors
```

Compiles fine with GCC 12, which is probably why this didn't come up in the kernel updates. I haven't tested targets other than mt7621.

### OpenWrt version

master, as of e2f9fa42044a2660f702a9b51b14cbde24a13702

### OpenWrt target/subtarget

ramips/mt7621

### Device

-

### Image kind

Self-built image

### Steps to reproduce

1. Clone OpenWrt (master)
2. Attempt to build with GCC 13
3. Select a ramips/mt7621 target (others I haven't tested)
4. Build

### Actual behaviour

Kernel build errors

### Expected behaviour

Kernel should successfully build

### Diffconfig

```text
timo@apollo /m/l/b/openwrt (master)> ./scripts/diffconfig.sh
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_redmi-router-ac2100=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_DEFAULT_PIE=y
# CONFIG_GCC_USE_VERSION_12 is not set
CONFIG_GCC_USE_VERSION_13=y
CONFIG_GCC_VERSION="13.1.0"
CONFIG_GCC_VERSION_13=y
# CONFIG_GDB is not set
CONFIG_KERNEL_CFLAGS="-Wno-int-to-pointer-cast"
CONFIG_LUCI_SRCDIET=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PKG_ASLR_PIE_ALL=y
# CONFIG_PKG_ASLR_PIE_REGULAR is not set
CONFIG_STRIP_KERNEL_EXPORTS=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MKLIBS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2023-05-23:
Upstream pending patch: https://lore.kernel.org/lkml/20230117164041.1207412-1-arnd@kernel.org/ https://github.com/openwrt/openwrt/pull/12586#issuecomment-1546775192

timocapa, 2023-05-24:
> https://lore.kernel.org/lkml/20230117164041.1207412-1-arnd@kernel.org/ [#12586 (comment)](https://github.com/openwrt/openwrt/pull/12586#issuecomment-1546775192)

Cool, I was looking for it but I'm not that good at browsing the mailing list, so I didn't find it. 

Doesn't seem like it's currently being back ported to older than 6.5 releases, so might be worth for someone to make a PR here with the patch.

commit d8e0163e (2023-05-30): generic: import patch fixing workqueue compilation with gcc-13

Already applied in wq/for-6.5 [0].

Fixes errors in the form of:
  kernel/workqueue.c: In function 'get_work_pwq':
  kernel/workqueue.c:705:24: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
    705 |                 return (void *)(data & WORK_STRUCT_WQ_DATA_MASK);
        |                        ^
  kernel/workqueue.c: In function 'get_work_pool':
  kernel/workqueue.c:733:25: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
    733 |                 return  [... cut]
closed 2023-05-30 by openwrt-bot