openwrt/openwrt #12692: luajit is built unexpectedly

zhanghua000, 2023-05-22:
### Describe the bug

I am building a custom image on Arch Linux. When I build my custom image, I found that `luajit` is built even it is not selected in config. As Arch Linux does not provide 32bit glibc development files by default, my build job fails.

### OpenWrt version

7d0fb2cc34cb7aaa2b6eaca4f14f2c20778d3e40

### OpenWrt target/subtarget

mipsel_24kc

### Device

GL-MT1300

### Image kind

Self-built image

### Steps to reproduce

1. Prepare build environment by installing openwrt-devel package on AUR
2. Build image with config provided below

### Actual behaviour

The build fails, says `/usr/include/gnu/stubs-32.h` is not found.
Output:
```
make[3]: Entering directory '/home/zhanghua/openwrt-clean/feeds/packages/lang/luajit'
rm -f /home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/.built
touch /home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/.built_check
make -j1 -C /home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3 HOST_CC="gcc -O2 -I/home/zhanghua/openwrt-clean/staging_dir/host/include -I/home/zhanghua/openwrt-clean/staging_dir/hostpkg/include -I/home/zhanghua/openwrt-clean/staging_dir/target-mipsel_24kc_musl/host/include -m32" CROSS="mipsel-openwrt-linux-musl-" DPREFIX=/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/ipkg-install/usr PREFIX=/usr TARGET_SYS=Linux TARGET_CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" BUILDMODE=dynamic
make[4]: Entering directory '/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3'
==== Building LuaJIT 2.1.0-beta3 ====
make -C src
make[5]: Entering directory '/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/src'
HOSTCC    host/minilua.o
In file included from /usr/include/features.h:515,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/limits.h:26,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/include/limits.h:205,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/include/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/include/limits.h:34,
                 from host/minilua.c:33:
/usr/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-32.h: No such file or directory
    7 | # include <gnu/stubs-32.h>
      |           ^~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:692: host/minilua.o] Error 1
make[5]: Leaving directory '/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/src'
make[4]: *** [Makefile:113: default] Error 2
make[4]: Leaving directory '/home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3'
make[3]: *** [Makefile:92: /home/zhanghua/openwrt-clean/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/.built] Error 2
make[3]: Leaving directory '/home/zhanghua/openwrt-clean/feeds/packages/lang/luajit'
time: package/feeds/packages/luajit/compile#0.55#0.24#0.73
    ERROR: package/feeds/packages/luajit failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/luajit/compile] Error 1
make[2]: Leaving directory '/home/zhanghua/openwrt-clean'
make[1]: *** [package/Makefile:114: /home/zhanghua/openwrt-clean/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhanghua/openwrt-clean'
make: *** [/home/zhanghua/openwrt-clean/include/toplevel.mk:231：world] Error 2

```

### Expected behaviour

The image built successfully.

[... 231 lines cut ...]
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zram-swap=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
# CONFIG_ZRAM_DEF_COMP_LZORLE is not set
CONFIG_ZRAM_DEF_COMP_ZSTD=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set

```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

micmac1, 2023-05-22:
Not OpenWrt issue. luajit is in packages repo. If anything from  OpenWrt base depended on it luajit would be in this repo.

Here luajit seems to get dragged in by your nginx package. Try this change and check if issue goes away.

```
diff --git a/net/nginx/Makefile b/net/nginx/Makefile
index 16767efdd..46be842e5 100644
--- a/net/nginx/Makefile
+++ b/net/nginx/Makefile
@@ -375,7 +375,7 @@ define naxsi
   NGINX_MODULES += nginx-mod-naxsi
 endef
 
-$(eval $(call module,lua, +luajit,ngx_http_lua, Enable Lua module))
+$(eval $(call module,lua, +PACKAGE_nginx-mod-lua:luajit,ngx_http_lua, Enable Lua module))
 $(eval $(call module,stream, +@NGINX_STREAM_CORE_MODULE,ngx_stream, Add support for NGINX request streaming.))
 $(eval $(call module,ubus, +libubus +libjson-c +libblobmsg-json +@NGINX_UBUS,ngx_http_ubus, Enable UBUS api support directly from the server.))
 $(eval $(call module,dav-ext, +@NGINX_DAV +libxml2,ngx_http_dav_ext, Enable the WebDAV methods PROPFIND OPTIONS LOCK UNLOCK.))
```

If this fixes it report it to nginx maintainers in packages.

zhanghua000, 2023-05-23:
Thank you for your help, this patch fixes the issue. I will report to nginx maintainers.

closed 2023-05-23 by zhanghua000