openwrt/openwrt #12712: TP-Link Archer C6 v3.2 kernel 5.10.176 crash: irq 25: nobody cared

diederikdehaas, 2023-05-23:
### Describe the bug

I noticed an unusual routing issue that didn't happen before while the config was the same as before, so I went looking in the logs and found there was a kernel crash:

```sh
Tue May 23 18:49:37 2023 kern.warn kernel: [277934.054695] sched: RT throttling activated
Tue May 23 18:49:54 2023 kern.err kernel: [277950.723736] irq 25: nobody cared (try booting with the "irqpoll" option)
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.730526] CPU: 0 PID: 402 Comm: irq/25-mt7530 Not tainted 5.10.176 #0
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.737195] Stack : 00000000 8072f628 809b0000 80840000 00000000 00000000 00000000 00000000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.745632]         00000000 00000000 00000000 00000000 00000000 00000001 80c0bda8 24a759e9
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.754063]         80c0be40 00000000 00000000 80c0bc50 00000038 8039e8a4 ffffffea 00000000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.762491]         80c0bc5c 00000177 807d4940 ffffffff 80c0bd88 806fe1e4 00000000 80700000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.770921]         807bffcc 809c0000 80080000 80840000 00000000 8040948c 00000000 80990000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.779352]         ...
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.781876] Call Trace:
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.781895] [<8039e8a4>] 0x8039e8a4
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.787966] [<80080000>] 0x80080000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.791522] [<8040948c>] 0x8040948c
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.795080] [<80007b08>] 0x80007b08
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.798637] [<80007b10>] 0x80007b10
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.802193] [<803839ac>] 0x803839ac
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.805750] [<8007ffe4>] 0x8007ffe4
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.809307] [<8008a10c>] 0x8008a10c
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.812864] [<8008a768>] 0x8008a768
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.816423] [<80080000>] 0x80080000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.819979] [<80086de8>] 0x80086de8
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.823535] [<804c9cd0>] 0x804c9cd0
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.827092] [<8008b384>] 0x8008b384
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.830649] [<800861bc>] 0x800861bc
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.834205] [<800861bc>] 0x800861bc
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.837761] [<8039f3e0>] 0x8039f3e0
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.841318] [<8039f6b8>] 0x8039f6b8
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.844876] [<8039f774>] 0x8039f774
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.848433] [<800861bc>] 0x800861bc
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.851989] [<80080000>] 0x80080000
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.855547] [<806e5e8c>] 0x806e5e8c
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.859104] [<8039ef14>] 0x8039ef14
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.862661] [<80003448>] 0x80003448
Tue May 23 18:49:54 2023 kern.warn kernel: [277950.866215]
Tue May 23 18:49:54 2023 kern.err kernel: [277950.867781] handlers:
Tue May 23 18:49:54 2023 kern.err kernel: [277950.870135] [<824d8599>] 0x80086e20 threaded [<77ee5b56>] 0x8049965c
Tue May 23 18:49:54 2023 kern.emerg kernel: [277950.876554] Disabling IRQ #25
``

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6 v3

### Image kind

Official downloaded image

### Steps to reproduce

None really, this is the first time it happened and I've had a 20+ day uptime with 22.03.05 previously.

### Actual behaviour

I can describe the result, which I guess is due to the kernel crash, but it's effects seems rather limited compared to what I'd expected with a kernel crash.

If I'm connected to the wireless network of the Archer C6-v3 (192.168.3.1/24), then I can ssh into my router. But what I can't do is ping my main router or any host on it (192.168.2.1/24) and consequently I can't connect to the internet.
(While creating this issue, I had to switch back and forth between my wireless networks to access the Archer router and GH (to post the results of the commands above).

When I'm connected to the wireless network of my other/main router, pinging 192.168.2.3 (my Archer router in my main network) fails with ``Destination Host Unreachable``.  
Normally that's not a problem as I have a firewall rule which allows access from 192.168.2.0/24 into my Archer router on port 22, 80 and 443

### Expected behaviour

No kernel crash and normally functioning routing

### Additional info

I help out with the Debian kernel team and the ``nobody cared`` bug sounded familiar (it's kind of funny msg to see):
https://bugs.debian.org/978043
That was for ``amd64``, but the latest *found* version is 5.10.162 and it's still not marked as fixed for the 5.10.x kernel, while it is fixed in 6.1.15.
That bug is *forwarded* to https://bugzilla.kernel.org/show_bug.cgi?id=201817, but there haven't been new msgs for almost 6 months now.

The stack trace I got with my router was/is missing the extra 'bits' I'm used to, so I have no way to compare them.
(FTR: I still don't know what to actually do with stack traces)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

diederikdehaas, 2023-05-23:
I'm quite sure this is an upstream kernel bug, but I can't seem to add additional labels?

LGA1150, 2023-05-24:
`irq/25-mt7530` is an interrupt pin connected to the internal MT7530 switch, to detect link changes. As it was working before, could it be a hardware issue?

diederikdehaas, 2023-05-24:
I guess anything is possible, but as it's only a few months old, that would surprise me.
How can I find out?

For completeness, here's the full dmesg output after I rebooted the device (everything seems to work now again):
```
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Apr 27 20:28:15 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is TP-Link Archer C6 v3
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 15 pages/cpu s30480 r8192 d22768 u61440
[    0.000000] pcpu-alloc: s30480 r8192 d22768 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0003da40
[    0.000000] Readback ErrCtl register=0003da40
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119472K/131072K available (7064K kernel code, 631K rwdata, 832K rodata, 1252K init, 243K bss, 11600K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000012] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007943] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.016999] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083158] pid_max: default: 32768 minimum: 301
[    0.087917] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095120] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.105705] rcu: Hierarchical SRCU implementation.
[    0.110750] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.118855] smp: Bringing up secondary CPUs ...
[    0.124128] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.124138] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.124150] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.124223] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178632] Synchronize counters for CPU 1: done.
[    0.210898] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.210907] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.210915] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.210960] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.269953] Synchronize counters for CPU 2: done.
[    0.300555] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.300563] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.300571] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300620] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.355139] Synchronize counters for CPU 3: done.
[    0.385003] smp: Brought up 1 node, 4 CPUs
[    0.393056] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 239 lines cut ...]
[   21.592887] device lan4 entered promiscuous mode
[   21.616388] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   21.623560] 8021q: adding VLAN 0 to HW filter on device wan
[   23.293078] br-lan: port 5(wlan1) entered blocking state
[   23.298429] br-lan: port 5(wlan1) entered disabled state
[   23.304528] device wlan1 entered promiscuous mode
[   23.309827] br-lan: port 5(wlan1) entered blocking state
[   23.315247] br-lan: port 5(wlan1) entered forwarding state
[   23.322702] br-lan: port 5(wlan1) entered disabled state
[   23.717822] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   23.724768] br-lan: port 5(wlan1) entered blocking state
[   23.730144] br-lan: port 5(wlan1) entered forwarding state
[... cut ...]
[   24.011253] br-lan: port 6(wlan0) entered blocking state
[   24.016703] br-lan: port 6(wlan0) entered forwarding state
[   24.322537] br-lan: port 6(wlan0) entered disabled state
[   24.450975] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   24.458379] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   33.628994] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.635902] br-lan: port 6(wlan0) entered blocking state
[   33.641230] br-lan: port 6(wlan0) entered forwarding state
```

brada4, 2023-05-24:
It is not kernel crash, it is unhandled IRQ, the effect seems that switch does not function right anymore.
Can you try entering switch configuration via wifi+web and perform some configuration change like adding blank VLAN, or bring down unconnected port?

diederikdehaas, 2023-05-24:
I'm willing to try/test anything, but I would need some guidance as I'm rather new to OpenWrt (but not Linux). I'm also comfortable with ``uci``. IOW: If you give me a set of buttons to push and/or commands to execute, I can do that.

diederikdehaas, 2023-05-24:
> see if it restores network functionallity

After a reboot, everything seems to work normally again. I'm now connected to my Archer's wifi and I can (obviously) connect to the internet. So all the symptoms I described earlier, are now gone.

brada4, 2023-05-24:
It is some atypical/unexpected signal from the switch ASIC, with logs wiped away hard to tell which locked up first - kernel or ASIC. Sorry for not mentioned, really when the thing happens again - try to snap at least `logead` and try to reconfigure switch and do another `logread` before rebooting it.

brada4, 2023-05-24:
Network/Switch

diederikdehaas, 2023-05-24:
> Network/Switch is the place.

My `Network` menu item does not have a `Switch` submenu item. I do have:
- Interfaces
- Wireless
- Routing
- DHCP and DNS
- Diagnostics
- Firewall

brada4, 2023-05-24:
Thats rather strange. It should be there.
 `service network restart` or restart that via web, you will lose connectivity shortly or till you get to power cycle. i.e start one ssh session, start `logread -f` and in other ssh session restart network. WiFi will flip but the 1st session will have log collected. Or collect UDP syslog via WiFi.

LGA1150, 2023-05-25:
As I said before, the IRQ pin is used to detect link changes. The IRQ service routine function is as follows:

```c
static irqreturn_t
mt7530_irq_thread_fn(int irq, void *dev_id)
{
	struct mt7530_priv *priv = dev_id;
	bool handled = false;
	u32 val;
	int p;

	mutex_lock_nested(&priv->bus->mdio_lock, MDIO_MUTEX_NESTED);
	val = mt7530_mii_read(priv, MT7530_SYS_INT_STS); // Read the interrupt status
	mt7530_mii_write(priv, MT7530_SYS_INT_STS, val); // Clear the interrupt status
	mutex_unlock(&priv->bus->mdio_lock);

	// Iterate through each PHY port to handle link changes
	for (p = 0; p < MT7530_NUM_PHYS; p++) {
		if (BIT(p) & val) {
			unsigned int irq;

			irq = irq_find_mapping(priv->irq_domain, p);
			handle_nested_irq(irq);
			handled = true;
		}
	}

	// If any link changes are detected, return IRQ_HANDLED, otherwise return IRQ_NONE
	return IRQ_RETVAL(handled);
}
```

The Linux kernel has detected it had returned `IRQ_NONE` for too many times, which means the switch asserts the interrupt pin, yet there are no link changes. This may also explain why you can't connect to the internet at the same time: the switch just stops working. This is possibly a hardware issue.

brada4, 2023-05-25:
No idea how that could go stray irq....