openwrt/openwrt #12714: WNR2200 with extroot reboots then opening LUCI

SimplyKot, 2023-05-24:
### Describe the bug

WNR2200 with extroot reboots then opening LUCI

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNR2200 (16M

### Image kind

Official downloaded image

### Steps to reproduce

Add USB drive support ([(https://openwrt.org/docs/guide-user/storage/usb-drives)]):

```
opkg update
opkg install kmod-usb-storage
opkg install kmod-usb-storage-uas

opkg install usbutils
lsusb -t
```

root@OpenWrt:~# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=uas, 480M

```
opkg install block-mount
block info | grep "/dev/sd"
```

/dev/sda1: UUID="f5383639-7a98-43cd-8a0a-29174983c7bb" LABEL="extroot" VERSION="1.0" TYPE="ext4"
/dev/sda2: UUID="c8ff74ed-5e68-d901-40a7-74ed5e68d901" LABEL="data" VERSION="1.0" TYPE="ext4"
/dev/sda3: VERSION="1" TYPE="swap"

```
opkg install e2fsprogs
opkg install kmod-fs-ext4
```

Configure extroot (https://openwrt.org/docs/guide-user/additional-software/extroot_configuration):

```
DEVICE="$(sed -n -e "/\s\/overlay\s.*$/s///p" /etc/mtab)"
uci -q delete fstab.rwm
uci set fstab.rwm="mount"
uci set fstab.rwm.device="${DEVICE}"
uci set fstab.rwm.target="/rwm"
uci commit fstab
```

```
DEVICE="/dev/sda1"
mkfs.ext4 -L extroot ${DEVICE}
```

```
eval $(block info ${DEVICE} | grep -o -e "UUID=\S*")
uci -q delete fstab.overlay
uci set fstab.overlay="mount"
uci set fstab.overlay.uuid="${UUID}"
uci set fstab.overlay.target="/overlay"
uci commit fstab
```

```
mount ${DEVICE} /mnt
tar -C /overlay -cvf - . | tar -C /mnt -xf -
reboot

```
Then edit /etc/rc.local :
```

# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

export PREINIT=1
mount_root

exit 0
```

At this point extroot is working:

root@OpenWrt:~# df -h /overlay /
Filesystem                Size      Used Available Use% Mounted on
/dev/sda1                 1.9G      2.7M      1.8G   0% /overlay
overlayfs:/overlay        1.9G      2.7M      1.8G   0% /

root@OpenWrt:~# grep -e /overlay /etc/mtab
/dev/mtdblock5 /rom/overlay jffs2 rw,noatime 0 0
overlayfs:/overlay /rom overlay rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work 0 0
/dev/sda1 /overlay ext4 rw,relatime,stripe=8191 0 0
overlayfs:/overlay / overlay rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work 0 0

### Actual behaviour

Router reboots the accessing web interface

[image]

### Expected behaviour

LUCI interface successfully opened in browser.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-24:
try to catch `logread -f` in shell session while triggering reboot.

SimplyKot, 2023-05-24:
> try to catch logread -f in shell session while triggering reboot.

brada4, 2023-05-24:
The anomalous occurrence is high 'sys' CPU usage and procd CPU usage spike.
For me procd uses nothing and sys CPU does not spike above 1-2%% outside heavy traffic.

SimplyKot, 2023-05-24:
By the way  OpenWrt 21.02.7, r16847-f8282da11e works fine!

brada4, 2023-05-24:
Lua starts, that is expected(you see incrementing PID in busybox top already.
But procd is too high if it crashes linux reboots.Not sure though with logs silent.

SimplyKot, 2023-05-25:
> You can try UDP syslog during crash

Still empty logs
[image]

brada4, 2023-05-25:
I can only speculate without proof that procd crashes after eating cpu, its the thing restarting services, and services (or their scripts) connect and tell it they are up.