openwrt/openwrt #12717: the main branch and master branch?

ilxp, 2023-05-24:
### Describe the bug

What is the difference between the main branch and the master branch?

### OpenWrt version

master

### OpenWrt target/subtarget

main

### Device

x86

### Image kind

Official downloaded image

### Steps to reproduce

What is the difference between the main branch and the master branch?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tapper82, 2023-05-24:
They are the same. Just too names for the same thing.

ptpt52, 2023-05-25:
This is truly a foolish act, renaming the default branch of Git from "master" to "main" in the name of so-called political correctness.

tapper82, 2023-05-25:
We had a vote about it. The vote past.

tapper82, 2023-05-25:
One of the OpenWrt devs pleas close this. Thanks.

ilxp, 2023-05-25:
It is best to have only one name for the same branch, main or master.Just choose one

ynezz, 2023-05-25:
https://lists.openwrt.org/pipermail/openwrt-adm//2023-February/002318.html

> Change the git server side to automatically update the 'master' 
branch, whenever an update is pushed to 'main'.
It's important to have a long transition period in order to avoid 
breaking downstream users' workflows.

closed 2023-05-25 by ynezz

ilxp, 2023-05-25:
i see。
“3. Developers simply change their git configs to always push to 'main'

Once this change is well established, we can look into removing 
'master', but we should definitely take our time with that.”

lynxthecat, 2023-05-26:
OpenWrt is supposedly inclusive, but it's OK for one of the main developers to have an avatar that mocks Christianity.

lynxthecat, 2023-05-26:
The OpenWrt developers [voted in favour of this politically correct change](https://openwrt.org/voting/2023-02-27) on the basis of fostering inclusivity, and see discussion with apparent justification [here](https://lists.openwrt.org/pipermail/openwrt-adm/2023-March/thread.html), but sadly it's apparently OK for [one of the developers to have an avatar that might be offensive to Christians](https://forum.openwrt.org/u/ansuel/summary).