openwrt/openwrt #12722: ramips: reg db needs update for Country Egypt 5GHz band

momothefox, 2023-05-25:
### Describe the bug

This is the current reg db set for EG 
```
:~# iw reg get
global
country EG: DFS-ETSI
        (2402 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
```
Although according to NTRA Wireless LANs (5150 - 5350) MHz 

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

ramips

### Device

MT7615DN

### Image kind

Self-built image

### Additional info

[https://www.tra.gov.eg/wp-content/uploads/2021/11/EGY-NTRA-June21-NFAT.pdf](url)
Pages 108-110 

[https://www.tra.gov.eg/wp-content/uploads/2021/11/EGY-NTRA-June21-Chart.pdf](url)
SHF Band 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-05-25:
Its not an OpenWrt bug, if the regulatory rules are out of date then you should submit the updated ones to wireless-regdb.
https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb

momothefox, 2023-05-25:
Thank You @robimarko

closed 2023-05-25 by momothefox