openwrt/openwrt #12734: mediatek/filogic: wireguard build error

dibdot, 2023-05-26:
### Describe the bug

Continued from https://forum.openwrt.org/t/when-openwrt-23-0x-is-planned-to-be-branched-off/159775/9?u=dibdot

Wireguard no longer builds in my environment since all warnings are treated as errors:

```
make -C /opt/openwrt-buildbot/TEST/source-master/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.112 KCFLAGS="-fmacro-prefix-map=/opt/openwrt-buildbot/TEST/source-master/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/opt/openwrt-buildbot/TEST/source-master/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu May 25 21:53:00 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/opt/openwrt-buildbot/TEST/source-master/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.112 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/opt/openwrt-buildbot/TEST/source-master/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.112'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC [M]  drivers/net/wireguard/allowedips.o
drivers/net/wireguard/allowedips.c: In function 'root_remove_peer_lists':
drivers/net/wireguard/allowedips.c:80:1: error: the frame size of 1040 bytes is larger than 1024 bytes [-Werror=frame-larger-than=]
   80 | }
      | ^
drivers/net/wireguard/allowedips.c: In function 'root_free_rcu':
drivers/net/wireguard/allowedips.c:67:1: error: the frame size of 1040 bytes is larger than 1024 bytes [-Werror=frame-larger-than=]
   67 | }
      | ^
cc1: all warnings being treated as errors
make[8]: *** [scripts/Makefile.build:289: drivers/net/wireguard/allowedips.o] Error 1
make[7]: *** [scripts/Makefile.build:552: drivers/net/wireguard] Error 2
make[6]: *** [scripts/Makefile.build:552: drivers/net] Error 2
make[5]: *** [Makefile:1906: drivers] Error 2
make[5]: Leaving directory '/opt/openwrt-buildbot/TEST/source-master/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.112'
make[4]: *** [Makefile:18: /opt/openwrt-buildbot/TEST/source-master/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.112/.modules] Error 2
make[4]: Leaving directory '/opt/openwrt-buildbot/TEST/source-master/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/opt/openwrt-buildbot/TEST/source-master/target/linux'
time: target/linux/compile#5.65#2.07#7.70
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/opt/openwrt-buildbot/TEST/source-master'
make[1]: *** [target/Makefile:23: /opt/openwrt-buildbot/TEST/source-master/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/opt/openwrt-buildbot/TEST/source-master'
make: *** [/opt/openwrt-buildbot/TEST/source-master/include/toplevel.mk:231: world] Error 2
```

@Ansuel 

### OpenWrt version

r23088-bf8187d5dc

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

see diffconfig

### Actual behaviour

Build errors out during wireguard build

### Expected behaviour

Build succeded

### Additional info

The error can be muted with an addtional CFLAG, e.g. `-Wno-error=frame-larger-than= ` ... of course that doesn't solve the root cause.

Build host:
```
root@t14s:/lib# uname -a
Linux t14s 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux`
```

### Diffconfig

```text
dirk@t14s:/opt/openwrt-buildbot/TEST/source-master$ ./scripts/diffconfig.sh 
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_DEFAULT_PIE=y
# CONFIG_GDB is not set
# CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR is not set
CONFIG_KERNEL_CC_STACKPROTECTOR_STRONG=y
CONFIG_KERNEL_STACKPROTECTOR_STRONG=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PKG_ASLR_PIE_ALL=y
# CONFIG_PKG_ASLR_PIE_REGULAR is not set
# CONFIG_PKG_CC_STACKPROTECTOR_REGULAR is not set
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_2=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-05-26:
So probably triggered by `CONFIG_KERNEL_CC_STACKPROTECTOR_STRONG=y` or `CONFIG_KERNEL_STACKPROTECTOR_STRONG=y` ?

ynezz, 2023-05-26:
@zx2c4 maybe of interest?

Ansuel, 2023-05-26:
@ynezz afaik we don't apply patch to wireguard and those option are also stuff upstream, but seems strange that this warning was not detected by kernel test bot...

dibdot, 2023-05-26:
> So probably triggered by `CONFIG_KERNEL_CC_STACKPROTECTOR_STRONG=y` or `CONFIG_KERNEL_STACKPROTECTOR_STRONG=y` ?

Yep, it's triggered by "Kernel space Stack-Smashing Protection". "None" or "Regular" works OK, "Strong" breaks the wireguard build.

dibdot, 2023-05-26:
> Good now I have a way to repro and send patch upstream

Thanks!
Please note, "Strong" is the default in all of my builds, but some targets are not affected: while x86 is broken as well, IPQ40xx_generic GL-A1300 builds are OK.

Ansuel, 2023-05-26:
depends on the config for gcc stack limit it does vary based on the arch