openwrt/openwrt #12739: There is no .factory for Archer C60

castillofrancodamian, 2023-05-26:
### Describe the bug

[image]

### OpenWrt version

Snapshot

### OpenWrt target/subtarget

ath79/generic

### Device

Archer C60 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-26:
Please examine device page. It explains how to install openwrt via web UI.

castillofrancodamian, 2023-05-26:
@brada4 I don't need that (I already know how to install that way). Recently there was factory image, as well as sysupgrade.

brada4, 2023-05-26:
Install 22.03 and then snapshot?

castillofrancodamian, 2023-05-26:
@brada4 I'm saying that -factory images are not being generated.

luizluca, 2023-06-24:
This is the error:

```
/builder/shared-workdir/build/staging_dir/host/bin/tplink-safeloader -B ARCHER-C60-V1 -V r23411-68ef2d1856 -k /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tplink_archer-c60-v1-kernel.bin -r /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin -o /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin.new -j   && mv /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin.new /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin || rm -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin
dd if=/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs+pkg=202d5a3e >> /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin
[ -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tplink_archer-c60-v2-kernel.bin -a -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs+pkg=202d5a3e ]
dd if=/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs+pkg=202d5a3e >> /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-factory.bin
support-list partition too big (more than 256 bytes): Success
cp /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v3-initramfs-kernel.bin /builder/shared-workdir/build/bin/targets/ath79/generic/openwrt-ath79-generic-tplink_archer-c60-v3-initramfs-kernel.bin
[ ! -s "/builder/shared-workdir/build/key-build" -o ! -s "/builder/shared-workdir/build/key-build.ucert" -o ! -s "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin" ] || { cp "/builder/shared-workdir/build/key-build.ucert" "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin.ucert" ; usign -S -m "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin" -s "/builder/shared-workdir/build/key-build" -x "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin.sig" ; ucert -A -c "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin.ucert" -x "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin.sig" ; fwtool -S "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin.ucert" "/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-sysupgrade.bin" ; }
[ -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tplink_archer-c60-v3-kernel.bin -a -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs+pkg=202d5a3e ]
dd if=/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs+pkg=202d5a3e >> /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v3-squashfs-sysupgrade.bin
Premature end of file
7441+1 records in
7441+1 records out
3809826 bytes (3.8 MB, 3.6 MiB) copied, 0.0268576 s, 142 MB/s
Unable to load signature file: No such file or directory
/builder/shared-workdir/build/staging_dir/host/bin/tplink-safeloader -B ARCHER-C60-V2 -V r23411-68ef2d1856 -k /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tplink_archer-c60-v2-kernel.bin -r /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin -o /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin.new -j  -S && mv /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin.new /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin || rm -f /builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade.bin
7441+1 records in
7441+1 records out
3809826 bytes (3.8 MB, 3.6 MiB) copied, 0.0245878 s, 155 MB/s
```

The `support-list partition too big (more than 256 bytes): Success` happens for all and only C60 models. I don't know this support-list partition but it is "too big".

brada4, 2023-06-24:
It may need 2-stage loader from factory, i.e factory image without luci and essentially with sysupgrade and LAN with ethernet only to be able to load full openwrt. Currently you need older factory image + new sysupgrade for essentially same effect.

castillofrancodamian, 2023-06-29:
```
Directory table size 13412 bytes (13.10 Kbytes)
        44.59% of uncompressed directory table size (30077 bytes)
Number of duplicate files found 150
Number of inodes 1381
Number of files 1032
Number of fragments 52
Number of symbolic links 217
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 131
Number of hard-links 0
Number of ids (unique uids + gids) 1
Number of uids 1
        unknown (0)
Number of gids 1
        unknown (0)
8828+1 records in
8828+1 records out
4519940 bytes (4.5 MB, 4.3 MiB) copied, 0.12596 s, 35.9 MB/s
8828+1 records in
8828+1 records out
4519940 bytes (4.5 MB, 4.3 MiB) copied, 0.0288333 s, 157 MB/s
support-list partition too big (more than 256 bytes): Success
cp: cannot stat '/home/franco/Escritorio/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin': No such file or directory
Skip JSON creation for non existing file /home/franco/Escritorio/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-c60-v1-squashfs-factory.bin

Calculating checksums...

```

rotanid, 2026-11-15:
Hi — we hit this in our Gluon-based community builds (tecff) and dug into the root cause, since this issue has been open for a while.

**Root cause**

Commit 25131d6230e0 ("tplink-safeloader: add Archer C59 / C60 CA ragion code", 2023-02-20) added a fourth regional entry to the ARCHER-C60-V1 support list:

```c
.support_list =
    "SupportList:\r\n"
    "{product_name:Archer C60,product_ver:1.0.0,special_id:00000000}\r\n"
    "{product_name:Archer C60,product_ver:1.0.0,special_id:43410000}\r\n"  /* CA, added 2023-02-20 */
    "{product_name:Archer C60,product_ver:1.0.0,special_id:45550000}\r\n"
    "{product_name:Archer C60,product_ver:1.0.0,special_id:55530000}\r\n",
```

The concatenated string is `14 + 4 × 65 = 274` bytes, but the C60 **v1** layout only reserves a 256-byte support-list partition:

```c
{"support-list", 0x11500, 0x00100},
```

so `tplink-safeloader` aborts with exactly the error @luizluca quoted above:

```
support-list partition too big (more than 256 bytes): Success
```

(the `: Success` suffix comes from `error(1, 0, …)` reporting errno 0).

With the previous three entries the list was `14 + 3 × 65 = 209` bytes and fit fine — which matches the observation that 22.03 still had a factory image for this device.

**Why the C59 is unaffected**

The C59 v1 layout uses a 64 KB support-list partition (`{"support-list", 0xe70000, 0x10000}`), so the same CA addition doesn't overflow it — only the C60 family has the tight 0x100 layout. (The C59 v1 list has five entries now: global/CA/EU/RU/US.)

**Still present in current releases**

Still the case in the latest releases of every series since (23.05.6 / 24.10.8 / **25.12.6** — no factory image for any C60 variant) and in current firmware-utils master, where the four-entry list and the 256-byte partition are unchanged.

**Possible fixes**

- drop the CA entry for ARCHER-C60-V1 only (trade-off: CA-market v1 devices lose web-UI installation), or
- enlarge the `support-list` partition by borrowing from the adjacent `device-id`/`profile` partitions (like other devices with longer lists do).

The code in question lives in the [firmware-utils](https://github.com/openwrt/firmware-utils) repository (`src/tplink-safeloader.c`, the `error(1, 0, "%s partition too big …")` checks).

**Reproduce**

Build ath79/generic for `tplink_archer-c60-v1` on any release since 23.05, or run the safeloader on the build artifacts:

```
tplink-safeloader -B ARCHER-C60-V1 -V test -k kernel.bin -r rootfs.bin -o out.bin -j
```

It fails immediately with the message above.

Happy to test a patch if one of the fix directions is preferred.

brada4, 2026-11-15:
Patch is to make multiple supportlist aliases (while tryingbto find other impaired devices)