openwrt/openwrt #12747: Dual Mode USB Support for QCA953x since change to ath79 (from ar71xx) not possible

mani9876, 2023-05-27:
### Describe the bug

I own an Gl.Inet AR300M-16, if I use the latest available ar71xx Firmware (19.07.9) I am able to use the USB in Device mode (kmod-usb-chipidea, kmod-usb-gadget)

As soon as I upgrade to a ath79 Firmware (21.xx.x or 22.xx.x) I am not able to use this feature anymore. There is no UDC device in /sys/class/udc available.

Maybe the Patch supplied here:
https://github.com/adron-s/QCA953x-usb-device-mode

Has not been ported to the at79 firmware?

Thanks!

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

22.03.5

### Device

Gl.Inet AR300M-16

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-27:
Gadget modules are present but OTG module is not built?
https://www.kernel.org/doc/html/latest/usb/chipidea.html

brada4, 2023-05-30:
Kernel doc says its usb-otg-fsm module needed, that is certainly not built. Try to modprobe chipidea module and have a look in /proc/module if it is ok.

mani9876, 2023-05-30:
> Kernel doc says its usb-otg-fsm module needed, that is certainly not built. Try to modprobe chipidea module and have a look in /proc/module if it is ok.

Hello, thanks for your reply!

I may be wrong - but there is no module chipidea, or usb-otg-fsm available 

```
root@OpenWrt:/lib/modules/5.10.176# ls
ath.ko                  nf_defrag_ipv4.ko       nft_masq.ko
ath9k.ko                nf_defrag_ipv6.ko       nft_nat.ko
ath9k_common.ko         nf_flow_table.ko        nft_numgen.ko
ath9k_hw.ko             nf_flow_table_inet.ko   nft_objref.ko
cfg80211.ko             nf_flow_table_ipv4.ko   nft_quota.ko
ci_hdrc.ko              nf_flow_table_ipv6.ko   nft_redir.ko
cmac.ko                 nf_log_common.ko        nft_reject.ko
compat.ko               nf_log_ipv4.ko          nft_reject_inet.ko
configfs.ko             nf_log_ipv6.ko          nft_reject_ipv4.ko
crc-ccitt.ko            nf_nat.ko               nft_reject_ipv6.ko
crc32c_generic.ko       nf_reject_ipv4.ko       nls_base.ko
drbg.ko                 nf_reject_ipv6.ko       phy-generic.ko
ehci-fsl.ko             nf_tables.ko            ppp_async.ko
ehci-hcd.ko             nfnetlink.ko            ppp_generic.ko
ehci-platform.ko        nft_chain_nat.ko        pppoe.ko
fsl-mph-dr-of.ko        nft_counter.ko          pppox.ko
g_mass_storage.ko       nft_ct.ko               roles.ko
gpio-button-hotplug.ko  nft_fib.ko              seqiv.ko
hmac.ko                 nft_fib_inet.ko         sha256_generic.ko
jitterentropy_rng.ko    nft_fib_ipv4.ko         slhc.ko
libcomposite.ko         nft_fib_ipv6.ko         udc-core.ko
libcrc32c.ko            nft_flow_offload.ko     ulpi.ko
libsha256.ko            nft_hash.ko             usb-common.ko
mac80211.ko             nft_limit.ko            usb_f_mass_storage.ko
nf_conntrack.ko         nft_log.ko              usbcore.ko

```

brada4, 2023-05-30:
> or usb-otg-fsm available

Upstream kernel discusses that to support particular controller.

mani9876, 2023-06-05:
> > or usb-otg-fsm available
> 
> Upstream kernel discusses that to support particular controller.

Hello,
can you give me a hint on how to implement the function on my own? What files are missing in the build?
Or do you know when a patch will be available?

brada4, 2023-06-05:
in package/kernel/linux/modules/usb.mk you need to add package for otg-fsm module which is to allow reversing USB port.

mani9876, 2023-06-05:
Hello,
thank you for the hint! but I am not able to find ane otg-fsm package in usb.mk. I've also downloaded OpenWRT v19 and I am not able to find the otg-fsm module, and also no Kconfig file. Would you be so kind and give me some more infos?

Thanks!

brada4, 2023-06-05:
otg.c in chipidea directory at least in current kernel

mani9876, 2023-06-06:
> otg.c in chipidea directory at least in current kernel

You are right, but I can see that the Makefile in the chipidea directory is already including otg.o/otgfsm.o or am I wrong?

```
manuel@debian:~/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-5.10.176/drivers/usb/chipidea$ cat Makefile
# SPDX-License-Identifier: GPL-2.0
obj-$(CONFIG_USB_CHIPIDEA)              += ci_hdrc.o

ci_hdrc-y                               := core.o otg.o debug.o ulpi.o
ci_hdrc-$(CONFIG_USB_CHIPIDEA_UDC)      += udc.o
ci_hdrc-$(CONFIG_USB_CHIPIDEA_HOST)     += host.o
ci_hdrc-$(CONFIG_USB_OTG_FSM)           += otg_fsm.o

# Glue/Bridge layers go here

obj-$(CONFIG_USB_CHIPIDEA_GENERIC)      += ci_hdrc_usb2.o
obj-$(CONFIG_USB_CHIPIDEA_MSM)          += ci_hdrc_msm.o
obj-$(CONFIG_USB_CHIPIDEA_PCI)          += ci_hdrc_pci.o
obj-$(CONFIG_USB_CHIPIDEA_IMX)          += ci_hdrc_imx.o usbmisc_imx.o
obj-$(CONFIG_USB_CHIPIDEA_TEGRA)        += ci_hdrc_tegra.o

```

brada4, 2023-06-06:
It depends on config options at kernel toplevel.

mani9876, 2023-06-06:
> It depends on config options at kernel toplevel.

I added "CONFIG_USB_OTG_FSM=y" in the usb.mk usb-chipidea definition, like this:

```
define KernelPackage/usb-chipidea
  TITLE:=Host and device support for Chipidea controllers
  DEPENDS:=+USB_GADGET_SUPPORT:kmod-usb-gadget @TARGET_ath79 +kmod-usb-ehci +kmod-usb-phy-nop +kmod-usb-roles
  KCONFIG:= \
        CONFIG_EXTCON \
        CONFIG_USB_CHIPIDEA \
        CONFIG_USB_CHIPIDEA_GENERIC \
        CONFIG_USB_CHIPIDEA_HOST=y \
        CONFIG_USB_CHIPIDEA_UDC=y \
        CONFIG_USB_CHIPIDEA_DEBUG=y \
        CONFIG_USB_OTG_FSM=y
  FILES:= \
        $(LINUX_DIR)/drivers/extcon/extcon-core.ko \
        $(LINUX_DIR)/drivers/usb/chipidea/ci_hdrc.ko \
        $(LINUX_DIR)/drivers/usb/common/ulpi.ko
  AUTOLOAD:=$(call AutoLoad,39,ci_hdrc,1)
  $(call AddDepends/usb)
endef

```

But still no luck, wasn't this what you meant?

Thanks!

brada4, 2023-06-06:
Thays what upstream kernel doc says, probably try all options on?

DragonBluep, 2023-06-07:
Remove line 197 or change it to `dr_mode = "otg";` should has some help.
And you need to build your own firmware from the source. Image builder may not work.
https://github.com/openwrt/openwrt/blob/598e058080b990cf59ff84d3d50b639f44f11614/target/linux/ath79/dts/qca953x.dtsi#L197

mani9876, 2023-06-07:
> Remove line 197 or change it to `dr_mode = "otg";` should has some help. And you need to build your own firmware from the source. Image builder may not work.
> 
> https://github.com/openwrt/openwrt/blob/598e058080b990cf59ff84d3d50b639f44f11614/target/linux/ath79/dts/qca953x.dtsi#L197

Thank you, I tried it, but there is also no "ci_hdrc" in /sys/class/udc.
Any more ideas :(?

Best regards
Manuel

brada4, 2023-06-07:
You added gprmally missibg module options as per upstream kernel. It says no other magic needed. Since it is in debug mode - try connecting some device without then with otg?

mani9876, 2023-06-13:
The Hardware is in OTG mode, Pull-Up Resistor is added, and with the old OpenWrt 19 everything is working fine.
Is there a possibility to check if the otg_fsm was built correctly? I am not able to see an otg_fsm.o (Objekt File for the Linker) So I think there is an entry missing in the make file, and the edited usb.mk file, usb-chipidea definition, isn't enough.