openwrt/openwrt #12748: [22.03] bpf-headers does not compile (also in buildbot)

hnyman, 2023-05-27:
### Describe the bug

cc @nbd168 @Ansuel  

bpf-headers does not compile for 22.03 any more. 
llvm-bpf toolchain clang fails with "unknown assembler".

I first noticed the breakage for my ipq806x/R7800 build for 22.03, but then noticed that the 22.03 buildbot seems to have failed compiling it at least for the last 8 days for all targets(?). 

I built libbpf and qosify successfully for 22.03 as r20137-749cc6ffc4 on 1 May 2023, so the breakage is rather recent.
Looking at the 22.03 OpenWrt main repo changes in May, there are not so many possible culprits.  Only the two fortify-headers changes look possible?  (And also three kernel bumps in May)

Examples:
https://downloads.openwrt.org/releases/faillogs-22.03/mips_24kc/base/bpf-headers/compile.txt
```
/builder/shared-workdir/build/sdk/staging_dir/host/llvm-bpf/bin/clang --target=mips-linux --prefix=mips-linux- --gcc-toolchain=/ -fno-integrated-as -Werror=unknown-warning-option: unknown assembler invoked
scripts/Kconfig.include:47: Sorry, this assembler is not supported.
make[5]: *** [scripts/kconfig/Makefile:71: oldconfig] Error 1
```
https://downloads.openwrt.org/releases/faillogs-22.03/aarch64_cortex-a53/base/bpf-headers/compile.txt
```
  HOSTLD  scripts/kconfig/conf
/builder/shared-workdir/build/sdk/staging_dir/host/llvm-bpf/bin/clang --target=mips64el-linux --prefix=mips64el-linux- --gcc-toolchain=/ -fno-integrated-as -Werror=unknown-warning-option: unknown assembler invoked
scripts/Kconfig.include:47: Sorry, this assembler is not supported.
make[5]: *** [scripts/kconfig/Makefile:71: oldconfig] Error 1
make[4]: *** [Makefile:615: oldconfig] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/bpf-headers/linux-5.10.179'
make[3]: *** [Makefile:123: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/bpf-headers/linux-5.10.179/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
time: package/feeds/base/bpf-headers/compile#17.80#11.94#32.82
```

Strangely even the x86 seems to be detected as "mips-something":
https://downloads.openwrt.org/releases/faillogs-22.03/x86_64/base/bpf-headers/compile.txt
```
/builder/shared-workdir/build/sdk/staging_dir/host/llvm-bpf/bin/clang --target=mips64el-linux --prefix=mips64el-linux- --gcc-toolchain=/ -fno-integrated-as -Werror=unknown-warning-option: unknown assembler invoked
scripts/Kconfig.include:47: Sorry, this assembler is not supported.
make[5]: *** [scripts/kconfig/Makefile:71: oldconfig] Error 1
```

Ps.
I tried also downloading the newer llvm-bpf from master and using it, but it fails similarly

### OpenWrt version

22.03 /  r20145-30fca5413b

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2023-05-27:
I tried to find the regression point, and after quick testing it looks like the culprit might be the linux 5.10.179 version bump by @graysky2 and @hauke 

A minimal config with just the router and qosify with bpf
* compiles ok with  5cc1bd5b2 , the 5.10.178 bump
* breaks with 2677220d81, the 5.10.179 bump

The 5.10.179 bump commit is small and looks innocent, so it might be something in upstream Linux.

Note: my testing was not done in my normal 22.03 build env by selectively checking out the older commits in the kernel dirs. I will test with a fully clean repo.

hnyman, 2023-05-27:
Testing with a clean repo showed that the reason is in the 5.10.178 bump.

Looking at the Linux upstream commits, it is likely these commits:

```
2023-04-20	kbuild: check CONFIG_AS_IS_LLVM instead of LLVM_IAS
2023-04-20	kbuild: Switch to 'f' variants of integrated assembler flag
2023-04-20	kbuild: check the minimum assembler version in Kconfig
```

* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.10.y&id=83f55e6f298e9e79a38b92d771179090ff0ac771
* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.10.y&id=d5f67f6d4ec4aaf8bcef95c35a70d7e537239cb1
* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.10.y&id=c36a251011793e1c86ab7c7a0e79ce6794a9bd2c

hnyman, 2023-05-28:
Looking at the differences between 5.15 and 5.10, one possibility is 
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.15.y&id=f12b034afeb3a977bbb1c6584dedc0f3dc666f14
'scripts/Makefile.clang: default to LLVM_IAS=1'

It inverses the default of the integrated assembler or not switch from an opt-in logic to opt-out:

> This commit flips the default from opt in via LLVM_IAS=1 to opt out via LLVM_IAS=0

Note that it modifies scripts/Clang,Makefile that does not exist in 5.10, as the commit  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.15.y&id=6f5b41a2f5a6314614e286274eb8e985248aac60   has not yet separated those lines from the main Makefile

hnyman, 2023-05-28:
Patching the bpf-headers Makefile to explicitly set internal assembler in llvm/clang seems to fix compilation in 22.03.

```
diff --git a/package/kernel/bpf-headers/Makefile b/package/kernel/bpf-headers/Makefile
index 0c1a02a0a1..4de0968ec1 100644
--- a/package/kernel/bpf-headers/Makefile
+++ b/package/kernel/bpf-headers/Makefile
@@ -53,7 +53,7 @@ KERNEL_MAKE := \
        $(MAKE) -C $(PKG_BUILD_DIR) \
                ARCH=$(BPF_KARCH) \
                CROSS_COMPILE=$(BPF_ARCH)-linux- \
-               LLVM=1 CC="$(CLANG)" LD="$(TARGET_CROSS)ld" \
+               LLVM=1 LLVM_IAS=1 CC="$(CLANG)" LD="$(TARGET_CROSS)ld" \
                HOSTCC="$(HOSTCC)" \
                HOSTCXX="$(HOSTCXX)" \
```

But I do not have enough knowledge about the role of bpf-headers, that I could judge if that is right.

Any input, @nbd168  ?

vincele, 2024-08-13:
@hnyman it looks like this should be fixed by the #12752 PR merge

commit 4a9eb94b (2023-06-10): bpf-headers: fix compilation with LLVM_IAS=1

Linux 5.10.178 includes backported commits that break the compilation
of bpf-headers, as the compilation gets confused which assembler to use.
Caused by Linux upstream commits just before the .178 tag:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=v5.10.178

2023-04-20	kbuild: check CONFIG_AS_IS_LLVM instead of LLVM_IAS
2023-04-20	kbuild: Switch to 'f' variants of integrated assembler flag
2023-04-20	kbuild: check the minimum assembler version in Kconfig

Explicitly use LLVM_IAS=1 to fix things.

Fixes #12748
PR #12752 (2023-06-10): [22.03] bpf-headers: fix compilation with LLVM_IAS=1
Linux 5.10.178 includes backported commits that break the compilation of bpf-headers in 22.03, as the compilation apparently gets confused on which assembler to use. 
Caused by Linux upstream commits just before the .178 tag: 
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=v5.10.178

2023-04-20	kbuild: check CONFIG_AS_IS_LLVM instead of LLVM_IAS
2023-04-20	kbuild: Switch to 'f' variants of integrated assembler flag
2023-04-20	kbuild: check the minimum assembler version in Kconfig 

Explicitly use LLVM_IAS=1 to fix th [... cut]
closed 2024-08-14 by hnyman