openwrt/openwrt #12756: Openwrt Wireguard does not provide random/reconnect/select node functions.

Juice7310, 2023-05-29:
### Describe the bug

Openwrt Wireguard app allows user to add multiple peers, however only the last peer will only be used and all the other will never be used. If the last server is down or slow down, then the wireguard connection will not switch to other servers.
I wish the wireguard app to support random connection to peers, automatic reconnect/switch and a checkbox to allow user to manually select a preferred peer server.

### OpenWrt version

22.04

### OpenWrt target/subtarget

wireguard

### Device

general

### Image kind

Official downloaded image

### Steps to reproduce

Install wireguard app
Do the wireguard config with multiple server
Connect Wireguard

### Actual behaviour

Only the last peer server will be used for connection. For example, if I add 5 servers' IP and pub key, only the fifth server will ever be used. If the 5th server is down, then all others will not be used. The user has to manually switch.

### Expected behaviour

All the peer server will be randomly used. If one of the server is down, the connection automatically switch to another.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-29:
Wireguard select node once at start.
https://lists.zx2c4.com/pipermail/wireguard/2017-November/002028.html

Juice7310, 2023-05-29:
> Wireguard select node once at start. https://lists.zx2c4.com/pipermail/wireguard/2017-November/002028.html

I don't think so, I have 4 routers of different brands with latest openwrt installed. And they all always use the last server in the list.

brada4, 2023-05-29:
Yes, the wireguard default and only behaviour is a bit awkward.
You can have 4 wireguard interfaces and juggle them with mwan3

schm0, 2023-06-04:
You can write a script that changes the peer with the wg tool.

luizluca, 2023-06-11:
Wireguard does not deal with failures, selection of server. It will, for example, only resolve DNS at start (wg userland tool) and never again. There is no concept of connection with wireguard and the interface will be configured and up no matter if the other side is alive. This is by design. It will only fail if it cannot setup the interface, like when remote endpoint is a FQDN and it could not resolve that into an address.

You can exploit the keepalive function to monitor if the wireguard link is ok. For example, there is a script that refresh DNS (/usr/bin/wireguard_watchdog for use in cron) to detect unresponsible peers with keepalive and endpoints defined. If that happens, it brings up the interface again to reconfigure it (forcing DNS to be resolved again).

Wireguard is not a complete solution but a building block for a VPN service. It does not even have dynamic address configuration. For dealing with fallback/load balancing servers, you need to externally detect non-responsible endpoints (pinging a target? checking keepalive?), select (randomly?) one from a list of servers, configure the interface using wg and so on. Another option is to setup all targets at the same time (they must use different public keys). Then you can use a routing tool like mwan3 or, if you control both sides, some dynamic routing protocol.

In my case, I implemented a script that launches on wg ifup an ad-hoc service that monitors the link. Once it detects a hang, it selects the next target in a list of options. I monitor the link using keepalive but tests a new target using ping as waiting for the keepalive to timeout would take too much time.

Doesi, 2023-06-23:
> 
> In my case, I implemented a script that launches on wg ifup an ad-hoc service that monitors the link. 

Would you mind sharing your script?

schm0, 2023-06-23:
here is one, for mullvad vpn:

`#!/bin/sh
. /lib/functions/network.sh
LC_ALL=C
readonly script_name="wg-rotate"
readonly temp_path="/tmp/${script_name}"
readonly mullvad_config_path="/mnt/data/mullvad/"
readonly regions=".*\/(al|bg|co|cz|hr|hu|md|mk|pl|ro|rs|sk)+-[a-zA-Z]{3}-wg-[0-9]{3}\.conf$"
readonly wg_interface="wgc0"

! network_is_up wan && exit
! network_is_up "${wg_interface}" && exit

# Create temp directory
[ ! -d "${temp_path}" ] && \
	mkdir -p "${temp_path}"

# Create tmp dummy config file 
wg_config="$(mktemp -p "${temp_path}" "${wg_interface}".XXXXXX)"

# Dump current wg config into the tmp file
wg showconf "${wg_interface}" > "${wg_config}"

# Get current endpoint
endpoint_curr="$(grep -Ei 'Endpoint' "${wg_config}" | \
								 grep -Eo '([0-9]{1,3}[\.]){3}[0-9]{1,3}')"

while true
do

# Get random new endpoint from one of the configs in path specified
config="$(find "${mullvad_config_path}" -type f | \
					grep -E "${regions}" | shuf -n1)"
publickey="$(grep -Ei 'PublicKey' "${config}" | \
						 grep -Eo '([a-zA-Z0-9]|/|\+)+=$')"
endpoint="$(grep -Ei 'Endpoint' "${config}" | \
						grep -Eo '([0-9]{1,3}[\.]){3}[0-9]{1,3}')"

# Check if new endpoint is actually new
[ "${endpoint}" = "${endpoint_curr}" ] && \
	continue

# Check if endpoint is online
if ! ping -c 1 -W 1 "${endpoint}" >/dev/null 2>&1; then
	logger -t "wg-rotate" "Endpoint seems down..."
	logger -t "wg-rotate" "Selecting a new one..."
	continue
fi

# Replace publickey and the endpoint in the tmp wg conf
sed -Ei -e 's|PublicKey = .+|PublicKey = '"${publickey}"'|' \
			  -e 's|Endpoint = .+:|Endpoint = '"${endpoint}"':|' \
			  "${wg_config}"

# Print some info to syslog
logger -t "wg-rotate" "Rotating endpoint..."
logger -t "wg-rotate" "Config: ${config}"
logger -t "wg-rotate" "EndPoint: ${endpoint}"

# Load new the wg config from the tmp file
wg syncconf "${wg_interface}" "${wg_config}"

# Clean temp files
[ -d "${temp_path}" ] && \
	rm -rf "${temp_path}"

exit 0

done
`

luizluca, 2023-06-23:
> > In my case, I implemented a script that launches on wg ifup an ad-hoc service that monitors the link.
> 
> Would you mind sharing your script?

[wireguard_failover.txt](https://github.com/openwrt/openwrt/files/11852549/wireguard_failover.txt)

Use it as a hotplug iface script 

You need to set your peer entry in /etc/config/network with:

```
config wireguard_xxx 'xxx_server'
        ....
        option endpoint_host '127.0.0.1' # let the script select the endpoint
        option endpoint_port '6969' # it does not matter
        option connectivity_test_target 'ping.target.reachable.only.though.wg'
        option persistent_keepalive '25'
        list alternative_endpoints 'wgsrv1.mydomain.com:5820'
        list alternative_endpoints 'wgsrv2.mydomain.com:5820'
        list alternative_endpoints 'wgsrv3.mydomain.com:5820'
```

It will try the first alternative_endpoints and fallback in cycles over the other ones respecting the order. If the first one answers ping and we are not using it, it will try it again. It is not perfect (it relies on ping) but it gets the job done.