openwrt/openwrt #12757: [NETGEAR WNDR3400 v1] Soft-bricked router after enabling 2nd wifi radio when 1st wifi radio was enabled; toggling wifi, resetting, and/or rebooting subsequently resulted in boot-loops

NintendoManiac64, 2023-05-29:
### Describe the bug

Enabling the second wifi radio through the usual luci interface resulted in the router becoming completely inaccessible (effectively a "brick" despite lights being on and the like; rebooting and/or resetting subsequently results in boot-loops)

Curiously, when I have an ethernet cable plugged in, the corresponding ethernet light on the router still lights up despite me being unable to communicate with the router at all.

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

I am unable to retrieve this information for obvious reasons

### Device

I am unable to retrieve this information for obvious reasons

### Image kind

Official downloaded image

### Steps to reproduce

I don't know if the initial step is required, but I was originally running OpenWrt v21 of some variety and, just today, updated to v22.03.5, making sure to also have it erase all settings.

The actual update worked without issue, and I was able to apply all my usual basic settings successfully:
- root password
- custom hostname of 'black'
- setting dropbear to only be accessable via LAN
- enabling the first wifi radio
- setting wifi signal strength to the weakest listed value
- adding `set -o noclobber` to the startup script

Heck, I even successfully installed the nano package. Before installation, free disk space was 1.77MB while, afterwards, it was something like 1.50MB.

After all of that, I went and clicked the "Enable" button for the second-listed wifi radio. Immediately, I lost connection to the router and could no longer access it.  I believe that the next thing I did was try to toggle wifi via the physical wifi button the router, but I'm not 100% sure on this. Afterwards I don't remember whether I next tried to reboot the router or if I tried to reset it via the small in-set button that requires using a pin or the like, but I do know that I tried both (I just don't remember which order I did it in).

### Actual behaviour

After trying to enable the 2nd wifi radio (while the 1st wifi radio is also enabled), you will be unable to access the router and will have no network access.  Subsequently toggling wifi, rebooting, and/or resetting the router will result in boot-loops.

### Expected behaviour

Enabling the 2nd wifi radio while the 1st wifi radio is enabled should either:

- 1. actually work without crashing the router

- 2. do nothing but not crash the router

　

Furthermore, using the physical wifi button on the router itself should either:

- 1. actually toggle on/off the wifi without crashing the router

- 2. do nothing but not crash the router

　

Also, turning off the router via its physical power button and then turning it back on should not cause the router to become inoperable.

And lastly, holding down the in-set reset button with a pin should actually reset the router back to its stock OpenWrt settings.

### Additional info

I recall back on OpenWrt 21 that, when fiddling with certain settings, the router would sometimes get into a similar "inaccessable" state that would require me to reset it with a pin via the in-set reset button.  However, a key difference is that resetting the router via this in-set reset button always worked on v21.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-29:
Watch the power LED on the router at boot - 
First comes netgear bootoloader that blinks all lights or one red
Then power button blinks fast 5/s for 1s - you need to press reset button at that moment to enter safe mode, i.e 192.168.1.1, root no password, no radios.
Read wiki on how to wipe overlay filesystem.

With 8MB flash overlay filesystem is rather small and more than few config changes may fill it up to no recovery.
Likely this helps:
https://github.com/openwrt/openwrt/issues/12679#issuecomment-1558853431

NintendoManiac64, 2023-05-30:
> Then power button blinks fast 5/s for 1s

I am not seeing this behavior on my router...

I've attached a video demonstrating the boot-loop in my opening post.

brada4, 2023-05-30:
Does not loop?
Please check: https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset

NintendoManiac64, 2023-05-30:
> Does not loop?

What do you mean? You can clearly see in my video above that it is very much looping, albeit in a pattern that does not match what you or the wiki says, which is what I meant by "I am not seeing this behavior on my router".

brada4, 2023-05-30:
Try reset button anyway, it is common case but not universal that led blinks to select failsafe mode.

NintendoManiac64, 2023-05-30:
Well there were only three buttons anyway (wifi, wps, and reset) so I just tried them all - that is, immediately when turning on the device, I just repeatedly pressed a given button over and over for 2 minutes straight.

Unfortunately, there was no change - the device just kept boot-looping for those entire 2 minutes.

(and obviously the one exception is that I didn't try repeatedly pressing the button since, it being a physical mechanical button switch like you find on the GameCube rather than a relay circuit or whatever like you find on the Wii or pretty much any PC made in the last 2 decades, means that repeatedly pressing the power button would instead have the device constantly turn on and off which I don't imagine would be healthy for its hardware...)

brada4, 2023-05-30:
Physical switch is mechanical interruptor, wont do any good.

Sorry to hear this. Only serial can help sadly.

My phone initially played video as 15s long.
It is all lights blinking every 35s, boot loop for certain
Failsafe mode should have prevented that next blink of death.

NintendoManiac64, 2023-05-30:
> Only serial can help sadly.

Does it make a difference that the wiki page for this router says "Ethernet works in CFE!" or does that still require serial?

- https://openwrt.org/toh/netgear/wndr3400#debricking

The funny thing is that I actually have PC hardware with a serial port, but I don't think that alone exactly helps solve the issue and I wouldn't be surprised if purchasing the necessary hardware would be a similar price to just buying another more OpenWrt-friendly router off ebay).

Thing is, I'm actually *extremely* savvy with PC hardware (I've successfully delidded CPU IHSes with nothing but a razor blade), but software is a weak point for me (that being said, I *do* know how to use a Linux terminal or a DOS command line - I just can't ever remember commands!), so I would at least be interested if you could fill me in on what sort of hardware would actually be required so that I can weigh up my options.

brada4, 2023-05-30:
CFE (cf BIOS) needs to be told via serial to enter recover via TFTP (just like f in 2-seconds openwrt waits before mounting overlay)

You need USB serial controller that can work at standard 12V and either voltage converter or its capability to also work at 5v 3.3v 2.5v serial interface.
Voltage meter to check voltage on 4-pin serial header on SoC
Usually enough metallic surface is exposed and no soldering needed.

Or/and flash programmer

...

raenye, 2023-06-25:
@NintendoManiac64 
I have a WNDR3400v1 somewhere, will dig it up and test.

In the meanwhile, you could try to recover your machine via TFTP with [nmrpflash](https://github.com/jclehner/nmrpflash).
Perhaps you could see whether the problem exists on snapshot.

NintendoManiac64, 2023-06-25:
> you could try to recover your machine via TFTP with [nmrpflash](https://github.com/jclehner/nmrpflash).

That did not work. With or without `-m` I just get `No response after 60 seconds. Bailing out.`; when I additionally append `-B` it just stays on `Uploading [firmware_file].chk` forever and the router itself behaves no differently.

I'd actually be willing to send the router (and optionally its admittedly coil-whine-y stock AC-adapter) to someone more skilled than me to try and figure out this bug, but they'd have to at least pay for shipping (I'm in Ohio).

ValdikSS, 2023-06-30:
Just off the top of my head, have you tried different PSU? You might get boot loop due to insufficient power.

NintendoManiac64, 2023-06-30:
> Just off the top of my head, have you tried different PSU?

Erm...

> Oh and, for reference, I tried using a different power supply. In fact, the power supplies I'm using are quite overkill - one PSU is even rated for 12v 3amps when the router itself only lists a requirement of 12v 1.5amps (but, even then, I recall measuring the actual power consumption of the router back when it was on OpenWrt v21 as being more in the upper-single digit range, like 5-8 watts).

The funny thing is that I usually make these lengthy sort of posts with every detail and the kitchen sink in order to "cut to the chase" and rule out all possibilities, yet it seems that time and time again this just causes a post that is so long that people miss those additional tidbits anyway, making me question why I even bother trying to rule out all other possibilities up-front if people are going to just ask about them regardless...

raenye, 2023-07-08:
> I have a WNDR3400v1 somewhere, will dig it up and test.

I installed snapshot (r23530) today.
Cannot make any wireless phy work with either `kmod-b43`?, `kmod-brcmsmac` or `kmod-brcm-wl`, but ethernet works and the device is accessible.Which wireless driver did you use?

> Broadcom BCM4716

Same here.

NintendoManiac64, 2023-07-08:
> Which wireless driver did you use?

Uhhhhh... whatever was used in the .chk image I linked to in the first post?  I didn't change any driver stuff from the defaults and I don't think I would even know how to...

That being said, there *is* the chance that the wireless never even actually worked but... I *think* it did? Problem is that it's been long enough now that I don't even quite remember if I just automatically used ethernet without even trying wifi until I ran into the soft-brick or if I did in fact use wifi at some point.

NintendoManiac64, 2023-07-11:
> Cannot make any wireless phy work with either kmod-b43, kmod-brcmsmac or kmod-brcm-wl, but ethernet works and the device is accessible.

That being said, isn't this a major regression none-the-less consider that wifi *definitely* worked for me on 21.x? (I don't recall the exact version but, according to my browser history, I was looking up information about this router almost exactly a year ago in mid-July 2022 so whatever version of stable OpenWrt was available then is almost certainly the version I used)

raenye, 2023-07-16:
> a major regression

Old router, and Broadcom drivers have always been a mess. Personally I think it's time for you to move on...
(or install Fresh Tomato)

NintendoManiac64, 2023-07-16:
> Old router, and Broadcom drivers have always been a mess. Personally I think it's time for you to move on... (or install Fresh Tomato)

While it was my newest router, it wasn't my *only* router. I have another router that's almost identical except colored white, lacks a USB port, is only 4MB RAM/32MB NAND, but is fully supported by OpenWrt 19.07.