openwrt/openwrt #12758: armvirt: Package kmod-phylink is missing dependencies

xlighting2017, 2023-05-29:
### Describe the bug

the issue occurs after target armvirt switch to Linux Kernel 6.1 (the same .config works with Kernel 5.15)
https://github.com/openwrt/openwrt/pull/12705

the error says:

```
time: package/kernel/gpio-button-hotplug/compile#0.14#0.03#0.16
make[3]: Entering directory '/workdir/openwrt/package/kernel/linux'
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp

SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/packages/.pkgdir/kernel/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.kernel_installed
Package kmod-phylink is missing dependencies for the following libraries:
libphy.ko
make[3]: *** [modules/netdevices.mk:123: /workdir/openwrt/bin/targets/armvirt/64/packages/kmod-phylink_6.1.29-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.18#0.09#2.26
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:120: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:114: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

even though in `.config` I explicitly set 
`
CONFIG_PACKAGE_kmod-libphy=y
`
and 
`
CONFIG_PACKAGE_kmod-phylink=y
`

the full .config file is attached
[config.txt](https://github.com/openwrt/openwrt/files/11590918/config.txt)

### OpenWrt version

latest commit @main

### OpenWrt target/subtarget

armvirt/64

### Device

armvirt

### Image kind

Self-built image

### Steps to reproduce

build with clean environment, but using the .config attached

### Additional info

other users see this error too:

_Originally posted by @PolynomialDivision in https://github.com/openwrt/openwrt/issues/12705#issuecomment-1564247535_

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chunkeey, 2023-05-29:
Try:

```diff --git a/package/kernel/linux/modules/netdevices.mk b/package/kernel/linux/modules/netdevices.mk
index dd458ece4c..35374490c8 100644
--- a/package/kernel/linux/modules/netdevices.mk
+++ b/package/kernel/linux/modules/netdevices.mk
@@ -112,6 +112,7 @@ define KernelPackage/phylink
   SUBMENU:=$(NETWORK_DEVICES_MENU)
   TITLE:=Model for MAC to optional PHY connection
   KCONFIG:=CONFIG_PHYLINK
+  DEPENDS:=+kmod-libphy
   FILES:=$(LINUX_DIR)/drivers/net/phy/phylink.ko
   AUTOLOAD:=$(call AutoLoad,15,phylink,1)
 endef
``

john-tho, 2023-05-30:
Yes, it does need the `libphy` dependency, and our package is missing it.

Further, there seems to be issues building it as a module.
When I dug into this in the armvirt 6.1 PR, I saw that the armvirt64 buildbot packages are empty for `kmod-phylink` and `kmod-sfp`. Buildbot log had this warning: https://github.com/openwrt/openwrt/blob/ef98dc3b3e07f529831c3ebe719d1a98bd43b568/include/kernel.mk#L257
https://github.com/openwrt/openwrt/pull/12705#issuecomment-1565044964
I was building with `ALL_KMODS`, so expecting packages of the module, but `PHYLINK` (and `SFP`, which depended on it) was seemingly always dropped when kernel did config.
I _guessed_ this might be because phylink does not have prompt text in Kconfig.

For my notes… It is being built by `CONFIG_PACKAGE_kmod-usb-net-asix-ax88179=y` from your config.

guidosarducci, 2023-05-30:
As @john-tho pointed out in the comment, the right change is probably closer to:
```
diff --git a/package/kernel/linux/modules/netdevices.mk b/package/kernel/linux/modules/netdevices.mk
index 934ba3c4f2d2..bb62e5da9bcf 100644
--- a/package/kernel/linux/modules/netdevices.mk
+++ b/package/kernel/linux/modules/netdevices.mk
@@ -111,7 +111,9 @@ $(eval $(call KernelPackage,libphy))
 define KernelPackage/phylink
   SUBMENU:=$(NETWORK_DEVICES_MENU)
   TITLE:=Model for MAC to optional PHY connection
-  KCONFIG:=CONFIG_PHYLINK
+  KCONFIG:=CONFIG_PHYLINK \
+	   CONFIG_SWPHY=y
+  DEPENDS:=+kmod-libphy
   FILES:=$(LINUX_DIR)/drivers/net/phy/phylink.ko
   AUTOLOAD:=$(call AutoLoad,15,phylink,1)
 endef
```

But more than that, `kmod-phylib` likely needs reworking around `CONFIG_MDIO_DEVRES` and `CONFIG_MDIO_DEVICE`. Some dependencies seem reversed. And I wasn't able to reproduce the original errors during my own building and testing....

commit 3c55d869 (2023-05-31): kernel: Make kmod-phylink depend on kmod-libphy

The CONFIG_PHYLINK Kconfig option in the kernel selects CONFIG_LIBPHY.
Add this dependency to fix the all kernel modules build on MIPS malta
and armvirt with kernel 6.1.

With kernel 5.15 mod-phylink and kmod-sfp are empty packages because
no OpenWrt kmod is selecting a module which needs sfp or phylink
support.

Fixes: #12758
closed 2023-05-31 by openwrt-bot