openwrt/openwrt #12760: Wireguard taking too much space

Juice7310, 2023-05-29:
### Describe the bug

I have a few routers which has 10/50/100MB of disk space respectively. However, the wireguard app as my only app always takes all the space. It used to run Openwrt 21.0.2 on the 10MB one and it was running fine. After I updated to the latest it started to take all the space doesn't matter what router I use, and always leave me 200kb of spare space. The router can then slow down to a few kb/s and maybe crash and require a reboot.

### OpenWrt version

22.0.4

### OpenWrt target/subtarget

wireguard

### Device

general

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-29:
Could you elaborate on exact packages and versions and 1 router name?

Djfe, 2023-05-30:
We can't magically make the apps you are using smaller
https://openwrt.org/supported_devices/864_warning

If it ran fine on openwrt 21, then you might've to stay with that release on that device or buy a new device with sufficient flash for newer Linux Kernels.
Linux as well as openwrt are actively being developed and over time they grow in size.

What is possible for developers: compile openwrt yourself and shave of those kilobytes to keep the device running. For example by not compiling packages like LuCI (Web interface) or removing (kernel) features you don't need like usb support.

brada4, 2023-05-30:
100MB persistent storage filling up seems something else, but more details are needed.

Djfe, 2023-05-30:
oh you are right, I misread it as "mostly the 10mb router is affected" my bad.
I agree, we can only help, if we know which devices are affected

jow-, 2023-06-08:
Insufficient details, not reproducible.

closed 2023-06-08 by jow-