openwrt/openwrt #12761: qihoo_360t7 STA wifi can not work!

stevenke1981, 2023-05-29:
### Describe the bug

qihoo_360t7 STA wifi can not work

Hostname | OpenWrt
Model | Qihoo 360T7
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/filogic
Firmware Version | OpenWrt SNAPSHOT r23147-4b8b2f4f97 / LuCI Master git-23.145.49097-2c9f108
Kernel Version | 5.15.113

### OpenWrt version

r23147-4b8b2f4f9

### OpenWrt target/subtarget

mediatek/filogic

### Device

Qihoo 360T7

### Image kind

Self-built image

### Diffconfig

```text
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-29:
- Can it scan
- Is regulatory domain set

stevenke1981, 2023-05-30:
1. Yes, it can scan.
2. Hmm, I tried again. I found the bug is main router 's setting, Redmi AX6 must enable "WIFI 5 compatibility mode", then the 
qihoo_360t7 STA wifi work.

very thanks for your remind

closed 2023-05-30 by stevenke1981

hqvv, 2023-06-02:
Same issue here...

The issue occurs on radio0 (2G) and radio1 (5G) when upper AP in AX mode.