openwrt/openwrt #12788: new wAP AC wireless phy issue

odon-noda, 2023-06-01:
### Describe the bug

#release/22.03
#target/ipq40xx
### Issue: Wireless

1. reverse order of phy interfaces
| phyx | AS-IS | TO-BE |
| phy0 | ...5..... | ...2.4.... |
| phy1 | ..2.4.. | ....5....... |
[https://forum.openwrt.org/t/new-wap-ac-rbwapg-5hacd2hnd/93532/25](https://forum.openwrt.org/t/new-wap-ac-rbwapg-5hacd2hnd/93532/25)

2. missing wlan0 and wlan1

Impact: usteer, fast roaming ... does not work in the net of APs

### OpenWrt version

r23147-4b8b2f4f97

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MikroTik wAP ac

### Image kind

Self-built image

### Steps to reproduce

# download
wget https://downloads.openwrt.org/snapshots/targets/ipq40xx/mikrotik/openwrt-imagebuilder-ipq40xx-mikrotik.Linux-x86_64.tar.xz
# extract
xz -d < openwrt-imagebuilder-ipq40xx-mikrotik.Linux-x86_64.tar.xz  | tar -xvf
# build
export PROFILE=mikrotik_wap-ac
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
export USER_PACKAGES="-ppp -ppp-mod-pppoe \
-ip6tables -odhcp6c -kmod-ipv6 -kmod-ip6tables -odhcpd-ipv6only \
-odhcpd \
-wpad-basic-mbedtls \
-wpad-basic-wolfssl wpad-wolfssl usteer \
-iptables \
uhttpd uhttpd-mod-ubus \
libiwinfo-lua \
luci-base luci-app-firewall luci-mod-admin-full luci-theme-bootstrap \
zram-swap"
export PACKAGES=$USER_PACKAGES
export DISABLED_SERVICES="firewall dnsmasq"
export BIN_DIR=
export EXTRA_IMAGE_NAME="hnet-`date -I`"
export ADD_LOCAL_KEY=1
cd openwrt-imagebuilder-ipq40xx-mikrotik.Linux-x86_64
make image
# deploy
cd bin/targets/ipq40xx/mikrotik/
scp -p openwrt-hnet-2023-05-29-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade.bin <target_AP>
# sysupgrade
<target_AP> sysupgrade -v openwrt-hnet-2023-05-29-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade.bin

### Actual behaviour

# iw dev
phy#1
        Interface phy1-ap0
                ifindex 8
                wdev 0x100000002
                channel 52 (5260 MHz), width: 80 MHz, center1: 5290 MHz
phy#0
        Interface phy0-ap0
                ifindex 9
                wdev 0x2
                channel 13 (2472 MHz), width: 20 MHz, center1: 2472 MHz

### Expected behaviour

# iw dev
phy#1
        Interface wlan1
                ifindex 6
                wdev 0x100000002
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
phy#0
        Interface wlan0
                ifindex 7
                wdev 0x2
                channel 112 (5560 MHz), width: 80 MHz, center1: 5530 MHz

### Diffconfig

```text
# openwrt-imagebuilder-ipq40xx-mikrotik.Linux-x86_64$ cat scripts/diffconfig.sh
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
# openwrt-imagebuilder-ipq40xx-mikrotik.Linux-x86_64$
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-01:
Can you show logread lines in regard to usteer failing?

brada4, 2023-06-02:
dash in interface name could be taken as [a-z] style range with some parser? Really the logread of failures (minus proximity reports with foreign MAC addresses) could be super useful.

odon-noda, 2023-06-03:
Hi @brada4 
switched phyx issue context:

**fast roaming with the new wAP AC does not work in my network**

In the detail:
- new wAC AC snapshot does not support wlanx yet. Example:
```
iwinfo wlan0 info
No such wireless device: wlan0
```
- I use wlan0, wlan1 'logical interfaces', not directly phy0, phy1. Example:
```
ubus call network.wireless status | jsonfilter -e '$.radio0.config.channel'
iwinfo wlan0 info | grep 'Master  Channel' | awk '{ print $4 }'
```

- I use usteer for fast roaming 802.11r,k,v
```
ubus call hostapd.wlan0 rrm_nr_list
ubus call usteer local_info
ubus call usteer remote_info
```
All my APs have the same order of phy0, phy1.
I maintain small network of 8APs (2 old wap AC, 6 TP-Link Archer C5-V2)
Everything custom build from the latest 22.03.

I use some custom utilities that rely on fix assignment wlan0=5 ...
Example:
```
ubus call hostapd.wlan0 rrm_nr_list	# displays list of all 5Ghz neighbors
```

brada4, 2023-06-03:
I agree renaming wlan0 to phy1-ap0 is confusing at best.
So you are saying default wlan0/wlan1 missing cause aproblem. 
Can `ip link altname` add the name wlan0 so your scripts work?

Can you back your claim on any problems whatsoever with usteer?

odon-noda, 2023-06-04:
Hi @brada4 
```ip link altname```
I was unsuccessful trying to find it for openwrt. Any advice?
Thanks

brada4, 2023-06-04:
indeed, but somebody introducing change should have kept old names.
`ip link property add dev wlan0 altname potat0`
`ifconfig potat0`
Seems not done given default ip tool does not support this config?

john-tho, 2023-06-04:
To answer why completely different hardware has different phy order? They probe in a different order.
In the ath79 wap, 2g is SoC@18100000, 5g is PCI0: https://github.com/openwrt/openwrt/blob/b6e0a24c492537e5bbfa015e2a3638ccc53c164b/target/linux/ath79/dts/qca9556_mikrotik_routerboard-wap-g-5hact2hnd.dts#L120
In the ipq4019 wap, 2g is SoC@a000000, 5g is SoC@a800000: https://elixir.bootlin.com/linux/latest/source/arch/arm/boot/dts/qcom-ipq4019.dtsi

> `ip link altname`

You likely would need to install `ip-full`

Why not just change this file on your device to give you the wireless ifname pattern you want? https://github.com/openwrt/openwrt/blob/main/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh

> indeed, but somebody introducing change should have kept old names.

This has not changed within a stable branch.

---

It seems like the assumptions you built your scripts with were not correct, fix them? Otherwise, if you _really_ want something changed in OpenWrt, you could submit a PR to give you your desired behaviour provided a config setting (something like `option legacy_ifnames '1'` for `wifi-iface`), and you might be able to build a kernel hack, and use a module param, or DTS property as config for the phy names.

brada4, 2023-06-05:
What was the intent of random names? why eth0 does not follow the new scheme?
No similarity to any other interface namIng scheme. 
https://www.freedesktop.org/software/systemd/man/systemd.net-naming-scheme.html

Djfe, 2023-06-10:
@odon-noda appears to be using main or openwrt-23.05 on his ipq40xx wAP-AC (the issue title should be renamed accordingly), the device has only been included in the new release candidate so far.

Depending on the device and target I have seen 5GHz on phy0/wlan0 a lot already. as has been said order is depends on the order they are detected with at boot.
There are tri-band routers like the Linksys MR8300.
It has QCA9888 for 5GHz channels 100-140 (or higher) and detects this one first. then it detects ipq40xx wifi chips: 2.4ghz and 5ghz (channels 36-64)

Your script can't rely on boot order being the same across devices and you need to adjust for that.

that being said: once you upgrade all your devices to openwrt 23, then all of them will employ the new naming scheme for wifi interfaces.
Maybe we could ask for a different naming scheme (feature request), but I don't have an idea for a better one.

jow-, 2023-06-10:
Not an issue. OpenWrt never guaranteed a particular radio/phy order. Probe capabilities and decide based on that which radio to use for 2.4GHz / 5GHz operation.

closed 2023-06-10 by jow-

brada4, 2023-06-11:
The problem is  not order of devices but intetface names becoming unguessable.

Djfe, 2023-06-11:
well OP stated as the first issue
> reverse order of phy interfaces

which is nothing new, regarding ipq40xx, the old wap ac used ath79 and phy's were at different pcie addresses.
If he had bought let's say a FRITZ!Box 4040 or whatever, the phy order would've been reversed as well (even several releases back) the behavior was always like this. He just expected 0 to always be 2.4GHz, because it was the case for the few devices he deployed. it's just a number, your script needs to actually check for the phy's frequency if you need it.

second issue
> missing wlan0 and wlan1

Which was an intentional change as OP found out, if he wants to discuss changing this to something else, we/he should probably open a separate issue just for discussing that (with no regard to the wap ac device which was only relevant because he was comparing to the older 22 release without releasing he was running snapshot/23 rc1)
wlanx in the old way probably won't come back (but that's just my guess, I don't have any say in the matter, discussion is welcome imo)

Apart from that, OP never really explained, why/how this impacted
> Impact: usteer, fast roaming ... does not work in the net of APs

His scripts don't work correctly anymore in 23 rc1, but are those features actually broken? probably not, right?
Scripts can be adjusted, even if it's a hassle.

@brada4 you actually listed some valid criticism, why the new naming scheme is inferior, if you care enough, please open a new clean issue about it 👍, so people can discuss this prior to the full release of 23 (discussing this during the release candidates phase feels appropriate to me) 
https://github.com/openwrt/openwrt/issues/12788#issuecomment-1575747731
https://github.com/openwrt/openwrt/issues/12788#issuecomment-1575881429

to summarize
- altnames could be useful atleast for the next release so people have time to adjust
- ifnames with dash might need some ticks around (you should probably confirm this one first)
- No similarity to any other interface namIng scheme.
https://www.freedesktop.org/software/systemd/man/systemd.net-naming-scheme.html

Overall I wouldn't call the new ifnames random or unguessable. There are just a few more options and you might need to know the number of available phy's now (if you actually want to guess names)

> why eth0 does not follow the new scheme?

just a guess: probably because it's more common for wifi to do more than one interface per phy (broadcast more than one ssid per phy)
For eth0 you would need to do VLANs, I think.

in the end this change might break a some people's work flow

brada4, 2023-06-11:
Ty for suggestion. I will.

> ifnames with dash 

nft at least. Though no elaboration on that from OP