openwrt/openwrt #12791: blockmount: Error during USB drive mount leaves the system without /dev /proc and /sys

M95D, 2023-06-01:
### Describe the bug

By mistake, I added "noacl" to an ext2 partition in Luci / Mount points. When I saved the settings, Luci crashed.
This was the only mount point enabled in Luci. I can't remember if the partition was already mounted manually when I saved the config or not. There were two other NFS dirs mounted manually.

I had a ssh session already open. The system was still up. I used ssh to investigate what happend. Kernel log showed usual mount messages for ext2 and an error that "noacl" is no logner supported for ext2. There was no other error or message.

The real problem was that somehow the system was left without any mounts. There was no /proc, no /sys, no /dev and no /tmp mounted. All these dirs were empty.
Most commands failed without /proc and /dev and I couldn't do much. **mount** won't list anything without /proc. **mountpoint** showed they were not mounted (probably didn't work without /proc either).

I tried to mount proc, dev and sys again without success. All returned "Invalid argument", with no new message in kernel log. I doubt that kernel messages were still being recored without /dev/kmsg, as I didn't see any new message after the crash. Logread didn't work. I was afraid to do a restart.

After a few attempts to fix it remotely, ssh failed and it won't connect again. There is nothing else I can do until I get home (4 days), so I can't test if this is a random one-time error or reproducible. Even if not reproducible, it's serious enough to try to investigate probable causes. How is it even possible to umount /dev ? Shouldn't it fail with "busy" error?

Thank you.

### OpenWrt version

SNAPSHOT r22977

### OpenWrt target/subtarget

brcm2710

### Device

Raspberry Pi 3B+

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-01:
Did you mount external fs over / ?
You can mount proc and sys in other place
`mount -t procfs none /tmp/proc`
and so on

M95D, 2023-06-01:
Correction: I still have access to the system with the dbclient's  -T option (don't allocate a pty). I also have strace installed.

No. I mounted it in /mnt/work, but it's not mounted now. **ls /** still shows OpenWrt's usual files.
I tried all possible mount commands. They all fail, but there's a different error now.
```
strace -Z mount -t tmpfs none /tmp
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "none", 0x7fe1e8b848, 0) = -1 ENOENT (No such file or directory)
mount("none", "/tmp", "tmpfs", MS_SILENT, NULL) = -1 EINVAL (Invalid argument)
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=32758, si_uid=0, si_status=111, si_utime=0, si_stime=0} ---
mount: mounting none on /tmp failed: Invalid argument
+++ exited with 255 +++

strace -Z mount -t devtmpfs none /tmp
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "none", 0x7ff7d25fb8, 0) = -1 ENOENT (No such file or directory)
mount("none", "/tmp", "devtmpfs", MS_SILENT, NULL) = -1 ENODEV (No such device)
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=32767, si_uid=0, si_status=111, si_utime=0, si_stime=0} ---
mount: mounting none on /tmp failed: No such device
+++ exited with 255 +++

strace -Z mount -t proc proc /proc
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
mount("proc", "/proc", "proc", MS_SILENT, NULL) = -1 EINVAL (Invalid argument)
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=304, si_uid=0, si_status=111, si_utime=0, si_stime=0} ---
mount: mounting proc on /proc failed: Invalid argument
+++ exited with 255 +++
```

brada4, 2023-06-01:
Why would you mount readonly fs over useful filesystem?
```
mkdir /tmp/omg
mount -t proc proc /tmp/omg
```
You should be able to mount /dev and bind-mount old / to some directory and remove entry remounting over /

M95D, 2023-06-01:
@brada4 
```
root@RPi3OpenWrt:~# mkdir /tmp/omg
root@RPi3OpenWrt:~# mount -t proc proc /tmp/omg
mount: mounting proc on /tmp/omg failed: Invalid argument
```

M95D, 2023-06-01:
More info:

1) It's not caused by the noacl option. It fails even with no options.

2) Mount works during boot, but fails if the same mount is disabled and reenabled after boot.

3) I enabled writing the system log in a file (not in tmpfs) and there's an error saved when the crash happens:
```
Fri Jun  2 00:25:33 2023 daemon.err blockd: failed to symlink /mnt/work->/tmp/run/blockd/sda (17) - File exists
```
/tmp/run/blockd/sda does not exist before applying the config.
/mnt/work is the dir where I want it to be mounted! Why would it try to create a symlink instead ?

brada4, 2023-06-02:
Please compare `block detect` output with `/etc/config/fstab`, and post highlighted diferences even if they led you to resolution.

mcbridematt, 2023-06-02:
I encountered something similar a while ago and have a draft patch. I haven't submitted it yet as it hasn't been regression tested on a "traditional" OpenWrt (MTD/NOR only) system yet.

If you can try with this patch, that would be helpful to see if it's the same issue.
https://gitlab.com/traversetech/ls1088firmware/openwrt/-/commit/c69bcba146e0bcd16a461046a652cf09a465ae83

M95D, 2023-06-03:
> Please compare `block detect` output with `/etc/config/fstab`, and post highlighted diferences even if they led you to resolution.

There are no differences.

> I encountered something similar a while ago and have a draft patch. I haven't submitted it yet as it hasn't been regression tested on a "traditional" OpenWrt (MTD/NOR only) system yet.

The description matches what I see on my system.

> If you can try with this patch, that would be helpful to see if it's the same issue. https://gitlab.com/traversetech/ls1088firmware/openwrt/-/commit/c69bcba146e0bcd16a461046a652cf09a465ae83

Will test.

M95D, 2023-06-04:
> If you can try with this patch, that would be helpful to see if it's the same issue.
> https://gitlab.com/traversetech/ls1088firmware/openwrt/-/commit/c69bcba146e0bcd16a461046a652cf09a465ae83
I build fstools with the patch and I installed it over the old package (opkg install --force-reinstall fstools.ipk)
It didn't fix the problem.

New info:
If the drive is mounted at boot, the crash happens even when the mount is disabled in Luci and changes are saved.

mcbridematt, 2023-06-04:
> (opkg install --force-reinstall fstools.ipk)

The `block` program is actually in the `block-mount` ipk/package. Try reinstalling that one.

M95D, 2023-10-25:
> I haven't submitted it yet as it hasn't been regression tested on a "traditional" OpenWrt (MTD/NOR only) system yet.

I used the patch on a "traditional" router with MTD: Linksys WRT-1900ACv1. It didn't cause any problems.

closed 2024-12-04 by M95D