openwrt/openwrt #12805: MTU of the wlan1 interface does not work

viniribeirossa, 2023-06-03:
### Describe the bug

I define an MTU for the wlan1 interface, but when restarting the router it returns to the default value, 1500

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

rapims/mt7621

### Device

D-Link DIR-878 A1

### Image kind

Official downloaded image

### Steps to reproduce

. I define mtu for wlan1 interface and it works, but when I restart the router it goes back to the default value, 1500.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-04:
How does the `/etc/config/network` section for wlan1 device look?

viniribeirossa, 2023-06-05:
```
config device
	option name 'wlan1'
	option mtu '1436'

```
After 

```
wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu
1436 qdisc noqueue master br-lan state UP qlen 1000
    link/ether f4:8c:eb:b7:de:f2 brd ff:ff:ff:ff:ff:ff
```
When I restart the router it goes back to the default value of 1500. And any changes I make to the wireless also go back to the default values

brada4, 2023-06-07:
Does not work on jumbo-capable physical interface without IP address assigned either. 
Just curious why you are reducing wlan mtu? It is capable of a bit more than 2k mtu always.

viniribeirossa, 2023-06-08:
The mt76 drivers are not fully compatible on wireless, the loading of websites becomes slow. So I want to reduce mtu of wlan1 to get better wireless browsing

brada4, 2023-06-08:
MSS clamping does not work for forwarded traffic presently especially if your uplink is not 1500MTU one. Please check https://github.com/openwrt/openwrt/issues/12112#issuecomment-1523745180 (patch -p1 <that.git.diff)

The problem of MTU not staying over reboot if no IP is configured on physical interface is still there.

viniribeirossa, 2023-06-08:
Aa interfaces wlan0 and wlan1 are connected to default interface br-lan

brada4, 2023-06-08:
This fixes everything going towards internet. If you get packet loss between 2 wifi interfaces depending on MTU its worse wrong than the common PMTU blackholes on the internet.

viniribeirossa, 2023-06-08:
I have slow performance on my wifi to load some websites. I have a 400MB internet down/up

brada4, 2023-06-08:
First try would be to clamp MSS to MTU on visible interface, as you confirm problem is not observable over br-lan only.

brada4, 2023-06-08:
please try.

viniribeirossa, 2023-06-08:
I did this on the wan interface, it forwards to all interfaces to fix

brada4, 2023-06-08:
`nft list ruleset` has to position packet mangling in right place to get NAT-ed traffic too
you can add "counter" between rst and option to see if it gets touched.
(restart firewall service after patching uc files on disk)
```
        chain mangle_postrouting {
                type filter hook postrouting priority mangle; policy accept;
                oifname { "eth0", "wg0" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
        }

        chain mangle_input {
                type filter hook input priority mangle; policy accept;
    // wrong place
        }

        chain mangle_output {
                type route hook output priority mangle; policy accept;
   // wrong place
        }
```
If it was wifi driver problem you should have regular glitches isolated on br-lan and not specific to external sites.

viniribeirossa, 2023-06-08:
Can you give me the steps to do this via LuCI or CLI?

viniribeirossa, 2023-06-08:
mine is like this 

```
    chain mangle_postrouting {
                type filter hook postrouting priority mangle; policy accept;
        }
        chain mangle_input {
                type filter hook input priority mangle; policy accept;
        }
        chain mangle_output {
                type route hook output priority mangle; policy accept;
        }
        chain mangle_forward {
                type filter hook forward priority mangle; policy accept;
                iifname "pppoe-wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone
wan IPv4/IPv6 ingress MTU fixing"
                oifname "pppoe-wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone
wan IPv4/IPv6 egress MTU fixing"
        }
```

brada4, 2023-06-08:
You have to patch ucode scripts as in linked issue - the code section has to be a file to passed as input to patch command.

Yes, they US are text script files, where in patch it says `/a/root/usr/` you might want to edit out `/root` part and just cd / and run patch command, it is to install, but small , i promise.

at mangle_forward level routing tables' outcome including MTU is not known, so nothing gets adjusted.
since you mention pppoe-wan, i assume that is <1500 MTU, so you see the consequence. Large CDN-s tune down MSS in server end to make people happy, so the "slowness" is not universally hitting you.

viniribeirossa, 2023-06-08:
How can I do this? Can you give me the steps? Please

brada4, 2023-06-08:
First select and copy code in https://github.com/openwrt/openwrt/issues/12112#issuecomment-1523745180
[image]

Then paste it to notepad(++) and change `a/root/usr` to plain `a/usr`

Then ssh to router and:
```
cat > firewall.patch
(paste the new file content)
Ctrl-D
```

Verify:
`cat firewall.patch` -> is it same as in notepad?

Now
`opkg install patch`
```
cd /
patch -p1 < /root/firewall.patch
(some warnings regarding line numbers likely)
fw4 restart
```

viniribeirossa, 2023-06-08:
I'm without my PC right now, can I do it with my Android phone?

brada4, 2023-06-08:
SSH client is needed. Not that it is the only one, but I am happy with termux and unexpected keyboard from f-droid store.
Then it is up to your precision character selection skill. I'd better wait for a full screen if I was you.

brada4, 2023-06-09:
Regarding MTU keeping->
MTU is set by netifd to specified mtu setting
MTU is reset by netifd to unspecified mtu6 setting of 1500 because of automatic LL address !!!
MTU is reset back to 1500 by hostapd with no chance to configure !!!