openwrt/openwrt #12815: Xiaomi AX3600 IPQ8074 problem witch WiFi - exception and radio is down

kmaras77, 2023-06-04:
### Describe the bug

Exception in kernel log, radio IPQ8074 down

 [47.657144] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received: 
[   47.657144] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01713-QCAHKSWPL_SILICONZ-1
[   47.657144] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
[   47.657144] 
[   47.657144]     :Excep  :0 Exception detectedparam0 :zero, param1 :zero, param2 :zero.
[   47.657144] Thread ID      : 0x00000061  Thread name    : WLAN_SCHED1  Process ID     : 0
[   47.657144] Register:
[   47.657144] SP : 0x4bfd1858
[   47.657144] FP : 0x4bfd1890
[   47.657144] PC : 0x4b28e9f8
[   47.657144] SSR : 0x00000003
[   47.657144] BADVA : 0x00000000
[   47.657144] LR : 0x4b28e9d8
[   47.657144] 
[   47.657144] Stack Dump
[   47.657144] from : 0x4bfd1858
[   47.657144] to   : 0x4bfd21e8
[   47.657144] 
[   47.704048] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
[   47.726307] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
[   47.734348] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
[   47.742001] ath11k c000000.wifi: failed to transmit frame -108
[   47.767025] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[   47.801979] ath11k c000000.wifi: failed to transmit frame -108
[   47.818950] ath11k c000000.wifi: failed to transmit frame -108
[   47.819011] ath11k c000000.wifi: failed to transmit frame -108
[   47.823746] ath11k c000000.wifi: failed to transmit frame -108
[   47.853377] ath11k c000000.wifi: failed to transmit frame -108
[   47.854953] ath11k c000000.wifi: failed to transmit frame -108
[   47.858117] ath11k c000000.wifi: failed to transmit frame -108
[   47.864038] ath11k c000000.wifi: failed to transmit frame -108
[   47.869741] ath11k c000000.wifi: failed to transmit frame -108
[   53.033344] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
[   53.033392] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
[   56.768263] ath11k_warn: 112 callbacks suppressed
[   56.768286] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   56.771979] ath11k c000000.wifi: failed to send pdev bss chan info request
[   56.781111] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   56.786331] ath11k c000000.wifi: failed to send pdev bss chan info request
[   56.794966] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   56.800710] ath11k c000000.wifi: failed to send pdev bss chan info request
[   56.809974] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   56.815150] ath11k c000000.wifi: failed to send pdev bss chan info request
[   56.823887] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   56.829530] ath11k c000000.wifi: failed to send pdev bss chan info request
[   61.883324] ath11k_warn: 100 callbacks suppressed
[   61.883345] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
[   61.887035] ath11k c000000.wifi: could not request fw stats (-108)
[   61.893487] ath11k c000000.wifi: failed to request fw pdev stats: -108
[   61.899685] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   61.906068] ath11k c000000.wifi: failed to send pdev bss chan info request
[   64.300970] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   64.301021] ath11k c000000.wifi: failed to send pdev bss chan info request
[   64.307831] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
[   64.314411] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 0
[   64.320886] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
[   66.893407] ath11k_warn: 38 callbacks suppressed
[   66.893430] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   66.897122] ath11k c000000.wifi: failed to send pdev bss chan info request
[   66.905972] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   66.911357] ath11k c000000.wifi: failed to send pdev bss chan info request
[   66.920528] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   66.925805] ath11k c000000.wifi: failed to send pdev bss chan info request
[   66.934273] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[   66.940176] ath11k c000000.wifi: failed to send pdev bss chan info request
[   66.948939] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[... 307 lines cut ...]
r23262-b6e0a24c49

### OpenWrt target/subtarget

 ipq807x/generic

### Device

 Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Build image and radio is down after few seconds.

### Actual behaviour

Exception in kernel logs, and there is no WiFi.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dick902, 2023-06-05:
I'm also encountered ŵith the same issue on Xiaomi AX9000, Xiaomi AX3600 and Dynalink DL-WRX36 with  the latest firmware WLAN.HK.2.9.0.1-01713!
After rollback to the previous firmware, problem solve.

robimarko, 2023-06-05:
Did the previous 2.9.0.1-01385 FW work fine?

raulmorales, 2023-06-05:
Hi,

in version r23147 and 2.9.0.1-01385 FW it works perfectly well

`[   10.097023] ath: Country alpha2 being used: US
[   10.097027] ath: Regpair used: 0x3a
[   10.107845] ath11k c000000.wifi: ipq8074 hw2.0
[   10.107882] ath11k c000000.wifi: FW memory mode: 1
[   10.111493] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
[   10.116202] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   11.402107] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   11.406780] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.446226] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[   11.453667] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   11.453718] ath11k c000000.wifi: fw_version 0x290c84a5 fw_build_timestamp 2023-03-25 07:34 fw_build_id WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
[   11.703993] ath11k c000000.wifi: htt event 48 not handled`

kmaras77, 2023-06-05:
With FW WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1 and OpenWrt 23.05-SNAPSHOT r23004 it generally works, but some errors occur and WLAN is disconnected for a while:

there is error in event log:
[30562.239254] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 10
[33281.831166] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
[35098.385257] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 4
[35107.585228] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[35763.663070] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1

robimarko, 2023-06-05:
Ok, I made a PR to revert to previous FW version.

closed 2023-06-05 by openwrt-bot